OpenTracksApp/OpenTracks #999: Export track to CSV

dennisguse, 2021-10-29:
Mainly for analytics purpose and debugging: it would be great to export the trackpoints into a really simple CSV format.

Import is optional.

pstorch, 2021-10-29:
Just curios: is it for your own development purposes? You know you can directly access the database via the AppInspector from AndroidStudio, while you debug the app, query the trackpoints table of with a trackid and export as CSV.

dennisguse, 2021-10-29:
Yes it is - sometimes AppInspector is just not enough:
* I have some devices with development mode disabled (for reasons)
* I don't always carry my development machine with me

For weird cases, I often use a spreadsheet program and then CSV is choice to go.
That would even work on the phone itself :sunglasses:

pstorch, 2021-10-29:
Ah yes, I already guessed that you have a special usecase for this. Just wanted to make sure you know that AppInspector DB access. Because I found out rather late 😉

dennisguse, 2021-10-30:
@pstorch +1
actually, I did not use the AppInspector that often so far :)

MrS1R1US, 2022-02-04:
It's very good idea!! It would be nice to choose the export method to csv, broken down by activity and time week / month / year

dennisguse, 2022-02-04:
@MrS1R1US That was not the intention, but we could also include something like this.

brianjmurrell, 2022-04-19:
How do you include all of the various extension items that may or may not exist from one track to another?

dennisguse, 2022-04-19:
@brianjmurrell Give me 5min and then you will see :)

commit ab88127c (2022-04-19): Add CSV as export format (TrackPoints only).

Fixes #999.
commit 76b5bd5e (2022-04-19): Add CSV as export format (TrackPoints only).

Fixes #999.
commit 048ddcf9 (2022-04-19): Add CSV as export format (TrackPoints only).

Fixes #999.
commit c3369ca1 (2022-04-23): Add CSV as export format (TrackPoints only).

Fixes #999.
closed 2022-04-23 by dennisguse