OpenTracksApp/OpenTracks #1000: OT does not detect Amazfit Bip paired to GB as a HR sensor on Android 6

ElongatedVeggie, 2021-10-30:
**Describe the bug**
After enabling all the required toggles for Amazfit Bip on GadgetBridge to be used by 3rd party apps ("Visible while connected" and according to GB wiki, `You can enable this feature in the Device specific settings: 3rd party realtime HR access. Do note, that some devices seem to have this actually ON all the time and it cannot be disabled`, which they listed "Amazfit Bip" as one of those devices that can not disable "3rd party realtime HR access"), my Android 6 device does not see my watch as a BLE HR device, ***while Android 8 device does detect it as a HR device***.

I have verified that my BLE HR sensor works using "nRF Toolbox" (see: [img](https://plik.artemislena.eu/file/2onGTFs9ANP0wbfi/ZY5xVy4LFpF8A1gH/db8c0801-ab65-46c6-8d65-4e3e9bf3b051.png)).

**To Reproduce**
1. Turn on all required switches on GB for discovery
2. Go to Bluetooth > HR in OT on Android 6
3. See: "None"

**Mobile information**
 - Device: LG G4 and LG G5 
 - OS: Android 6 and Android 8
 - Version "v3.20.2/v3.20.2" (Code: 4448)
 - GadgetBridge Version: 0.61.0 

**Wearable information**
 - Device: Amazfit Bip
 - Firmware: 1.1.6.48 Non-Latin

**Additional information**
Since there are no clear indicator of bugs, no crashes or anything that is giving an error and me having never worked on BLE device detection, I do not know what additional info I can provide (such as logs) to help with debugging and fixing any problems, if there are. I can however provide requested information if the way to obtain them does not require root privileges (I can use ADB).

dennisguse, 2021-10-30:
Sounds like an issue with Android's BLE subsystem on Android 6 (it was still kind of new back then).

For the device selection, we tell Android to look for devices that announce HR capability in their announcement packets.
However, the announcement information might not contain this information (not the case here as we have a workaround for Amazfit in place and works for you on Android 8) or the announcement information is not processed by Android correctly or the filter is not applied by Android correctly.
This might fix your problem by disabling the filter and just show all devices: https://github.com/OpenTracksApp/OpenTracks/tree/bleDiscoveryDisabled%231000

PS/ If the announcement packet does not contain the HR service info (the device may not primarily be a HR sensor and space is limited in these packages/ it is consuming energy), then you need to establish a connection to a device and do a service discovery. That's what the nRF toolbox is doing, but it is not really convenient for sensor selection (also it is way more complicated to implement).

ElongatedVeggie, 2021-10-30:
>This might fix your problem by disabling the filter and just show all devices: https://github.com/OpenTracksApp/OpenTracks/tree/bleDiscoveryDisabled%231000

I will test it out in the morning. Will it ever be merged to the main branch as a regular feature where you can select between those 2 filter modes?
Thanks.

dennisguse, 2021-10-30:
From my perspective, I would rather not as it is quite some effort to maintain - the number of tiny details quickly add up.
Let's see if it helps first :)

ElongatedVeggie, 2021-10-31:
>Let's see if it helps first :)

Yes. After a painful Android Studio experience, it does work. It shows the device and it gives the heart rate data.

>it is quite some effort to maintain

I understand. I guess I could hack it each release myself because I am the one of the only users with this problem it seems.

dennisguse, 2021-10-31:
Cool!

If you want to create a PR for this; go ahead.
Yesterday, I realized that this code is not touched that often anymore.
Adding such an option would be okay from a maintenance perspective ;)

A really cool solution would to enhance the BLE search dialog by adding a button to search without the filter (in case nothing is found).

ElongatedVeggie, 2021-10-31:
I wish I could, but I do not know how to do it. I would be a tester for that feature though, of course.

This is all from me. Thanks for the hack!
I will leave closing the issue to you, in case there is something to add.

dennisguse, 2021-11-01:
Not that much to add - every now and then you can just apply the change and that's it.
That should not be too much of an issue :D

closed 2021-11-01 by dennisguse

dennisguse, 2021-12-28:
I added an option to disable the filtering of services.
Can you check that the device is now discovered?
Changes will be in tomorrows nightly build.

ElongatedVeggie, 2021-12-28:
Tested, it does work as expected. Thanks for that!

dennisguse, 2021-12-28:
Thanks for the quick feedback!
Have fun using it!

PR #1064 (2021-12-28): Settings: add option to disable filter of BLE sensors.
Fixes #1000.

Some devices do not announce supported service in their discovery package and this prevents users to use those with OpenTracks.
Affected devices are usually smartwatches that can measure heart rate.
commit bb290e2e (2021-12-28): Settings: add option to disable filter of BLE sensors.

Fixes #1000.