OpenTracksApp/OpenTracks #1003: Proposal to obtain better GPS speed values

vlmendz, 2021-10-31:
**Is your feature request related to a problem? Please describe.**
Using OpenTracks I have noticed some problems with the GPS speed (for example, maximum speed lower than the current one). Other people have reported related problems (e.g. #978, #990).

**Describe the solution you'd like**
Modify the way OpenTracks processes GPS speed. The proposed changes are:

On TrackStatisticsUpdater:

1) Use a lower value for MAX_ACCELERATION (at least for positive acceleration). Tentatively it could be around 0.7 - 1g.

2) Calculate the acceleration with respect to the previous received location (preferably to the last one with a valid acceleration) and not to the last one recorded.

3) Discard the use of speedBuffer_mps (with a working distance and low speed filter and a more effective acceleration filtering perhaps better speed values could be obtained).

4) Move the acceleration filter to TrackRecordingManager.onNewTrackPoint() so that all the locations filtering is done in one place and the statistics correspond to what is actually recorded.

dennisguse, 2021-10-31:
Follow up to https://github.com/OpenTracksApp/OpenTracks/discussions/994

dennisguse, 2021-11-15:
Did I got it right? 

*  use the covered distance from the previously stored location to the current one to derive the speed?
and then ignore the GPS provided speed (2.; and then removing speedBuffer_mps [I really like this idea])?
  I mean, speedBuffer_mps should not be required for BLE-based sensors...

* (regarding 4.) I am not sure as this is (more or less) specific for GPS.
   (my assumption) when using a BLE sensor for cycling/ running, we do not need to filter for this. Those sensor are less likely to have such inaccurate speed that this might be required.

vlmendz, 2021-11-16:
(regarding 4.) I am not sure as this is (more or less) specific for GPS.
All of the above is specific to GPS (not to other types of sensors that determine speed "mechanically").

The idea is to directly use GPS speed, using a minimum value for stop detection (already fully implemented in 3.22.0 and working very well) and a maximum acceleration value to exclude "peaks" and positions probably determined with poor quality (an improbable acceleration value serves as an unreliable positioning indicator).
 
It is better to calculate the acceleration as it says in (2) to avoid "smoothing" it due to posible filtering of previous positions.

In short: the gps speed value would be used directly (as is done with locations) and acceleration would be the third criterion for filtering locations (in addition to the minimum speed and distance already in use).

P.s. Graphical comparison of GPS speed and calculated speed:

The chart was created with GPXSee (www.gpxsee.org) using a track I recorded (walking) to test the changes introduced by #1010.

[image]
☝️Continuous line: GPS speed. Dashed line: calculated speed (distance/time).

dennisguse, 2021-11-17:
One question: why (4)?
What is the benefit of filtering by acceleration before storing a location?
I mean storing it is basically is almost cost-free and it could be removed in post-processing.
And why should it be better than the estimation acceleration?

@vlmendz  similar proposal to last time: want to create a PR and we can iterate on it?

Regarding unwanted locations: Great to hear!

vlmendz, 2021-11-18:
What is the benefit of filtering by acceleration...
The benefit is to get a more reliable track and statistics (especially the maximum speed). For example, if you are recording a track riding a bike and the acceleration calculated in the last position is more than 1g, it can be assumed that at least the speed of that position is wrong (and that the position in general is not very reliable).

You could ask the opposite question: what do you gain by saving a position with (at least) the wrong speed almost certainly?

    it could be removed in post-processing
post-processing where? outside of OpenTracks?

    And why should it be better than the estimation acceleration?
I don't understand the question🤷

    similar proposal to last time: want to create a PR and we can iterate on it?
Okay, I'll create it

dennisguse, 2021-11-20:
Another though: we would still store the GPS reported speed for each TrackPoint, but just compute the TrackStatistics differently.

rgmf, 2022-01-23:
About max. speed, it seems that it's getting worse in the last releases. In my last riding biking:

[image]

I rode downhill for more than 3km with an average of more than 40km/h so I don't understand how max. speed was 32,5 km/h.

[image]

Also, today's running I got this value for max. speed:

[image]

But I got intervals below 4:10.

@dennisguse can I work on this? Is it blocked by your current work?

@vlmendz do you know some QGIS tutorial/resource to learn how to use it? It could be useful to learn to fix some issues in OpenTracks.

dennisguse, 2022-01-23:
@rgmf go ahead.
I don't know about any recent changes in this code. 

PS I believe that our speed filtering is not optimal as it does not take time between measurements into account. Also it should smoothen better for outliers. Might even take GPS accuracy into account.

And for sensors: the data should be way more reliable and thus less processing needed.
But we can not distinguish the data source at the moment in TrackStatisticsUpdater.

vlmendz, 2022-01-24:
https://www.qgis.org/es/docs/index.html
@rgmf  ☝️

rgmf, 2022-01-25:
@dennisguse @vlmendz reading the TrackStatisticsUpdater, it seems to me that MAX ACCELERATION computation is wrong.

From code:
```java
Speed maxAcceleration = Speed.of(MAX_ACCELERATION * timeDifference.toMillis());
```

But I think, it should be:
```java
Speed maxAcceleration = Speed.of(MAX_ACCELERATION * timeDifference.getSeconds());
```

I mean: we have to do: `acceleration times seconds` instead of `acceleration times milliseconds`, right?

Because: `2g = 0.02m/s^2`
Then to get speed in `m/s (mps)` we have to do: **[acceleration in g]** times **[time in seconds]**

dennisguse, 2022-01-25:
MAX_ACCELERATION is (for some reason) in milliseconds....
But you are right: it looks weird - might be some refactoring mistake.
Best guess: MAX_ACCELERATION should be in seconds.

vlmendz, 2022-01-26:
This way of using acceleration is confusing to me. I would prefer to directly use the maximum acceleration value and compare it with the calculated acceleration:
MAX_ACCELERATION = 9.8 (1g or the value you decide to use)
acceleration = (speedDifference / timeDifference)
if acceleration.greaterThan(MAX_ACCELERATION) {...

vlmendz, 2022-01-26:
Another way:

max_speedDifference = MAX_ACCELERATION * timeDifference
if speedDifference.greaterThan(max_speedDifference) {...

Anyway, if acceleration is calculated using the last recorded point instead of last received point, this could mask spike in speed due to possibly greater timeDifference

dennisguse, 2022-06-11:
At least for altitude: the smoothing was broken (and now removed)  #1241.
For speed the same smoothing is applied, which might be broken as well.
so, we might get one step closer by also removing the SpeedRingBuffer.

@vlmendz @rgmf some opinion needed.

vlmendz, 2022-06-11:
@dennisguse I agree with removing the SpeedRingBuffer (if you reread the beginning of this issue you will see that it is the first point I wrote).

It would only remain optionally (points 2 and 3) to improve the filtering of peaks (in my opinion they are rare).

rgmf, 2022-06-11:
@dennisguse I'd remove SpeedRingBuffer. I think discarding impossible speed in `isValidSpeed` should be enough.

Also, I think `maxAcceleration` should be computed like this:

`Speed maxAcceleration = Speed.of(MAX_ACCELERATION * timeDifference.toSeconds());`

Read my old comment: https://github.com/OpenTracksApp/OpenTracks/issues/1003#issuecomment-1021482321

rgmf, 2022-06-11:
@dennisguse do you want me to do those changes and check them out next week?

- Removing ring buffer and
- Computing maxAcceleration as I'm pointing out.

dennisguse, 2022-06-11:
@rgmf go ahead if you have the time :sunglasses:

rgmf, 2022-06-14:
@dennisguse I didn't know, but max speed is not working well. For instance, here you have the first series of values of MAX. SPEED:

```
40,4
24,4
27,8
21,0
25,4
26,2
23,3
26,4
24,1
25,6
24,0
25,5
24,1
25,3
24,0
25,3
24,2
25,2
24,2
24,3
24,4
25,3
24,6
25,8
25,1
25,8
25,2
26,1
25,5
```

I didn't have any layout with MAX. SPEED and I've never seen this bad behavior of MAX. SPEED.

This is because `TrackStatisticsUpdater` is been resetting every once in a while and `maxSpeed` is set to 0 every once in a while and then `average.greaterThan(currentSegment.getMaxSpeed())` is true.

I'm going to try to fix it and then I'll remove ring buffer.

dennisguse, 2022-06-14:
Do we need this?
```
    private void reset(TrackPoint trackPoint) {
        if (currentSegment.isInitialized()) {
            trackStatistics.merge(currentSegment);
        }
        currentSegment.reset(trackPoint.getTime());
        currentSegment.setMaxSpeed(trackStatistics.getMaxSpeed());

        lastTrackPoint = null;
        speedBuffer.reset();
        resetAverageHeartRate();
    }
```

or this:
```
    private void updateSpeed(@NonNull TrackPoint trackPoint, @NonNull TrackPoint lastTrackPoint) {
        if (!trackPoint.isMoving()) {
            speedBuffer.reset();
        } else if (isValidSpeed(trackPoint, lastTrackPoint)) {
            speedBuffer.setNext(trackPoint.getSpeed());
            Speed average = speedBuffer.getAverage();
            if (average.greaterThan(trackStatistics.getMaxSpeed())) {
                currentSegment.setMaxSpeed(average);
            }
        } else {
            Log.d(TAG, "Invalid speed. speed: " + trackPoint.getSpeed() + " lastLocationSpeed: " + lastTrackPoint.getSpeed());
        }
    }
```

PS/ did not test any of the two solutions...
And I would be curious when we introduced this issue :sunglasses:

dennisguse, 2022-06-14:
Now that I think about it: the TrackStatistics is computed by segment and on fetching to the TrackStatisticsUpdater.TrackStatistics and TrackStatisticsUpdater.currentSegment are merged - so technically the higher maxSpeed should be used. Anyhow, we have no test coverage for maxSpeed in TrackStatisticsUpdaterTest.

dennisguse, 2022-06-14:
And checking the RingBuffer: `RingBuffer.getAverage()` requires at least one item to return an average - so the buffer might not be averaging as we expect it to.

This may explain the weird data that you see: 
First measured speed 40.4
 -> RingBuffer.setNext(40.4)
 -> RingBuffer.getAverage() => 40.4
Second measured speed 28.0
...
And we really need test coverage here....

rgmf, 2022-06-15:
> Now that I think about it: the TrackStatistics is computed by segment and on fetching to the TrackStatisticsUpdater.TrackStatistics and TrackStatisticsUpdater.currentSegment are merged - so [...]

Reading the code I can't understand what's happening, but "something" is changing the value of `currentSegment.maxSpeed`.

> And checking the RingBuffer: RingBuffer.getAverage() requires at least one item to return an average - so the buffer might not be averaging as we expect it to.

Anyway, `maxSpeed` never should get a lower value.

I keep going... ;)

rgmf, 2022-06-15:
@dennisguse I think I get it. Relevant code:

```java
    public Speed getAverageMovingSpeed() {
        return Speed.of(totalDistance, movingTime);
    }

    public Speed getMaxSpeed() {
        return Speed.max(maxSpeed, getAverageMovingSpeed());
    }
```

Like you can see, `getMaxSpeed()`  doesn't return `maxSpeed` directly. It cannot be like that.

dennisguse, 2022-06-15:
@rgmf this was implemented in 3f99263b5a57639f15631d0cfd90f6e051751ff7
(the before time)

rgmf, 2022-06-15:
> @rgmf this was implemented in [3f99263](https://github.com/OpenTracksApp/OpenTracks/commit/3f99263b5a57639f15631d0cfd90f6e051751ff7) (the before time)

It's not the problem :(

I think there are more than one thread accessing to TrackStatisticsUpdater 🤔 Is that possible? I'm thinking that because I cannot understand values I print for debugging porpose.

dennisguse, 2022-06-16:
Actually yes.

Everytime a new TrackPoint is stored AS WELL AS every second (to update the UI).

https://github.com/OpenTracksApp/OpenTracks/blob/1c3a2bd2e419dc9f5cf3446d2e8316e0b90a8996/src/main/java/de/dennisguse/opentracks/services/TrackRecordingManager.java#L114

rgmf, 2022-06-23:
@dennisguse I give up :(

I'm not able to catch the ¿bug? that make maxSpeed computation not working. TrackStatisticsUpdate.currentSegment.maxSpeed is resetting to 0 and something else is changing the value.

dennisguse, 2022-06-24:
@rgmf how often does it happen?

rgmf, 2022-06-24:
> @rgmf how often does it happen?

I have no idea because I've never set a layout with MAX SPEED. It's a data I don't care too much.

dennisguse, 2022-06-30:
Filtering of speed is now completely removed.
Let's see how this works out ;)

closed 2022-06-30 by dennisguse

dennisguse, 2022-07-01:
@vlmendz The removal of the speedBuffer is in v4.0.3
If you have any further suggestions, let us know.
PS/ sorry it took so long; too much discussion, too little doing (from my side).

vlmendz, 2022-07-01:
Results of a quick test of speedBuffer removal:

Maximum speed displayed in OT ui: 28.2 km/h

Maximum speed value exported in gpx file: 28.2 km/h

commit 9a116c9b (2022-04-16): TrackStatistics: compute max speed by traveled distance over time instead of GPS provided estimates.

Fixes #1003.
commit 65b04b09 (2022-04-16): TrackStatistics: compute max speed by traveled distance over time instead of GPS provided estimates.

Fixes #1003.