OpenTracksApp/OpenTracks #1006: NPE; first reported with v3.21.0

dennisguse, 2021-11-01:
Reported via Google Play; don't know how to reproduce.

```
java.lang.NullPointerException: 
  at de.dennisguse.opentracks.fragments.IntervalsFragment.updateIntervals (IntervalsFragment.java:190)
  at de.dennisguse.opentracks.fragments.IntervalsFragment.lambda$new$0$IntervalsFragment (IntervalsFragment.java:57)
  at de.dennisguse.opentracks.fragments.IntervalsFragment$$ExternalSyntheticLambda0.onSharedPreferenceChanged (Unknown Source:2)
  at de.dennisguse.opentracks.fragments.IntervalsFragment.onResume (IntervalsFragment.java:140)
  at androidx.fragment.app.Fragment.performResume (Fragment.java:3039)
  at androidx.fragment.app.FragmentStateManager.resume (FragmentStateManager.java:607)
  at androidx.fragment.app.FragmentStateManager.moveToExpectedState (FragmentStateManager.java:306)
  at androidx.fragment.app.FragmentManager.executeOpsTogether (FragmentManager.java:2189)
  at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute (FragmentManager.java:2100)
  at androidx.fragment.app.FragmentManager.execSingleAction (FragmentManager.java:1971)
  at androidx.fragment.app.BackStackRecord.commitNow (BackStackRecord.java:305)
  at androidx.viewpager2.adapter.FragmentStateAdapter$FragmentMaxLifecycleEnforcer.updateFragmentMaxLifecycle (FragmentStateAdapter.java:726)
  at androidx.viewpager2.adapter.FragmentStateAdapter$FragmentMaxLifecycleEnforcer$1.onPageScrollStateChanged (FragmentStateAdapter.java:632)
  at androidx.viewpager2.widget.CompositeOnPageChangeCallback.onPageScrollStateChanged (CompositeOnPageChangeCallback.java:87)
  at androidx.viewpager2.widget.CompositeOnPageChangeCallback.onPageScrollStateChanged (CompositeOnPageChangeCallback.java:87)
  at androidx.viewpager2.widget.ScrollEventAdapter.dispatchStateChanged (ScrollEventAdapter.java:426)
  at androidx.viewpager2.widget.ScrollEventAdapter.onScrolled (ScrollEventAdapter.java:214)
  at androidx.recyclerview.widget.RecyclerView.dispatchOnScrolled (RecyclerView.java:5173)
  at androidx.recyclerview.widget.RecyclerView$ViewFlinger.run (RecyclerView.java:5338)
  at android.view.Choreographer$CallbackRecord.run (Choreographer.java:999)
  at android.view.Choreographer.doCallbacks (Choreographer.java:797)
  at android.view.Choreographer.doFrame (Choreographer.java:728)
  at android.view.Choreographer$FrameDisplayEventReceiver.run (Choreographer.java:984)
  at android.os.Handler.handleCallback (Handler.java:883)
  at android.os.Handler.dispatchMessage (Handler.java:100)
  at android.os.Looper.loop (Looper.java:237)
  at android.app.ActivityThread.main (ActivityThread.java:8167)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:496)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1100)
```

dennisguse, 2021-11-01:
I don't get it. This should not be happening :man_shrugging:

rgmf, 2021-11-02:
I don't understand. The NPE cannot be there :thinking: 

When I fixed #996 I got errors in several parts of the code but the error was in IntervalStatistics.Interval.add

I'd bet this issue is the same one. We'll see when new release was launched.

dennisguse, 2021-11-02:
It occurred for v3.20.1.... the installation was not yet updated to v3.21.0

closed 2021-11-02 by dennisguse