OpenTracksApp/OpenTracks #1007: Recording of points without meeting the conditions of the distance and speed filters

vlmendz, 2021-11-02:
**Describe the bug**
These unnecessary points are recorded following a pattern:

Before each point that meets the conditions of the filters, the previous point received is recorded (see images):

[image]
[image]
☝️Points of a fragment of a recorded track (distance interval=5m)

[image]
☝️Distances between the points of a fragment of a recorded track (distance interval=5m)

Examining the TrackRecordingManager.onNewTrackPoint() function
I see a possible cause of this behavior:

	void onNewTrackPoint(TrackPoint trackPoint, Distance thresholdHorizontalAccuracy) {

        //TODO Figure out how to avoid loading the lastValidTrackPoint from the database

        TrackPoint lastValidTrackPoint = getLastValidTrackPointInCurrentSegment(trackId);

        //Storing trackPoint

        // Always insert the first segment location
        if (!currentSegmentHasTrackPoint()) {
            insertTrackPoint(trackId, trackPoint);
            lastTrackPoint = trackPoint;
            
            // Posible 2nd modification:   ******************************
            // Posible solution for TODO above?
            lastValidTrackPoint = trackPoint; 
            
            return;
        }

        Distance distanceToLastTrackLocation = trackPoint.distanceToPrevious(lastValidTrackPoint);
        if (distanceToLastTrackLocation != null) {
            if (distanceToLastTrackLocation.greaterThan(maxRecordingDistance)) {
                
                // Remove also this?  ******************************              
                insertTrackPointIfNewer(trackId, lastTrackPoint);

                trackPoint.setType(TrackPoint.Type.SEGMENT_START_AUTOMATIC);
                insertTrackPoint(trackId, trackPoint);

                lastTrackPoint = trackPoint;
                
                // Posible 2nd modification:   ******************************
                // Posible solution for TODO above?
                lastValidTrackPoint = trackPoint; 
                
                return;
            }
            
            
            if (trackPoint.hasSensorData() || (draistanceToLastTrackLocation.greaterOrEqualThan(recordingDistanceInterval) && trackPoint.isMoving())) {
                
                //************************************
                // 1st modification:   ******************************
                // Remove next line. Next function (combined with the last line of onNewTrackPoint) appears to be the cause of the unwanted recording:
                //insertTrackPointIfNewer(trackId, lastTrackPoint);
                //************************************

                insertTrackPoint(trackId, trackPoint);

                lastTrackPoint = trackPoint;
                
                // Posible 2nd modification:   ******************************
                // Posible solution for TODO above?
                lastValidTrackPoint = trackPoint; 
                
                return;
            }
        }

      
        Log.d(TAG, "Not recording TrackPoint, idle");
        lastTrackPoint = trackPoint;
    }
**To Reproduce**
Record a track at low speed.

**Technical information**
 - Device: Redmi Note 7
 - OS: MIUI 12.5.1 (Android 10 based)
 - Version OpenTracks 3.21.0

dennisguse, 2021-11-02:
The pasted code is not the most recent version.

@vlmendz In any case: creating a PR makes our discussions way easier :)

vlmendz, 2021-11-03:
The pasted code is not the most recent version.

I see 🤦

    @vlmendz In any case: creating a PR makes our discussions way easier :)

How I create it? (sorry for my GitHub shortcomings)

commit c0b14750 (2021-11-08): Proposed solution for recording points that do not meet filter conditions (#1010)

Only store a TrackPoint if traveled distance is greater than recording distance interval.
Code seemed to be a leftover of the removed idle/non-idle code.

Fixes #1007.
commit 08ba7d50 (2021-11-08): Only record a TrackPoint if the traveled distance is greater than recording distance interval.
Code seemed to be a leftover from the removed idle/non-idle functionality.

Fixed #1007.
commit 16ee8a97 (2021-12-28): Only record a TrackPoint if the traveled distance is greater than recording distance interval.
Code seemed to be a leftover from the removed idle/non-idle functionality.

Fixed #1007.
closed 2021-11-08 by dennisguse