OpenTracksApp/OpenTracks #1020: PreferencesUtils.getSharedPreferences

vsevjednom-cz, 2021-11-15:
Where is `PreferencesUtils.getSharedPreferences` in `v3.22.0`? I can't find it.
Class `PreferencesUtils` is in settings right now but `public static SharedPreferences getSharedPreferences(Context context) {`is missing. I need set and get some variables.

And you probably forgot replace sound annoucement: https://github.com/OpenTracksApp/OpenTracks/issues/862#issuecomment-917682202_

dennisguse, 2021-11-15:
It getSharedPreferences got deleted and replaced by a global static variable: https://github.com/OpenTracksApp/OpenTracks/issues/961
See for the actual changes: https://github.com/OpenTracksApp/OpenTracks/commit/eac3d4b3dbfa2cc565301237539d106504e95965

Regarding: https://github.com/OpenTracksApp/OpenTracks/issues/862#issuecomment-917682202
Can you create a PR?

vsevjednom-cz, 2021-11-15:
Hmm, solved, I hope.

Original code:
```
sharedPreferences = PreferencesUtils.getSharedPreferences(this);
SharedPreferences sharedPref = getSharedPreferences(getString(R.string.Prefs), Context.MODE_PRIVATE);
String nick = sharedPref.getString(getString(R.string.something), "");
```

New code:
```
SharedPreferences sharedPref = getSharedPreferences(getString(R.string.Prefs), Context.MODE_PRIVATE);
String nick = sharedPref.getString(getString(R.string.something), "");
```

And:
From this:
```
sharedPreferences = PreferencesUtils.getSharedPreferences(getApplicationContext());
DocumentFile dir = PreferencesUtils.getDefaultExportDirectoryUri(sharedPreferences, getApplicationContext());
```

To this:
`DocumentFile dir = PreferencesUtils.getDefaultExportDirectoryUri(getApplicationContext());`

> Regarding: [#862 (comment)](https://github.com/OpenTracksApp/OpenTracks/issues/862#issuecomment-917682202)
> Can you create a PR?

Of course. This evening.

Thanks!

closed 2021-11-15 by vsevjednom-cz

vsevjednom-cz, 2021-11-15:
I'm trying to create PR but create button is still inactive. I don't understand branches 👽.

dennisguse, 2021-11-15:
@vsevjednom-cz You can do this, but this is better:
```
PreferencesUtils.getString() 
```

or use this (if you really need the sharedPreferences object).
```
PreferencesUtils.sharedPreferences
```

Then sharedPreferences are only instantiated once and we always yse them in the same way.

dennisguse, 2021-11-15:
@vsevjednom-cz just create a branch locally + commit, push it to your fork, and then open the PR via Github.com

vsevjednom-cz, 2021-11-15:
> `PreferencesUtils.getString()`

But what is `int keyId`? My preference names are strings, for example: `String nick = sharedPref.getString(getString(R.string.nick), "");`

> `PreferencesUtils.sharedPreferences`

`'sharedPreferences' has private access in 'de.dennisguse.opentracks.settings.PreferencesUtils'`

My solution above works. So I don't need to change it I hope.

> Then sharedPreferences are only instantiated once and we always yse them in the same way.
Looks good 🤩.

dennisguse, 2021-11-15:
`keyId` would be in this case `R.string.nick`; it looks like you also extracted the actual preference names into one xml file.
This has the advantage that one can later rename the variables in the code without needing to modify the actual preferences stored on the phones.

PS/ did you have any trouble so far with integrating your changes?