OpenTracksApp/OpenTracks #1025: Export is sometimes (very rarely) not working

mega-stoffel, 2021-11-18:
Hej there!
My tour from yesterday, November 17th, is somehow not able to export.

**Describe the bug**
I usually commute in the morning, stop this track (in the underground parking without any connection). This works fine. In the afternoon I just select this track and click on "continue track" and ride home. This works fine, almost every time. Also all my other recordings work fine.
Just a few weeks ago, there was one track, which the app couldn't export. But it showed me the attached error (this was on October 20th) when I stopped the recording back home. And this happened again yesterday. Today everything was fine again, with the same setup.
[image]

**To Reproduce**
I can't really say, how to reproduce it. And I even tried to continue this "broken" track to create somehow a new file. But this didn't help.
The track itself is not broken. I attached it to an eMail and sent it to myself. This file was ok and I could upload it to strava.
The automatically exported track includes only the morning commute, but it doesn't include the ride home (because of the error).

I believe this is a hard one and I don't really expect a solution. But I wanted to let you know and offer my help regarding this issue - in case it could be of any use (without programming skills or an android infrastructure around).

**Technical information**
 - Device: Fairphone 3+
 - OS: Android 10, security patch of November 3rd.
 - Version: 3.22.0 4519

dennisguse, 2021-11-18:
Any chance to get a logcat?

The error messages states that the folder could not be accessed (not sure if we just catch all errors and just print this message).
When you set in the settings, we acquired write permissions.
My first thought was that this permission was revoked as it was granted long ago.

rgmf, 2021-11-20:
@mega-stoffel It happened to me with a track. I cannot be exported to a nextcloud folder but it does in others folders. Is that your case?

I know where the issue is but I don't know if we can do anything, because it's in `DocumentFile.createFile` or in `DocumentFile.getUri`.

mega-stoffel, 2021-11-20:
it might be related to nextcloud, yes. That's where my files end up.
And @dennisguse: I doubt it is a general issue, because usually it's working. Just today I successfully recorded a new track (as on most of the days since I use OpenTracks). How can I get such a "logcat"?

dennisguse, 2021-11-20:
@mega-stoffel This can be done via AndroidStudio: https://developer.android.com/studio/debug/am-logcat
Requires enabling developer options on your phone.
And the logs are stored for some time - so the phone does not need to be connected to the computer when it happens.
You filter by "OpenTracks" in the logs.

mega-stoffel, 2021-11-23:
I think I have some logcat output via the LogcatReader app (https://github.com/darshanparajuli/LogcatReader).
Is it ok to share it here? Or should I send it somewhere else?

pstorch, 2021-11-23:
This is the error I get when exporting tracks to a Nextcloud directory:

```
2021-11-23 07:09:16.602 16484-17097/? W/DocumentsContract: Failed to create document
    android.os.ParcelableException: java.io.FileNotFoundException: File could not be created
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:82)
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:79)
        at android.os.Parcel.readParcelable(Parcel.java:3293)
        at android.os.Parcel.createExceptionOrNull(Parcel.java:2380)
        at android.os.Parcel.createException(Parcel.java:2369)
        at android.os.Parcel.readException(Parcel.java:2352)
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:190)
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:142)
        at android.content.ContentProviderProxy.call(ContentProviderNative.java:732)
        at android.content.ContentResolver.call(ContentResolver.java:2415)
        at android.provider.DocumentsContract.createDocument(DocumentsContract.java:1377)
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:53)
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:45)
        at de.dennisguse.opentracks.util.ExportUtils.getExportDocumentFileUri(ExportUtils.java:94)
        at de.dennisguse.opentracks.util.ExportUtils.exportTrack(ExportUtils.java:43)
        at de.dennisguse.opentracks.io.file.exporter.ExportService.onHandleWork(ExportService.java:51)
        at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:396)
        at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:387)
        at android.os.AsyncTask$3.call(AsyncTask.java:394)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:923)
     Caused by: java.io.FileNotFoundException: File could not be created
        at java.lang.reflect.Constructor.newInstance0(Native Method)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:343)
        at android.os.ParcelableException.readFromParcel(ParcelableException.java:56)
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:82) 
        at android.os.ParcelableException$1.createFromParcel(ParcelableException.java:79) 
        at android.os.Parcel.readParcelable(Parcel.java:3293) 
        at android.os.Parcel.createExceptionOrNull(Parcel.java:2380) 
        at android.os.Parcel.createException(Parcel.java:2369) 
        at android.os.Parcel.readException(Parcel.java:2352) 
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:190) 
        at android.database.DatabaseUtils.readExceptionFromParcel(DatabaseUtils.java:142) 
        at android.content.ContentProviderProxy.call(ContentProviderNative.java:732) 
        at android.content.ContentResolver.call(ContentResolver.java:2415) 
        at android.provider.DocumentsContract.createDocument(DocumentsContract.java:1377) 
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:53) 
        at androidx.documentfile.provider.TreeDocumentFile.createFile(TreeDocumentFile.java:45) 
        at de.dennisguse.opentracks.util.ExportUtils.getExportDocumentFileUri(ExportUtils.java:94) 
        at de.dennisguse.opentracks.util.ExportUtils.exportTrack(ExportUtils.java:43) 
        at de.dennisguse.opentracks.io.file.exporter.ExportService.onHandleWork(ExportService.java:51) 
        at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:396) 
        at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:387) 
        at android.os.AsyncTask$3.call(AsyncTask.java:394) 
        at java.util.concurrent.FutureTask.run(FutureTask.java:266) 
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) 
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) 
        at java.lang.Thread.run(Thread.java:923) 
2021-11-23 07:09:16.603 542-548/? E/statsd: Predicate -7037417284711607308 dropping data for dimension key (10)0x2010101->10540[I] (10)0x30000->*job*/de.dennisguse.opentracks.nightly/de.dennisguse.opentracks.io.file.exporter.ExportService[S] 

```

dennisguse, 2021-11-24:
@pstorch Anything suggestion what to do in this case?

@mega-stoffel Does it look the same as the one from @pstorch?
You can just search for `Caused by: java.io.FileNotFoundException: File could not be created`

If it is different, you can either share the relevant part here __or__ send it by mail to me: opentracks-play@dennisguse.de

PS/ a logcat may contain sensitive information, so sharing it with others may be incur some risks.

closed 2021-11-29 by pstorch

pstorch, 2021-11-29:
It's closed by my commit message, but I'm not 100% sure if it is solved completely. Should we keep it open for a while?

commit 5e3063e5 (2021-11-26): fix #1025 exportdir failed to create document

fix #254 illegal filename characters
commit bb7efaa8 (2021-11-29): fix #1025 exportdir failed to create document

fix #254 illegal filename characters