OpenTracksApp/OpenTracks #1029: Sensor data: compute and store average for each trackpoint

dennisguse, 2021-11-19:
**Is your feature request related to a problem? Please describe.**
We are storing the latest reported data from a sensor, whenever a new TrackPoint is created (i.e., a new GPS location reported) [except for distance, which is summed].
Storing data this way is not entirely correct.

**Describe the solution you'd like**
Compute and store the average.
For example, compute the average of reported cadence sensor since the previous TrackPoint was stored.
And we need to take time between sensor data into account:
* t0 previous trackpoint stored
* t1 cadence=10
* t3 cadence=20
* t4 cadence=30
* t5 next trackpoint store with avg cadence of 70/4.

Affects:
* cadence 
* heartbeat

hercek, 2023-12-31:
Cycling power should be computed as weighted average as well. That is for the case when more BLE power points gets merged into one TrackPoint.

dennisguse, 2023-12-31:
@hercek this becomes possible now. 
As of version 4.9.6, the sensor processing got refactored. 
Now for each data stream there is one aggregator. 
Currently, the only "fancy" ones are for cycling cadence as well as cycling speed. 
The others are still "most recent" data point. 

And we have averaging for the interval computation for voice announcements. 
Maybe this can be used. 

PS this refactoring is also the reason for the current state of bugginess...

hercek, 2023-12-31:
If I noticed it correctly the averaging for voice announcements is for multiple TrackPoints which get accumulated into one Interval. While Agregators should handle multiple (BLE) measurements which get accumulated into one TrackPoint.

So it is similar but not the same. When the (weighted) average gets implemented even in all the Agregators then it will be as good as it could possibly be :)

Anyway I quite like OpenTracks even as they are now. Nice job!

hercek, 2024-01-02:
I do not see where Aggregator is reset when its data are pushed to a TrackPoint. There typically should be a few BLE data points in one TrackPoint. These BLE data points should be averaged over the TrackPoint time span.

TrackPointCreator will push SensorManager data to a TrackPoint and then resets SensorDataSet which resets the Aggregators. But aggregators reset does not do anything. This does not matter for power data since they are instantaneous. Actually cadence data seem to be instantaneous as well (compute value just overwrites the last value instead of averaging with pervious values). This hypothetical averaging of course also needs reset to work.

Heat rate does not seem to be averaged in aggregators yet as well.

Looks like there is no averaging at the aggregator level. But there is still the averaging at the Interval level. I'm not sure interval level averaging works correctly. The code looks fishy to me. But it likely works well enough since my tests on my bicycle indicate so. I have a conflict between what I see in the code and the reality. That is called romanticism :)

dennisguse, 2024-01-02:
You are right, there is no averaging.
The reset() call is only relevant for Cycling Cadence as well as Cycling Speed/Distance.
However, averaging is possible to implement in the Aggregators.

PS/ The intervals are purely done on the stored TrackPoints and since those do not contain all the data, it might be slightly off.