OpenTracksApp/OpenTracks #1038: [Feature Request] AMOLED Dark Theme (pure black)

ae5960e8-a6fc-491f-b252-898ecf59af95, 2021-11-28:
Please consider adding an option for a pure black/AMOLED black UI theme to the app. Thank you!

dennisguse, 2021-11-28:
Duplicate of  #1038.

Sadly, I didn't have the time to look into this, but it shouldn't be too difficult.
The F-Droid app has support for this, and we might be able to use the same approach.
@ae5960e8-a6fc-491f-b252-898ecf59af95 If you have time for this kind of research, it would be great.
Adding it to OpenTracks should be relatively simple (I hope)...

closed 2021-11-28 by dennisguse

ae5960e8-a6fc-491f-b252-898ecf59af95, 2021-11-28:
I'm not experienced in Android development, but recently another app implemented a pure black/AMOLED dark theme. Here's how Currencies (FOSS app) did it:
https://github.com/sal0max/currencies/commit/9b8cb791b5550d17edbf8cc7d8053eff7ebc63be