OpenTracksApp/OpenTracks #1039: Feature request: Export all/selected tracks to CSV

pil-in-a, 2021-12-03:
Hello.

It would be great to have possibility to export all/selected tracks to simple CSV, which will contain basic info of the track (date, distance, avg speed, description, etc.). Every track as a line.

As a biker, sometimes I need some quick info, how many kilometers I've ridden since some exact date (e.g. for maintenace - how many km I have with new chain etc,). 
Or maybe do some other simple analytics which I am struggling to make from several GPX files. I haven't found any suitable SW to do so.

Other alternative could be to show Aggregated stats from selected tracks (maybe add some more stats).

dennisguse, 2021-12-03:
CSV: related to #999.
Kind of easily doable; as usual the UI/UX is a problem.
@pil-in-a Any suggestions?

@rgmf Or could we extend the aggregated stats page for this?

pil-in-a, 2021-12-03:
For aggregated stats, the icon could stay in top bar after selection but it will lead to agg. stats for selected tracks. 

For export CSV, there could be "Export stats CSV" in hamburger menu after selection. It will generate CSV (named by date and time?) and export it to default export folder.
The other option could be a button "Export stats CSV" on the aggregated stats page, which will export that.

rgmf, 2021-12-03:
@dennisguse @pil-in-a Aggregated Stats could have a between dates filter. That way, it would be more general and I think it covers what @pil-in-a needs, isn't it?

About CSV export option: I'd integrate it with GPX, KML... options. Otherwise, it'd be a little confusing. We could do what @pil-in-a point out about the burger menu but for all kind of formats.

pil-in-a, 2021-12-03:
@rgmf I think aggregated stats from selection would be more general in my opinion because all tracks between dates suits only in one specific case (for my purpose good enough). But I can imagine, someone would like to pick up only some training route performed once in a while and leave all the commuting between. Although the exported CSV will handle this task very easily.

dennisguse, 2021-12-04:
This issue is also related: https://github.com/OpenTracksApp/OpenTracks/issues/395

PR #1040 (2021-12-12): Aggregated stats for only selected tracks.
**Describe the pull request**
Aggregated stats has been extended to see aggregated stats only for tracks selected in the contextual menu from TrackListActivity.

Users that select tracks (open contextual menu) can see now a new item menu called "Aggregated Stats".

AggregatedStatisticsActivity expect an optional list of Track.Id and show only aggregated stats from those Track.Ids. If this list is empty or not exists then it shows aggregated stats from all tracks.

It partially fixes #1039 (see comment: https://github.com/OpenTracksApp/OpenTracks/issue [... cut]
commit ba43df43 (2021-12-20): AggregatedStatisticsActivity: added filter option (by category, start and/or end date).
Fixes #1039.
commit 92f44ca7 (2021-12-20): AggregatedStatisticsActivity: added filter option (by category, start and/or end date).
Fixes #1039.
commit ad48e3a2 (2021-12-20): AggregatedStatisticsActivity: added filter option (by category, start and/or end date).
Fixes #1039.
closed 2021-12-24 by rgmf