OpenTracksApp/OpenTracks #1046: Moving time is no working.

rgmf, 2021-12-12:
**Describe the bug**
Using nightly version I've figured out that moving time is not working. When you stopped the moving time stopped too, but then moving time jump forward all the time you was stopped.

For example:
- Start an activity and stopped after 30 minutes.
- Take a 5 minutes rest (moving time remains in 30 minutes).
- Resume the activity and then moving time jump to 35 minutes instead of continue from 30 minutes.

**To Reproduce**
Follow the example described above.

dennisguse, 2021-12-12:
Damn; sounds like a regression due to our recent changes in `TrackRecordingManager.onNewTrackPoint`: i.e., the removal of idle/non-idle code.

@rgmf Could you check how the actual recorded TrackPoints look like (mainly there TYPE)?
Best guess: they will be of TYPE=TRACKPOINT and then the TrackStatisticsUpdater has trouble to determine if it is just a lot of time between TrackPoints or if you were actually not moving.

Problematic code should be in `TrackStatisticsUpdater`.

dennisguse, 2021-12-12:
Finally in `TrackStatisticsUpdater`: 
```
        if (lastTrackPoint.isMoving()) {
            currentSegment.addMovingTime(movingTime);
        }
```

It is a regression: 
1. Record TrackPoints while moving: type TRACKPOINT with speed > moving threshold
2. Then stop moving (almost immediately): nothing will be recorded (here is the issue: we do not record a non-moving TrackPoint here anymore; but we don't anymore).
2. Wait
3. Then start moving again will create new TrackPoints:  type TRACKPOINT with speed > moving threshold

PS: we need test coverage for this in an integration test for TrackRecordingService.

dennisguse, 2021-12-12:
Change was released with v3.22.0

rgmf, 2021-12-12:
> @rgmf Could you check how the actual recorded TrackPoints look like (mainly there TYPE)? Best guess: they will be of TYPE=TRACKPOINT and then the TrackStatisticsUpdater has trouble to determine if it is just a lot of time between TrackPoints or if you were actually not moving.

Yes, you are right. I see three TrackPoints of type -2 (SEGMENT_START_MANUAL) and three more of type 1 (SEGMENT_END_MANUAL) because I paused and then resumed the track exactly three times. The rest of the TrackPoints are type 0 (TRACKPOINT).

rgmf, 2021-12-12:
>     1. Record TrackPoints while moving: type TRACKPOINT with speed > moving threshold
> 
>     2. Then stop moving (almost immediately): nothing will be recorded (here is the issue: we do not record a non-moving TrackPoint here anymore; but we don't anymore).
> 
>     3. Wait
> 
>     4. Then start moving again will create new TrackPoints:  type TRACKPOINT with speed > moving threshold

That's right. The issue is in the second point. There have should be a segment there, right?

dennisguse, 2021-12-12:
Not really, OpenTracks creates segments only on distance to previous location.
For moving time, we just use `TrackPoint.isMoving()`, but we only store TrackPoints __if__ they were considered moving.
From `TrackRecordingManager`: 
```
            if (distanceToLastTrackLocation.greaterOrEqualThan(recordingDistanceInterval) && trackPoint.isMoving()) {
                insertTrackPoint(trackId, trackPoint);
                return true;
            }
```

Basically: moving time is broken.
It might only be working if an automatic segment was created (due to segment distance).

commit 407e56b5 (2021-12-13): Revert "Only record a TrackPoint if the traveled distance is greater than recording distance interval."

This reverts commit 08ba7d50c73197a1230eca8cc72dc1d501d7a6af.

Fix #1046.
commit 4c7d9384 (2021-12-19): Revert "Only record a TrackPoint if the traveled distance is greater than recording distance interval."

This reverts commit 08ba7d50c73197a1230eca8cc72dc1d501d7a6af.

Fix #1046.
commit 3be1348d (2021-12-21): Merge pull request #1047 from OpenTracksApp/movingTime#1046

Restore moving time implementation via idle/non-idle TrackPoints.
Fixes #1046.
commit 6fe6cee4 (2021-12-21): BLE: CSC remove workarounds as those are not necessary.

The implementation of OpenTracks for the CSC parsing was broken and fixed in 35de5b6b264ad43b1c343a8323961db18b4874f4

Fixes #1046.
closed 2021-12-21 by dennisguse