OpenTracksApp/OpenTracks #1050: Wahoo blue SC sensor

JaniKe74, 2021-12-16:
This sensor has speed/distance sensor and also cadence sensor. The problem is that information from cadence is coming to speed/distance fields and speed info is coming to cadence field. Cadense field is still showing value in rpm, but it's following tyre rotation. If I stop rotating cranks, speed goes to zero.

dennisguse, 2021-12-16:
Interesting!
Link to the sensor: https://de-eu.wahoofitness.com/devices/bike-sensors/wahoo-blue-sc-speed-and-cadence-sensor

I had some trouble with cadence as well as speed sensors that provided the wrong kind of data aka a cadence sensor providing speed and vice versa.
So, for check first for one and if it is not there check the other.

Relevant code: 

https://github.com/OpenTracksApp/OpenTracks/blob/b287967bc4f40c8f760b3dcdc0ac87f95e8df174/src/main/java/de/dennisguse/opentracks/services/sensors/BluetoothConnectionManager.java#L206

 https://github.com/OpenTracksApp/OpenTracks/blob/b287967bc4f40c8f760b3dcdc0ac87f95e8df174/src/main/java/de/dennisguse/opentracks/services/sensors/BluetoothConnectionManager.java#L230

JaniKe74, 2021-12-16:
I have tested also only one sensor at a time for both sensors and it's same situation, data goes to wrong field. If i use this same Wahoo sensor in Sports Tracker app it is working correctly, so the sensor should work correctly.

dennisguse, 2021-12-16:
That sounds as if we parse the data in the wrong order and the implement workarounds just "fix" this.

Here is the spec: https://www.bluetooth.com/wp-content/uploads/Sitecore-Media-Library/Gatt/Xml/Characteristics/org.bluetooth.characteristic.csc_measurement.xml

The actual byte parsing is happening in BluetoothUtils.

@JaniKe74 It would be awesome if you could debug this.
The actual received payload from sensor is the most important thing to get.

JaniKe74, 2021-12-16:
Well, I don't know what I need to do and how to do..

dennisguse, 2021-12-18:
Depends on your Tech Skills :)
Option 1: can you compile and install OpenTracks (via AndroidStudio) on your phone?
If yes, then you can just put a break point and wait until you receive data.

Option 2: you can use a Android BLE application and try connect to the sensor manually.
(i.e., connect to the device and subscribe to the correct service).
This one should be able to do it: 
https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp

Option 3: use a Linux machine and use bluetoothctl to connect to device and 

Option 4: I need a device for about  30min :)

JaniKe74, 2021-12-19:
I download that app what was in option 2. Then I connect to the Wahoo and took screenshot and dowloaded log file. I don't know have those information what you need, but you can give me more info what to do. [image][image][image]

dennisguse, 2021-12-19:
@JaniKe74 Nice!
Just checked my sensor with the nRF connect app and it can export raw values!
For this, you were already in the right menu: just get a measurement from the 0x2A5B  service.
1. start your sensor
2. open the nRF connect app and connect to the sensor
3. go to 0x2A5B (CSC measurement)
4. either enable notification / or get one measurement only
  When it received data, it will show the wheel revolution count (actually decoded) etc.
5. long-press the 0x2A5B section and select share

My sensor data (Garmin Speed sensor):
```
CSC Measurement
0x2A5B
Properties: NOTIFY
Parsed value: Wheel rev: 1209336,
Last wheel event time: 51125 ms
Raw value: (0x) 01-F8-73-12-00-B5-C7 
```
The 01 indicates that mine only measures wheel.
Yours should have a 03 in the beginning as it measures crank and wheel.

PS: those sensors only count revolutions: computation of the rest is happening on app level.

dennisguse, 2021-12-19:
```
CSC Measurement
0x2A5B
Properties: NOTIFY
Parsed value: Wheel rev: 0,
Last wheel event time: 272 ms,
Crank rev: 33,
Last crank event time: 60358 ms
Raw value: (0x) 03-00-00-00-00-10-01-21-00-C6-EB
```

From https://github.com/OpenTracksApp/OpenTracks/issues/1054#issue-1084190726

JaniKe74, 2021-12-19:
Okay, I send the data already, but it is a new issue. I didn't know how to put it in this same issue.

dennisguse, 2021-12-19:
I just checked the parsing and the data handling: looks okay.
At least I couldn't find anything wrong ;)

@JaniKe74 Can you try this APK?
I removed the swapping of wheel vs crank workaround: if we were looking for wheel and t was not present, we checked cadence as well and vice versa.
If there is a problem somewhere in OpenTracks you will not see any data using this variant.
Current hypothesis: there is something wrong where we connect to the sensor (BluetoothRemoteSensorManager).

[OpenTracksUpstream-releasePlayStore.zip](https://github.com/OpenTracksApp/OpenTracks/files/7741659/OpenTracksUpstream-releasePlayStore.zip)

JaniKe74, 2021-12-19:
Yeah, now it was working correctly. I tested one sensor at a time and both sensors together and it was working in all the situations.

dennisguse, 2021-12-20:
@JaniKe74 I still don't know why this works, but YOLO.
I checked with my Garmin Speed 2 and this works as well (without the workaround).
During the week I (hopefully) will get my hands on two Wahoo sensors (again) for testing.
If everything works, then we can release this change.

It might just be that there was a bug in another place that was fixed by this workaround, but the bug got fixed ;)

PS/ I sent you the GooglePlay variant (correctly signed); you can stick with it for now.
If you got OpenTracks via PlayStore, it will automatically be updated; otherwise you must manually switch the F-Droid version when it get's released.

PS/ nice debugging work by the way :D

dennisguse, 2021-12-21:
@JaniKe74 Did you actually ever use your sensor with a pre-v3.22.0 version of OpenTracks?

JaniKe74, 2021-12-21:
No, only that last one v3.22.0.

dennisguse, 2021-12-21:
@JaniKe74 then I don't get it :)
I introduced this issue likely in this commit: 35de5b6b264ad43b1c343a8323961db18b4874f4
I saw that the parsing of the data was (by me) implemented the wrong way around, but that change was not released with v3.22.0 as it happened afterwards.

But I tried to test using a Garmin Speed and the current main branch is not working anymore (likely due to this commit), but it works with the fix I send you.
Are you building OpenTracks yourself? Or are you using the nightly build?

JaniKe74, 2021-12-21:
I downloaded Open Tracks from F-Droid maybe year and half ago, but there was this problem with sensor. Later I tested it again, because it get update but problem was still there and then I make this issue. When you send that apk couple days ago, it was first time when sensor worked correctly. I'm not a coder, just user. But I like open source software and i'm using linux and so on, but i don't have skills to make code.

dennisguse, 2021-12-21:
@JaniKe74 Thanks for reporting this! Don't hesitate to tell us if something is broken or can be improved.
We will figure then out (somehow), how to get it done :)

PS/ I just released v3.23.0, but it does not contain this fix as I want to be sure that the unneeded workaround was the actual issue. My main motivation is understand if I understood how it works :)
Just keep using the one I sent you; I should have time to test this problem before the end of the year!

JaniKe74, 2021-12-21:
Okay, good to know about that 3.23.0 version. It was interesting being  part of this and thanks for fixing it :).

dennisguse, 2021-12-22:
@JaniKe74 Changes are merged; tomorrows nightly build run will have this change: https://opentracksapp.com/#Nightlies
For F-Droid: we are waiting for some days after v3.23.0 becomes available on F-Droid before creating a new release.
So, it will be after X-Mas.

closed 2021-12-22 by dennisguse

dennisguse, 2022-01-09:
This fix is now released with v3.24.0 - took quite a while, but happy cycling :D

PR #1056 (2021-12-22): Wahoo#1050
Fixes #1050 .

Will be merged ~before~ after v3.23.0.