OpenTracksApp/OpenTracks #1058: Fix movingtime_with_pauses on API23

dennisguse, 2021-12-22:
Somehow, this integration test is failing on API23 (especially CI).
Did not figure out exactly why: the mocking of the Clock seems to be not be working properly.

@dennisguse here they are:

At the beginning of the test:

```
Start Time: 1970-01-01T00:00:00Z
Stop Time: 1970-01-01T00:00:00Z
```

Just before  `expected: <PT19M> but was:<PT18M>` fail:

```
Start Time: 1970-01-01T00:00:00Z
Stop Time: 1970-01-01T00:19:00Z
```

Only FYI: this is the code I wrote:

```java
170. Log.e("probando", "Start Time: " + contentProviderUtils.getTrack(trackId).getTrackStatistics().getStartTime());
171. Log.e("probando", "Stop Time: " + contentProviderUtils.getTrack(trackId).getTrackStatistics().getStopTime());
172. assertTotalTime.apply(19 * 60);
```

_Originally posted by @rgmf in https://github.com/OpenTracksApp/OpenTracks/issues/1047#issuecomment-997379271_

dennisguse, 2021-12-27:
Took me quite a while.

The problem is:
1. on API23 a new segment is created at second 17
2. because in a previous test the max recording distance interval is reduced from 200m to 50m (for testing)
3. and then this test fails.

Actual problem: ServiceTestRule is not restarting (i.e., onDestroy() + onCreate()) the service on API23.
Thus, the preferences in TrackRecordingManager are used in the subsequent tests.
Same issue with the clock.

in addition, the TrackRecordingService.TrackRecordingManager was not set to null in onDestroy, so it might have been reused.

All changes added to #1033.

Only problem: I have the feeling the lifecycle of TrackRecordingService must be refactored (like only create managers if needed and set them to null afterwards again). Also do we really need to start with STICKY?

closed 2021-12-28 by dennisguse

rgmf, 2021-12-28:
@dennisguse awesome catch :tada: :tada: :tada:

dennisguse, 2021-12-28:
But to be honest...... I really need to rework the status handling in TrackRecordingService.
Things must be created when needed and not in onCreate.
Took me some hours :)

commit 98aff4cc (2021-12-27): Tests: ServiceTestRule does not (always) restart the service.
Thus, resetting it's status is not happening and then leading to failures in subsequent tests.

Fixes #1058.
commit 3f049f42 (2021-12-28): Tests: ServiceTestRule does not (always) restart the service.
Thus, resetting it's status is not happening and then leading to failures in subsequent tests.

Fixes #1058.
commit 7d0d87e9 (2021-12-28): Tests: ServiceTestRule does not (always) restart the service.
Thus, resetting it's status is not happening and then leading to failures in subsequent tests.

Fixes #1058.