OpenTracksApp/OpenTracks #1059: Optimize idle/non-idle detection

dennisguse, 2021-12-22:
We detect idle/non-idle changes only using the GPS/ sensor-provided speed.
As the speed is inherently noisy (especially for GPS), this leads to creating to much TrackPoints due to idle/non-idle changes.

Some ideas:
* filter the speed (using the speedbuffer from TrackStatistics)
* require a certain time before switching between idle/non-idle
* require a certain amount of GPS locations switching between idle/non-idle

Background infos: 
We removed idle/non-idle TrackPoints, but needed to re-implement them as otherwise the moving time cannot be computed.
Reverting the changes was done here: https://github.com/OpenTracksApp/OpenTracks/pull/1047

Version without idle/non-idle detection: v3.22.0

dennisguse, 2021-12-22:
@vlmendz Any other ideas?

vlmendz, 2021-12-23:
@dennisguse As a starting point to experiment, I recommend you create a branch using version 3.22 as a base, and in this branch replace TrackRecordingManager.onNewTrackPoint() with the version I put below, compile and test outdoors.

If this modification works as I expect, it should record only the first and last position received at each stop (when moving, it should work as the 3.22 does). As I have understood from your comments, this is the necessary condition for the current statistics logic of OpenTracks to calculate the time in motion.

    boolean onNewTrackPoint(TrackPoint trackPoint) {
        //Storing trackPoint

        // Always insert the first segment location
        if (!currentSegmentHasTrackPoint()) {
            insertTrackPoint(trackId, trackPoint);
            currentSegmentHasTrackPoint = true;
            PreviousPoint_IsMovig = trackPoint.isMoving();
            if PreviousPoint_IsMovig {
              PreviousIdlePoint = null;
            }
            else {
              PreviousIdlePoint = trackPoint;
            }
            return true;
        }

        Distance distanceToLastTrackLocation = trackPoint.distanceToPrevious(lastStoredTrackPoint);
        if (distanceToLastTrackLocation != null) {
            if (distanceToLastTrackLocation.greaterThan(maxRecordingDistance)) {
                trackPoint.setType(TrackPoint.Type.SEGMENT_START_AUTOMATIC);
                insertTrackPoint(trackId, trackPoint);
                return true;
            }

            If trackPoint.isMoving() then 
            {
              if (distanceToLastTrackLocation.greaterOrEqualThan(recordingDistanceInterval) then
              {
                if PreviousPoint_IsMovig = false then 
                {                  
                  PreviousPoint_IsMovig = true;
                  // Recording last idle point
                  insertTrackPoint(trackId, PreviousIdlePoint);
                }
                insertTrackPoint(trackId, trackPoint);
                return true;
              }
            }
            else  
            {
              if PreviousPoint_IsMovig then 
              {                
                PreviousPoint_IsMovig = false;
                // Recording first idle point
                insertTrackPoint(trackId, trackPoint);
                PreviousIdlePoint = trackPoint;
                return true;
              }
              PreviousIdlePoint = trackPoint;              
            }
        }

        Log.d(TAG, "Not recording TrackPoint, idle");
        return false;
    }

P.s. Tentatively, I added the following fields to the TrackRecordingManager class:

    private boolean PreviousPoint_IsMovig;
    private TrackPoint PreviousIdlePoint;

dennisguse, 2021-12-24:
One additional challenge is that GPS locations are not provided at a fixed interval and this might also lead to some unwanted effects.

@vlmendz I need to finish the other things first (at least #1033 and #1042).

rgmf, 2021-12-24:
I have an explanation for this (and I think it could help here): https://github.com/OpenTracksApp/OpenTracks/pull/1047#issuecomment-994414826

When you stop then moving time stops too. If you resume your activity but you stop again before distance threshold then moving time back to the value of the last point. It makes sense but it's an strange behavior.

vlmendz, 2022-02-19:
A problem with the original idle/non idle logic is when "(distanceToLastTrackLocation.greaterOrEqualThan(recordingDistanceInterval))" is not met, the code will check the idle status without taking into account the current value of the speed. This seems to produce the recording of many unwanted points.

The code proposed above attempts to correct this:

1) assuming the device is stopped only if the speed is less than the idle threshold

2) assuming that the device is moving again when the speed is greater than the idle threshold AND when "(distanceToLastTrackLocation.greaterOrEqualThan(recordingDistanceInterval)" is met

3) using a flag to only record 2 idle points: when entering the non moving state and when leaving it

Niekto350, 2022-02-19:
I can confirm this, it's especially visible during low speed recordings like a hike for example (~3km/h). If I walk for 60 minutes nonstop, the app sometimes shows 15 minutes in motion, most of the time it's wrong by about 1/2 of the recording duration. The speed graph is also very buggy / spiky.

I was thinking that the problem could be the noise from the gps sensor which could be smoothed out by calculating short scale exponential moving average of the values. That would be something like an improved combination of the 2nd and 3rd choice mentioned in the first post. The main benefit would be that the state would be changing gradually with new data points.

vlmendz, 2022-03-19:
[image]
[image]

☝️These images show a fragment of the first track (and corresponding time and speed data) that I recorded with opentracks, on January 8, 2021, using the latest APK available on F-Droid on that date.

[image]
[image]

And these ☝️ show a track recorded a few days ago using a nightly APK (build #567). Both recordings were made with the same device.

Examining the recorded speed values it can be concluded, with 100% certainty, that the recording of unwanted points in both cases is not caused by the poor accuracy and/or quality of the speed data provided by the GPS (in both cases idle threshold speed value was much lower than the recorded speed values).

The only version of OT without this problem that I have used is 3.22, but due to the way OT calculates time in motion, this functionality was broken in this version and the changes were reverted.

For users whose pattern of use is to be on movement and relatively fast all or most of the time, this problem is irrelevant and they will probably never notice it.

But for users who habitually make multiple stops or move slowly enough for the distance filter to do its job, the recording of unwanted points noticeably worsens the statistics, especially distance and time in motion.

A recording logic similar to the one I posted above a while ago, could perhaps restore the ability to record without unwanted points achieved in version 3.22 while maintaining the ability to determine time in motion without further modifications.

dennisguse, 2022-03-21:
@vlmendz thanks for reminding me.
I understood the problem and why it happens.

The problem a combination of TrackRecordingManager.lastTrackPoint:
* https://github.com/OpenTracksApp/OpenTracks/blob/ecfcc180b967f21b0852da2bb26ee3eba9af6a59/src/main/java/de/dennisguse/opentracks/services/TrackRecordingManager.java#L215
* https://github.com/OpenTracksApp/OpenTracks/blob/ecfcc180b967f21b0852da2bb26ee3eba9af6a59/src/main/java/de/dennisguse/opentracks/services/TrackRecordingManager.java#L226

I still need to figure out how we can fix this properly.

Issue occurs when (all TrackPoints considered moving):
* t=1, distance=0 (will be stored)
* t=2, distance=1 (will be assigned to lastTrackPoint and not immediately stored)
* t=3, distance=10 (will be stored and triggers storage of t=2)
Anyhow, we could ignore t=2 as it does not add useful information.

commit acb69a35 (2022-04-03): Adhere to recording distance interval for TrackPoint storage.

Fixes #1059.
PR #1166 (2022-04-10): Adhere to recording distance interval for TrackPoint storage.
Fixes #1059.

A valid LastTrackPoint (with distance) is not stored if it did not met the recording distance requirement and no change idle/non-idle took place.

@vlmendz @niekto350 could you test this?
closed 2022-04-10 by dennisguse