OpenTracksApp/OpenTracks #1060: Slow tracking refresh

vsevjednom-cz, 2021-12-24:
Tracking is lazy sometimes. Distance is refreshing randomly. I've got set 10 meters but sometimes it works fast (2-3 meters) and sometime very slowly (100-200 meters). I'm using `10 m` in `recordingDistanceInterval`. But there's not any change when I set for example `1 m`.

**To Reproduce**
1. Start tracking.
2. Sometimes it works good but when I tap to pause and continue - there is a problem with tracking. Sometimes it works lazy right away.

I tried to look into output log in Android studio at home but there isn't anything weird.

**Technical information**
 - Device: Honor KSA-LX9
 - OS: Android 9
 - Version:  9.1.0.320

Maybe some problem in my Android studio? I tried reinstall it but without changes.

dennisguse, 2021-12-25:
It depends on two things:
1. how often is your GPS providing updates?
   This might depend on a lot of factors, but largely device and surroundings.
2. how the reported speed is and how precise is it?
   In v3.22.0, we removed the storage of idle/non-idle TrackPoints and this broke the moving time.
   In v3.23.0 (also pre-v3.22.0), those are stored again.
   PS: depending on the surroundings GPS provided speed is very noisy.

And now the details:
OpenTracks stores a TrackPoint if it is considered moving (above `speed threshold`) and distance to previous TrackPoint is large enough (above `recording distance interval`).
In case of idle/non-idle changes, a TrackPoint is stored on every switch.
So, depending on your `speed threshold`, GPS accuracy etc, the later might happen more often.
And in this case, the `recording distance interval` is not considered.
_Only stored TrackPoints are used to compute statistics that are stored._
Code for this is in the `TrackRecordingManager`.

In addition, for the UI TrackPoints might be used (i.e., the most recent ones), if they are not stored.
This allows to update altitude as well as speed for the UI only.

vsevjednom-cz, 2021-12-25:
OMFG. I'm an idiot. Thank you, thank you, thank you! I set autopause speed to 5 km/h for testing few days ago.  But I forgot to bring it back. I searched problem in my Android studio 🤣. Thank you, thank you, thank you!

closed 2021-12-25 by vsevjednom-cz

dennisguse, 2021-12-25:
PS/ autopause is quite problematic if you use GPS only.
It gets even worse, if the difference between your expected speed is quite low (like walking and using 5kmh).
Then OpenTracks will create a LOT of idle/non-idle changes - even using 2.5kmh for this case might be problematic depending on the GPS accuracy at the location.

PS/ don't worry :)
And happy X-mas!