OpenTracksApp/OpenTracks #1063: pine time

JohanvdS, 2021-12-27:
I am using pine time watch but it is not connected when I asked for heart rate connection. It couldn't find the watch. Can this implemented? Gadgetbridge has no problem and also fittotrack can connect the watch. So if this is a problem ? Here it is or is the watch not included in the software? I hope it will be soon implemented.

Yours

Johan**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

dennisguse, 2021-12-27:
Not a problem.
We are filtering devices based upon the services that they announce.
I guess it is not announcing itself as a heartrate device, right?
What UUID is the PineTime announcing?

We have a workaround for Mibands in place as they also don't do this.

JohanvdS, 2021-12-27:
https://github.com/InfiniTimeOrg/InfiniTime/releases

You can find it here. I am not a developer in that case

Yours

Johan-- 
 Verzonden met Tutanota, de veilige en advertentievrije postbus.

dennisguse, 2021-12-27:
@JohanvdS Can you coordinate it with them?
What we need is the UUID that the PineTime is announcing in it's BLE announcement package.
Just reference this issue.

dennisguse, 2021-12-28:
I added an option to disable the filtering; will be in tomorrows nightly.
Should be working now.

https://github.com/OpenTracksApp/OpenTracks/pull/1064

closed 2021-12-28 by dennisguse

JohanvdS, 2021-12-29:
Thanks, i couldn't finds the UUID directly. I love to seen the work by development as I am not one. I am coaching people's to get a job lol. So not much time for invest into other interest things.

JohanvdS, 2021-12-29:
Hi Denis, 
One thing more I am not using the Google play option but the f-droid. No google on my phone ( volla.online )

Yours

Jihan
-- 
 Verzonden met Tutanota, de veilige en advertentievrije postbus.

dennisguse, 2021-12-29:
The F-Droid and the PlayStore built are identical: only difference the PlayStore is compiled on my machine.
PS/ don't worry; already reporting something helps :)

JF002, 2021-12-29:
Hello! Thanks for considering adding support for the PineTime in OpenTracks!
When I implemented the heart rate sensor, I tried to follow the BLE specification of the Heart Rate service:
 - Heart Rate Service : 0x180D
 - Measurement characteristic : 0x2A37

Here's a screenshot of NRFConnect, which detects the heart rate service correctly:
[image]

You'll find the corresponding source code here : https://github.com/InfiniTimeOrg/InfiniTime/blob/develop/src/components/ble/HeartRateService.h#L28

Is there anything else I can do so OpenTracks recognizes the PineTime as a heart rate measurement device?

JohanvdS, 2021-12-29:
What Telegram can do for a litle help :)

dennisguse, 2021-12-29:
@JF002 :)

The pine time should be working with the next release with one ugly nitpick.
I added an option to disable the filtering for supported devices.
Here is the problem for BLE discovery: upon initiating discovery nearby devices announce their presence.
This is just one package and contains the device name and (some) but not all service UUIDs.
For typical BLE sensors, those contain the useful service UUIDs (e.g., heartrate), but for multiple purpose devices (like smartwatches) that is often not the case.
And that's basically how Android's Bluetooth stack handles device discovery.

The next thing one can do is: connect to a device and execute a service discovery.
But that is costly (in terms of energy) and slow operation, so not useful for this purpose.

What we need is the UUID that the pine time is announcing :)

We have this for the Miband family: https://github.com/OpenTracksApp/OpenTracks/blob/35de5b6b264ad43b1c343a8323961db18b4874f4/src/main/java/de/dennisguse/opentracks/util/BluetoothUtils.java#L50

Thanks to some hours of remote debugging with @vanous

dennisguse, 2021-12-29:
@JohanvdS this is how nimble is configured to announce the supported service.
Don't know where this is actually configured in Infinitime...

https://mynewt.apache.org/latest/tutorials/ble/bleprph/bleprph-sections/bleprph-adv.html

PS/ https://github.com/InfiniTimeOrg/InfiniTime/blob/8ab959b4dcd0778c7637ba7934d24dbe59634327/doc/ble.md#heart-rate

BLE heartrate can be send either as UINT8 or UINT16 defined by the first bit.
So far I have only seen UINT16.

dennisguse, 2021-12-29:
I guess I found it.

Relevant code is in NimbleController:162, i.e., `fields.uuid128 = &dfuServiceUuid`.
And Infinitime announces this UUID:
```
        .value = {0x23, 0xD1, 0xBC, 0xEA, 0x5F, 0x78, 0x23, 0x15, 0xDE, 0xEF, 0x12, 0x12, 0x30, 0x15, 0x00, 0x00}};
```

https://github.com/InfiniTimeOrg/InfiniTime/blob/3b0b48020d96353fc6cd114aa80fc2fec98363a3/src/components/ble/NimbleController.h#L123

@JF002 can you confirm this?
And would you have time to test an OpenTracks build that includes this as special heart rate device?

dennisguse, 2021-12-29:
This should fix the issue: https://github.com/OpenTracksApp/OpenTracks/pull/1069
Alternative solution: InfiniTime also announces heartrate for advertisements (i.e., dfuServiceUuid).

JF002, 2021-12-30:
> @JF002 can you confirm this?
I confirm! The DFU service is advertised so that NRFConnect detects the PineTime as a DFU-capable device.

I guess I could add the UUID of the heart rate service. Is there any downside to add multiple UUID in the ADV packet?

dennisguse, 2021-12-30:
Moved to https://github.com/InfiniTimeOrg/InfiniTime/issues/895