OpenTracksApp/OpenTracks #1078: Move buttons to bottom bar

contrapunctus-1, 2022-01-03:
**Is your feature request related to a problem? Please describe.**
Buttons being placed along the top make one-handed use difficult and require more hand movement than a bottom-bar-oriented design.

**Describe the solution you'd like**
Ideally, all buttons would be moved to a bottom bar. The ones that affect me the most (and are probably easiest to relocate) are "Show on map", "Share", and the overflow menu items in the track details view.

dennisguse, 2022-01-05:
Agreed.

@contrapunctus-1 Any suggestions on the how the final result should look like?

contrapunctus-1, 2022-01-05:
@dennisguse I made some quick and dirty mockups. Five buttons between the record and stop buttons looks a little cluttered, but there's enough space for them on my 5.5" screen phone. (That said, there are smaller screen phones too...) Alternatively, the stop button could be removed while no track is recording, since it does nothing at that time...

[image]
[image]

rgmf, 2022-01-05:
I was thinking in something similar to FOSS Browser. I mean, move bar from top to bottom.

What do you think?

dennisguse, 2022-01-06:
I am all for reworking the UI; just don't want to spent too much time on it ;)
Having these overflow menu spread out looks very promising!

rgmf, 2022-02-10:
Only for notes: here it's documented a solution for this:

https://material.io/components/app-bars-bottom

rgmf, 2022-03-06:
I'm working on this. I copy/paste here some prototypes (it takes me some time so if this is not what you all expect, please let me know :smile: ):

PS/ The idea is remove pause button too (the pause function). So the only button it shows is recording and stop ones, depending it's recording or not.

[image]

[image]

dennisguse, 2022-03-06:
Looks neat!

Neat! 
Why not start with the simple solution to just move the bar down and center the start/stop button? We can add fancyness later (aka transparency, overlay layout). ;)

And could we add the current total time left of the button? 

About pause: I don't use it that often, but sometimes it is useful. 
Should we overload the button?
Aka start -> long press -> pause.
Pause -> short press -> resume.
Pause -> long press -> stop.

PS I would love to remove the pause functionality from TrackRecordingService as it makes it kind of complicated... If we only handle it in the UI our life would be way simpler.

rgmf, 2022-03-06:
I was thinking on this use case:

[image]

On stopped layout user can:
- Change name and activity type before save (and before auto-import).
- Back to recording (it's a logic pause but we can remove the pause functionality from TrackRecordingService).

rgmf, 2022-03-06:
> Why not start with the simple solution to just move the bar down and center the start/stop button? We can add fancyness later (aka transparency, overlay layout). ;)
> 

Yes, that could be great, but I have to look into the possibility of add to FAB buttons on that bottom bar.

> And could we add the current total time left of the button?
> 

I hope so :smile: 

> About pause: I don't use it that often, but sometimes it is useful. Should we overload the button? Aka start -> long press -> pause. Pause -> short press -> resume. Pause -> long press -> stop.
>
> PS [...]

I hardly ever use it. But the use case described on my last thread do the same and also reduce the impact of: #824 and #983

dennisguse, 2022-03-07:
Amazing - I would definitely use it this way!

commit 5976828b (2022-05-08): UI refactoring with a Bottom App Bar.
Only one button to play/stop.
Stop Activity to edit track just recorded, resuming or finishing.
Logo color to FF6D00.
Fixes #1078 #437 #575 #1094.
commit e9b939f1 (2022-05-08): UI refactoring with a Bottom App Bar.
Only one button to play/stop.
Stop Activity to edit track just recorded, resuming or finishing.
Logo color to FF6D00.
Fixes #1078 #437 #575 #1094.
commit da62d264 (2022-05-08): UI refactoring with a Bottom App Bar.
Only one button to play/stop.
Stop Activity to edit track just recorded, resuming or finishing.
Logo color to FF6D00.
Fixes #1078 #437 #575 #1094.
closed 2022-05-08 by rgmf