OpenTracksApp/OpenTracks #1079: Send/export data to an external URL/Server

steinm, 2022-01-05:
I wonder if there is any convenient way to export the track data to an external server by e.g. send it as json via http to a well definded api? Opentracks does not access the internet by design, so currently the only feasable way is to share or save the track data and than forward it to whereever it is supposed to go. For  many users that is probably to difficult. Would it be an option to have something like the OSM dashboard which is called from Opentracks and forwards the data to the external api?

pstorch, 2022-01-05:
Yes, you can implement an app which uses the same API as OSM Dashboard and send your tracks to a URL of your choice.

vsevjednom-cz, 2022-01-05:
> 4. (@vsevjednom-cz might have some more insights)

Yes! 😋

@steinm: you can use directory with gpx exports. I'm doing something like this:
```
        DocumentFile dir = PreferencesUtils.getDefaultExportDirectoryUri(getApplicationContext());
```

There is not possible to do automatic upload because it depends on big edit of the original code of OpenTracks. I created UploadActivity and add upload icon to the `track_list.xml`. There's async upload at FTP. You must change manifest, of course. I'm using this:
```
    <uses-permission android:name="android.permission.INTERNET"/>
    <uses-permission android:name="android.permission.ACCESS_NETWORK_STATE" />
    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE" />
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"
        tools:ignore="ScopedStorage" />
```
I'm new in Java 😲, this code can be better I think.

vsevjednom-cz, 2022-01-23:
>        ` DocumentFile dir = PreferencesUtils.getDefaultExportDirectoryUri(getApplicationContext());`

Update for v3.24:
`Uri dir = PreferencesUtils.getDefaultExportDirectoryUri();` 🙃

And then something like this:
```
            final String[] split = dir.getPath().split(":");
            String path = Environment.getExternalStorageDirectory() + File.separator + split[1];
            File directory = new File(path);
```

dennisguse, 2022-04-16:
Nothing to do here.

closed 2022-04-16 by dennisguse