OpenTracksApp/OpenTracks #1080: still some issues with Export to NextCloud

mega-stoffel, 2022-01-05:
**Describe the bug**
As mentioned already in #1025, my export to a NextCloud shared directory is sometimes not working. Most of the tracks are exporting perfectly fine after I record them. But unfortunately sometimes not. I haven't found a pattern, yet.
I tried to get a logcat output and I guess, this might be the error message. It's maybe not even related to OpenTracks, but some misbehavior of NextCloud?

```
[01-05 21:59:45.684 15568:15621 E/ExportUtils]Unable to open exportDocumentFile content://org.nextcloud.documents/tree/1405706866%2F12103/document/1405706866%2F16011java.io.FileNotFoundException: Failed to open document for writing 54f6640c_2022-01-04 08_34.gpx
	at android.database.DatabaseUtils.readExceptionWithFileNotFoundExceptionFromParcel(DatabaseUtils.java:149)
	at android.content.ContentProviderProxy.openAssetFile(ContentProviderNative.java:631)
	at android.content.ContentResolver.openAssetFileDescriptor(ContentResolver.java:1523)
	at android.content.ContentResolver.openOutputStream(ContentResolver.java:1229)
	at android.content.ContentResolver.openOutputStream(ContentResolver.java:1205)
	at de.dennisguse.opentracks.util.ExportUtils.exportTrack(ExportUtils.java:49)
	at de.dennisguse.opentracks.io.file.exporter.ExportService.onHandleWork(ExportService.java:62)
	at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:396)
	at androidx.core.app.JobIntentService$CommandProcessor.doInBackground(JobIntentService.java:387)
	at android.os.AsyncTask$3.call(AsyncTask.java:378)	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
	at java.lang.Thread.run(Thread.java:919)
```
**Technical information**
 - Device: Fairphone FP3
 - OS: Android 10
 - OpenTracks version: latest from FDroid, 3.23.0
 - OpenTracks commit id (only for nightly builds): [e.g. 4b333bb]

pstorch, 2022-01-06:
Thanks for the logcat. I'm still analyzing this.
What I see is that the directory gets not writable after some time. In the nightly you can now see an indication this is the case.

pstorch, 2022-01-07:
It still happens: after updating the nightly the write permission is gone one the Nextcloud directory. So somehow the persistable permission is not working. I still don't know why.

dennisguse, 2022-01-07:
@pstorch Could it be that we also need to request `FLAG_GRANT_PREFIX_URI_PERMISSION`?
For the export, we actually need to get access to URIs below the one we actually requested, right?

https://developer.android.com/reference/android/content/Intent#FLAG_GRANT_PREFIX_URI_PERMISSION

And are there differences between Android flavours?

PS/ I have no test setup available for this.

pstorch, 2022-01-07:
Interesting flag. Didn't know about it. With local directories it works without. And actually the permission vanishes at the parent directory, which should be an exact match. But anyway, I can give it a try.

pstorch, 2022-01-09:
Some other error I found in the logcat when opening the settings activity:

```
2022-01-09 12:44:28.892 29547-29547/de.dennisguse.opentracks.debug E/ActivityThread: Failed to find provider info for org.nextcloud.documents
2022-01-09 12:44:28.893 29547-29547/de.dennisguse.opentracks.debug W/DocumentFile: Failed query: java.lang.NullPointerException: Attempt to invoke interface method 'boolean android.database.Cursor.moveToFirst()' on a null object reference
2022-01-09 12:44:28.893 1022-3755/? I/AppsFilter: interaction: PackageSetting{c78af43 de.dennisguse.opentracks.debug/10473} -> PackageSetting{249cc88 com.nextcloud.client/10338} BLOCKED
2022-01-09 12:44:28.893 29547-29547/de.dennisguse.opentracks.debug E/ActivityThread: Failed to find provider info for org.nextcloud.documents
2022-01-09 12:44:28.893 29547-29547/de.dennisguse.opentracks.debug W/DocumentFile: Failed query: java.lang.NullPointerException: Attempt to invoke interface method 'boolean android.database.Cursor.moveToFirst()' on a null object reference
2022-01-09 12:44:28.974 472-472/? I/io_stats: !@   8,0 r 12409800 422031840 w 10954507 375611948 d 870350 328442000 f 2913825 4089688 iot 9647340 8093993 th 102400 102400 67108 pt 8 inp 0 0 449645.136
2022-01-09 12:44:29.019 1022-3518/? E/TaskPersister: File error accessing recents directory (directory doesn't exist?).

```

pstorch, 2022-01-09:
Some more interesting reads:

https://github.com/nextcloud/android/issues/7821
https://github.com/nextcloud/android/issues/5702
https://github.com/nextcloud/android/issues/8336

But I still don't know if one of them is applicable to our problem.

dennisguse, 2022-01-28:
I found something that might be related.
When we call `TrackRecordingService.endCurrentTrack()`, it executes `TrackRecordingService.endRecording(true)` and this calls `stopSelf()` before calling `ExportUtils.postWorkoutExport()`.

So, there is the potential of a race condition as the service might be already getting destroyed during the export initialization (especially if this takes longer).

@pstorch You can try #1113.
Commit is https://github.com/OpenTracksApp/OpenTracks/tree/762d9704068fe26ecf780764285ff90d179645b2

pstorch, 2022-01-28:
Ah, this was the other way around a few commits ago. I've changed it, because I found a potential error in `postWorkoutExport()` which could cause an Exception and then the recording wasn't stopped properly. And I thought the recording should be stopped properly before exporting it. But if this interferes with the Service lifecycle, this might be a problem.

dennisguse, 2022-01-28:
Yes, recording should be stopped before exporting (although it should also work if the recording is running), but the service should yet stop itself.

mega-stoffel, 2022-01-30:
Thanks a lot for your work! :-)

mega-stoffel, 2022-02-10:
Nextcloud obviously made some changes in their latest release 3.19.0, which was published this weekend via the f-droid store.
I need to reconfigure my export directory with this version.
Let's see, how it behaves now. :-)

dennisguse, 2022-04-02:
@mega-stoffel Is this still a problem?

mega-stoffel, 2022-04-03:
@dennisguse - I don't think so. I haven't seen any export errors since the latest updates.
If it appears again, I'll let you know.

Thanks a lot for your work!

dennisguse, 2022-04-04:
@mega-stoffel 
Kudos are going to @pstorch

commit 762d9704 (2022-01-28): TrackRecordingService: call stopSelf after triggering post-workout export.

Might fix #1080.
commit 3ae60151 (2022-02-07): TrackRecordingService: call stopSelf after triggering post-workout export.

Might fix #1080.
commit a7fd6b8d (2022-02-14): TrackRecordingService: call stopSelf after triggering post-workout export.

Might fix #1080.
closed 2022-04-04 by dennisguse