OpenTracksApp/OpenTracks #1082: `StringUtilsTest.testFormatDateTime()` failing

dennisguse, 2022-01-07:
`StringUtilsTest.testFormatDateTime()`.

Using `DateTimeFormatter.FormatStyle.MEDIUM` appends if the time is before lunch (e.g., AM); it even does this for ES and DE.

Test fails on CI and my machine.

In addition, the default locale of the device is modified and thus subsequent tests may fail afterwards.

_Originally posted by @dennisguse in https://github.com/OpenTracksApp/OpenTracks/issues/1081#issuecomment-1006940238_

dennisguse, 2022-01-07:
@rgmf Any ideas?

rgmf, 2022-01-07:
In Spanish we never add AM and PM after time, in fact nobody knows what that mean 😅

I'd modify that test to be more flexible with that.

rgmf, 2022-01-07:
PS/ in my local machine all tests passes (using AVD and real phones).

dennisguse, 2022-01-07:
In German, we also don't ;)

I get:
For US: AM
For ES: a. m.
For DE: vorm.

Kind of weird; never seen this in German.
I got this consistently in the emulator as well as on a LineageOS device.

And my LineageOS device behaves differently than my emulator as it produces some commas differently. 
I am mindblown...

PS/ the assert has the expectation first and as 2nd actual

dennisguse, 2022-01-07:
Can we simplify the expectation by defining it as string and not creating it from LocaleDateTime?

commit 9d2c3a0d (2022-01-07): StringUtilsTest.testFormatDateTime removed. Fixes #1082.
commit 674ccfd2 (2022-01-08): StringUtilsTest.testFormatDateTime removed. Fixes #1082.
commit eba4918b (2022-01-08): StringUtilsTest.testFormatDateTime removed. Fixes #1082.
closed 2022-01-08 by rgmf