OpenTracksApp/OpenTracks #1086: Crash while paused tracking in v3.24.0

vsevjednom-cz, 2022-01-11:
OpenTracks `v3.24.0` crashes while tracking is paused. Just tap to the `By time` and wait one second.

**To Reproduce**
1. Start tracking.
2. Pause tracking.
3. Tap to the `By time`.
4. Wait one second.

Log in Android studio:
```
E/pentracks.debu: [qarth_debug:]  get PatchStore::createDisableExceptionQarthFile method fail.
E/AndroidRuntime: FATAL EXCEPTION: main
    Process: de.dennisguse.opentracks.debug, PID: 18415
    java.lang.NullPointerException: Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference
        at de.dennisguse.opentracks.chart.ChartView.drawPointer(ChartView.java:852)
        at de.dennisguse.opentracks.chart.ChartView.onDraw(ChartView.java:597)
        at android.view.View.draw(View.java:20483)
        at android.view.View.updateDisplayListIfDirty(View.java:19294)
        at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4375)
        at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4348)
        at android.view.View.updateDisplayListIfDirty(View.java:19254)
        at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4375)
        at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4348)
        at android.view.View.updateDisplayListIfDirty(View.java:19254)
        at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4375)
        at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4348)
        at android.view.View.updateDisplayListIfDirty(View.java:19254)
        at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4375)
        at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4348)
        at android.view.View.updateDisplayListIfDirty(View.java:19254)
        at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4375)
        at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4348)
        at android.view.View.updateDisplayListIfDirty(View.java:19254)
        at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4375)
        at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4348)
        at android.view.View.updateDisplayListIfDirty(View.java:19254)
        at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4375)
        at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4348)
        at android.view.View.updateDisplayListIfDirty(View.java:19254)
        at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4375)
        at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4348)
        at android.view.View.updateDisplayListIfDirty(View.java:19254)
        at android.view.ViewGroup.recreateChildDisplayList(ViewGroup.java:4375)
        at android.view.ViewGroup.dispatchGetDisplayList(ViewGroup.java:4348)
        at android.view.View.updateDisplayListIfDirty(View.java:19254)
        at android.view.ThreadedRenderer.updateViewTreeDisplayList(ThreadedRenderer.java:740)
        at android.view.ThreadedRenderer.updateRootDisplayList(ThreadedRenderer.java:746)
        at android.view.ThreadedRenderer.draw(ThreadedRenderer.java:861)
        at android.view.ViewRootImpl.draw(ViewRootImpl.java:3583)
        at android.view.ViewRootImpl.performDraw(ViewRootImpl.java:3379)
        at android.view.ViewRootImpl.performTraversals(ViewRootImpl.java:2711)
        at android.view.ViewRootImpl.doTraversal(ViewRootImpl.java:1633)
        at android.view.ViewRootImpl$TraversalRunnable.run(ViewRootImpl.java:7943)
        at android.view.Choreographer$CallbackRecord.run(Choreographer.java:1092)
        at android.view.Choreographer.doCallbacks(Choreographer.java:893)
        at android.view.Choreographer.doFrame(Choreographer.java:812)
        at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:1078)
        at android.os.Handler.handleCallback(Handler.java:907)
        at android.os.Handler.dispatchMessage(Handler.java:105)
        at android.os.Looper.loop(Looper.java:216)
        at android.app.ActivityThread.main(ActivityThread.java:7625)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:987)
```

And I'm still below than 0 meters above sea level in v3.24.0 🤪.

**Technical information**
 - Device: Honor KSA-LX9
 - OS: Android 9
 - Version:  9.1.0.320

vsevjednom-cz, 2022-01-11:
You can tap to the `By time` while tracking 😲. Pause in not required. Crash 🤪.

vsevjednom-cz, 2022-01-11:
... and there was a crash of OpenTracks at random time after 63 kms (about two hours of tracking) at my road bike. I'm confused. I don't know the reason.

dennisguse, 2022-01-11:
@vsevjednom-cz Abou the 2nd problem it might be again the lack of a monotonic clock or something else :)
To get a log: you can increase the logcat size on your device to get them later when you have a computer at hand.

vsevjednom-cz, 2022-01-11:
I didn't know about logcat, thanks! But there isn't anything from the time of the crash (20.18). There's only some information from 20.35. This is the time when I arrived at home and I saw the OpenTracks crashed.

Here is the log. I don't know if it will help you. It looks like some error in my phone 🤪.
```
2022-01-11 20:35:02.245 30832-30892/? A/libc: Fatal signal 6 (SIGABRT), code -6 (SI_TKILL) in tid 30892 (pool-2-thread-3), pid 30832 (racks.osmplugin)
2022-01-11 20:35:02.436 30925-30925/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2022-01-11 20:35:02.437 30925-30925/? A/DEBUG: Build fingerprint: 'HONOR/KSA-L29EEA/HWKSA-M:9/HONORKSA-LX9/9.1.0.272C431:user/release-keys'
2022-01-11 20:35:02.437 30925-30925/? A/DEBUG: Revision: '0'
2022-01-11 20:35:02.437 30925-30925/? A/DEBUG: ABI: 'arm'
2022-01-11 20:35:02.437 30925-30925/? A/DEBUG: pid: 30832, tid: 30892, name: pool-2-thread-3  >>> de.storchp.opentracks.osmplugin <<<
2022-01-11 20:35:02.437 30925-30925/? A/DEBUG: signal 6 (SIGABRT), code -6 (SI_TKILL), fault addr --------
2022-01-11 20:35:02.437 30925-30925/? A/DEBUG: Abort message: 'Error, cannot access an invalid/free'd bitmap here!'
2022-01-11 20:35:02.437 30925-30925/? A/DEBUG:     r0  00000000  r1  000078ac  r2  00000006  r3  00000008
2022-01-11 20:35:02.437 30925-30925/? A/DEBUG:     r4  00007870  r5  000078ac  r6  d1574ef4  r7  0000010c
2022-01-11 20:35:02.438 30925-30925/? A/DEBUG:     r8  00000001  r9  d2a36200  r10 16ee3ed0  r11 13ebd6c0
2022-01-11 20:35:02.438 30925-30925/? A/DEBUG:     ip  d1574e90  sp  d1574ee0  lr  f376ac31  pc  f3762a7e
2022-01-11 20:35:02.623 30925-30925/? A/DEBUG: backtrace:
2022-01-11 20:35:02.623 30925-30925/? A/DEBUG:     #00 pc 0001da7e  /system/lib/libc.so (abort+58)
2022-01-11 20:35:02.623 30925-30925/? A/DEBUG:     #01 pc 00007055  /system/lib/liblog.so (__android_log_assert+156)
2022-01-11 20:35:02.623 30925-30925/? A/DEBUG:     #02 pc 000d208b  /system/lib/libandroid_runtime.so (android::bitmap::toBitmap(_JNIEnv*, _jobject*)+34)
2022-01-11 20:35:02.623 30925-30925/? A/DEBUG:     #03 pc 000d0f83  /system/lib/libandroid_runtime.so (android::CanvasJNI::drawBitmapMatrix(_JNIEnv*, _jobject*, long long, _jobject*, long long, long long)+6)
2022-01-11 20:35:02.623 30925-30925/? A/DEBUG:     #04 pc 003f3a83  /system/framework/arm/boot-framework.oat (offset 0x3ef000) (android.graphics.BaseCanvas.nDrawBitmapMatrix+162)
2022-01-11 20:35:02.623 30925-30925/? A/DEBUG:     #05 pc 008981a7  /system/framework/arm/boot-framework.oat (offset 0x3ef000) (android.graphics.Canvas.drawBitmap+302)
2022-01-11 20:35:02.623 30925-30925/? A/DEBUG:     #06 pc 00100e53  /data/app/de.storchp.opentracks.osmplugin-Tg4nppUeQwpFFIMCeNIBFg==/oat/arm/base.odex (offset 0x60000) (org.mapsforge.map.android.graphics.AndroidCanvas.drawBitmap+402)
2022-01-11 20:35:02.623 30925-30925/? A/DEBUG:     #07 pc 000dcde1  /data/app/de.storchp.opentracks.osmplugin-Tg4nppUeQwpFFIMCeNIBFg==/oat/arm/base.odex (offset 0x60000) (org.mapsforge.core.mapelements.SymbolContainer.draw+288)
2022-01-11 20:35:02.623 30925-30925/? A/DEBUG:     #08 pc 000a0005  /data/app/de.storchp.opentracks.osmplugin-Tg4nppUeQwpFFIMCeNIBFg==/oat/arm/base.odex (offset 0x60000) (org.mapsforge.map.layer.renderer.CanvasRasterer.drawMapElements+324)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #09 pc 00106a0b  /data/app/de.storchp.opentracks.osmplugin-Tg4nppUeQwpFFIMCeNIBFg==/oat/arm/base.odex (offset 0x60000) (org.mapsforge.map.layer.renderer.DatabaseRenderer.executeJob+930)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #10 pc 000a06e5  /data/app/de.storchp.opentracks.osmplugin-Tg4nppUeQwpFFIMCeNIBFg==/oat/arm/base.odex (offset 0x60000) (org.mapsforge.map.layer.renderer.MapWorkerPool$MapWorker.run+396)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #11 pc 003d669d  /system/framework/arm/boot.oat (offset 0x132000) (java.util.concurrent.ThreadPoolExecutor.processTask+1068)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #12 pc 003d7c1f  /system/framework/arm/boot.oat (offset 0x132000) (java.util.concurrent.ThreadPoolExecutor.runWorker+62)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #13 pc 003d4957  /system/framework/arm/boot.oat (offset 0x132000) (java.util.concurrent.ThreadPoolExecutor$Worker.run+54)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #14 pc 00270911  /system/framework/arm/boot.oat (offset 0x132000) (java.lang.Thread.run+64)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #15 pc 0042e975  /system/lib/libart.so (art_quick_invoke_stub_internal+68)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #16 pc 00401821  /system/lib/libart.so (art_quick_invoke_stub+224)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #17 pc 000a82b9  /system/lib/libart.so (art::ArtMethod::Invoke(art::Thread*, unsigned int*, unsigned int, art::JValue*, char const*)+136)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #18 pc 0035e9cd  /system/lib/libart.so (art::(anonymous namespace)::InvokeWithArgArray(art::ScopedObjectAccessAlreadyRunnable const&, art::ArtMethod*, art::(anonymous namespace)::ArgArray*, art::JValue*, char const*)+52)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #19 pc 0035f725  /system/lib/libart.so (art::InvokeVirtualOrInterfaceWithJValues(art::ScopedObjectAccessAlreadyRunnable const&, _jobject*, _jmethodID*, jvalue*)+320)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #20 pc 00381203  /system/lib/libart.so (art::Thread::CreateCallback(void*)+866)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #21 pc 00064d55  /system/lib/libc.so (__pthread_start(void*)+22)
2022-01-11 20:35:02.624 30925-30925/? A/DEBUG:     #22 pc 0001ec41  /system/lib/libc.so (__start_thread+24)
```

pstorch, 2022-01-12:
Oh, this is a crash in the OSMDashboard. From the Mapsforge library? I think I saw something similar on Play Store, but I don't know what I can do about it. Maybe I can report it to Mapsforge.

PR #1087 (2022-01-12): Bugfix: ChartView needs data.
Fixes #1086.

@vsevjednom-cz Does this fix the issue?
Don't know exactly how I broke it - too late for an investigation :)
commit bc1edec3 (2022-01-12): Bugfix: ChartView needs data.

Fixes #1086.
closed 2022-01-12 by dennisguse