OpenTracksApp/OpenTracks #1088: Record heartrate history even if position is unavailable or does not change

crosser, 2022-01-12:
Currently OpenTracks only saves history along with the GPS track, including the heart rate.

It would be useful if one could use the app to record their heartrate when they, for example, do a workout in a gym, where GPS position is be unknown, or known, but constant.

One way to represent it would be to make a KML track with dummy / constant trackpoints that still go every second even when position does not change, with `<when>` element only, and fill the heartrate section in `ExtendedData` with real data.  (There could be other ways too.)

dennisguse, 2022-01-12:
Nice timing!
v3.24.0 can now be used without a GPS, but requires distance sensors (e.g., for cycling or running).
Such tracks can only be exported completely using KML/KMZ as here the latitude/longitude is optional.

Also BLE sensors are now event-driven handled, so we can store the data every time we get something new.

Only change required for this issue: we would need to ignore the distance to the previous TrackPoint.
We could make it configurable, so all TrackPoints fulfill the distance requirement.

Relevant code: `TrackRecordingManager.onNewTrackPoint()`

abe-winter, 2022-04-19:
+1! I tried opentracks in the gym and thought it was broken until I found the idle speed setting

naive non-android suggestion here, but is there a hacky solution of just adding a 0 setting to the idle speed threshold in settings.xml?

https://github.com/OpenTracksApp/OpenTracks/blob/c271dd32ac1c1dae492801eac27be172e65e6b2b/src/main/res/values/settings.xml#L187

then isMoving would always be true IMO

https://github.com/OpenTracksApp/OpenTracks/blob/5ea651d3408db0b8e637b62ac9ce6344ae6740ba/src/main/java/de/dennisguse/opentracks/data/models/Speed.java#L67-L69

I'm happy to submit this PR if it's in any way helpful

dennisguse, 2022-04-19:
@abe-winter Almost :)
The speed does not matter, but the distance to the previous location.

We could just set this to 0m and everything would work fine (default in the settings is 10m):
https://github.com/OpenTracksApp/OpenTracks/blob/5ea651d3408db0b8e637b62ac9ce6344ae6740ba/src/main/java/de/dennisguse/opentracks/services/TrackRecordingManager.java#L175

However, the recording distance applies some implicit smoothing to the noisy GPS data.
So, adding 0m is quite a nice hack, but then the user would need to reconfigure OpenTracks each time when he switches between indoor use (no distance) and outdoor use (with distance).

If you want to take a look - have fun :)

abe-winter, 2022-04-19:
ah, thanks for the pointer to code

I understand the logic in that `onNewTrackPoint` function, I can submit this PR if you want.

Alternate approach to this: should I add a 'gym' or 'indoor' category which disables the location-based logic? That potentially solves your problem of the user having to edit prefs every time they switch

dennisguse, 2022-04-20:
That would be an option - but we need to wait at least until #1155 is merged as this changes a lot of the UI code and also restructures the bottom bar (formerly TrackController).

Until now we lack the logic to disable/ ignore the GPS: if it is available (i.e., not turned off by Android or the permissions not granted), OpenTracks will use it.

I guess the only viable option is to add this a setting (to record sensor data if the distance does not change).
However, this raises the question: should we then store each data point (BLE sensor data is now processed event-driven)?

dennisguse, 2022-04-20:
The 2nd idea would be like our distance filtering (ignore everything until a threshold is reached).
Sounds simple enough - it affects heartrate and powermeter.
We might have an edge case for sensors that provide one these and distance in addition (like Tickr X and Stryd).
But we can fine tune it later :)

Another aspect regarding the 1st idea - we should process the event at it's time of creation (aka store it).
Data that we store must have a monotonously increasing time (this is a hard requirement) as we need to be able to compute statistics without reloading all data and ordering it.

So, I am in favor of the 2nd concept.

@pstorch @rgmf What do you think?

For maintenance reasons: new libs are not my favorite solution.
Until now we just have androidx and some Google UI extensions - this keeps maintenance effort quite low.
Also mixing different threading approaches within the same application might require quite some mental effort.
So, implementing 2 is possible, but you would need to do this manually via Threading / Handler.

dennisguse, 2022-04-20:
Another aspect: we filter new data in GPSHandler (depending on the accuracy) and in the TrackRecordingManager (based on traveled distance).
And a question is where the filter (and likely later aggregation) is done for distance-less sensor data.

abe-winter, 2022-04-20:
(thanks for your continued help here)

from skimming gpshandler + trackrecordingmanager, it looks like:
- GPSHandler is registering [here](https://github.com/OpenTracksApp/OpenTracks/blob/0eda1eed1dd4516caf724a701f6941bcac4e3c90/src/main/java/de/dennisguse/opentracks/services/handlers/GPSHandler.java#L151) for updates with `minTimeMs=PreferencesUtils.getMinRecordingInterval`, `minDistanceM=0`
- `PreferencesUtils.getThresholdHorizontalAccuracy` is the accuracy filter you mentioned (why is this important? because gps accuracy is low when you're indoors?)
- [GPSHandler.onLocationChanged](https://github.com/OpenTracksApp/OpenTracks/blob/0eda1eed1dd4516caf724a701f6941bcac4e3c90/src/main/java/de/dennisguse/opentracks/services/handlers/GPSHandler.java#L124) is calling `TrackPointCreator.onNewTrackPoint`, which fills the trackpoint from bluetooth sensors + stores it via `Callback.newTrackPoint`
- `TrackRecordingManager` does distance filtering [here in onNewTrackPoint](https://github.com/OpenTracksApp/OpenTracks/blob/0eda1eed1dd4516caf724a701f6941bcac4e3c90/src/main/java/de/dennisguse/opentracks/services/TrackRecordingManager.java#L170-L180), based on `PreferencesUtils.getRecordingDistanceInterval`, and the distance calculation varies based on hasLocation / hasSensorDistance

re event frequency, it seems like the cases are:
1. in the GPS pathway, event frequency is limited by `minRecordingInterval` (which is [zero by default](https://github.com/OpenTracksApp/OpenTracks/blob/0eda1eed1dd4516caf724a701f6941bcac4e3c90/src/main/res/values/settings.xml#L151), I think) + `recordingDistanceInterval`. stationary heart rate can be supported by setting `recordingDistanceInterval=0` and minRecordingInterval to nonzero
2. with sensor distance, `recordingDistanceInterval` should work fine indoors
3. with no distance sensor + GPS turned off, I'm guessing no data gets written?
4. the pathological case is if `hasSensorDistance` is true, but the user has also set `recordingDistanceInterval=0` -- in this case, it seems like every sensor event will get written

^ in case 4, is the simplest fix to respect `minRecordingInterval` in `TrackRecordingManager.onNewTrackPoint`? (by comparing lastTrackPoint.time). And potentially disallowing minRecordingInterval=0?

I don't totally understand why there's a need for a min distance filter -- hypothetically, if you capture all sensors (including location / distance) at the end of every time interval, does that produce the necessary time series with simpler logic? as a jogger / hiker, I'm potentially interested in my heart rate after I've stopped moving so I can track recovery, so I might set minDistance=0 even if I'm outdoors and moving

pstorch, 2022-04-20:
> So, I am in favor of the 2nd concept.
> 
> @pstorch @rgmf What do you think?

Introducing rxjava only for this one feature feels a bit too heavy. I once contributed to an Android App which uses rxjava, but nearly everywhere, so it was part of its DNA. In OT I would also go for the second option.

rgmf, 2022-04-20:
> So, I am in favor of the 2nd concept.
> 
> @pstorch @rgmf What do you think?

I think rxjava would great for threading but introduce it at this point to the app would be hard because it would not make sense mixing several threading optiond.

I'd use the second option either.

dennisguse, 2022-04-21:
>     * GPSHandler is registering [here](https://github.com/OpenTracksApp/OpenTracks/blob/0eda1eed1dd4516caf724a701f6941bcac4e3c90/src/main/java/de/dennisguse/opentracks/services/handlers/GPSHandler.java#L151) for updates with `minTimeMs=PreferencesUtils.getMinRecordingInterval`, `minDistanceM=0`

Yes.

>     * `PreferencesUtils.getThresholdHorizontalAccuracy` is the accuracy filter you mentioned (why is this important? because gps accuracy is low when you're indoors?)

Just FYI.

> * [...]

Yes. And a small change is scheduled to be merged to prevent a race condition: #1167.

> * `TrackRecordingManager` does distance filtering [here in [...]

Yes, either we take the distance (from sensor) or we compute it relative to the previous GPS location.

> re event frequency, it seems like the cases are:
>
>     1. in the GPS pathway, event frequency is limited by `minRecordingInterval` (which is [zero by [...]

Yes, but no. Until now a TrackPoint must have a distance (i.e., non null) to be stored and a heartrate/power meter TrackPoint has `hasSensorDistance() == null`.

>     2. with sensor distance, `recordingDistanceInterval` should work fine indoors

Indeed it does - even outdoors with GPS disabled (e.g., to save battery).

>     3. with no distance sensor + GPS turned off, I'm guessing no data gets written?

Yes, due to 1.

>     4. the pathological case is if `hasSensorDistance` is true, but the user has also set `recordingDistanceInterval=0` -- in this case, it seems like every sensor event will get written

Yes.
 
> ^ in case 4, is the simplest fix to respect `minRecordingInterval` in `TrackRecordingManager.onNewTrackPoint`? (by comparing lastTrackPoint.time). And potentially disallowing minRecordingInterval=0?

Likely; minRecordingDistance interval of 0s was so far just used for GPS, so we could introduce a second preference for BLE data.
TrackPoints that have distance must still comply with minRecordingDistance.
TrackPoints without distance should comply with the __new__ minRecordingInterval?

> zoom-out design question: I don't totally understand why there's a need for a min distance filter -- hypothetically, if you capture all sensors (including location / distance) at the end of every time interval, does that produce the necessary time series with simpler logic? as a jogger / hiker, I'm potentially interested in my heart rate after I've stopped moving so I can track recovery, so I might set minDistance=0 even if I'm outdoors and moving

GPS may be very noise due to weather, the actual location, and eve time (and it was even worse for phones in 2010, when MyTracks was implemented [best guess]).
So, even if you don't move the measured coordinates will fluctuate around your location.
This creates two problems:
1. we would store too much data, because there is no real added benefit of storing this
2. we would overestimate the actual distance you moved
By requiring a certain distance to be traveled (or not in case of low accuracy), we limit the impact of the problem.
The assumption is that you are moving in direct line (i.e., shortest path) from your previous location to the next one.

We could handle heartrate sensor data differently though.

abe-winter, 2022-04-21:
thanks for the detailed response + continued attention on this

for clarity, let's call your **new** minRecordingInterval param something like `minSensorInterval`?

This didn't make sense to me:

> Yes, but no. Until now a TrackPoint must have a distance (i.e., non null) to be stored and a heartrate/power meter TrackPoint has hasSensorDistance() == null.

^ For case 1, I meant that if we set `recordingDistanceInterval`=0, GPS events would come through even if stationary, and heart rate events would be attached to them via `fill()`. Are you saying that's not correct?

I'm imagining three ways forward at this point:
1. **config-only workaround**: leave GPS active in the gym, play around with minimum time / distance / accuracy configs, maybe add minSensorInterval
2. **reuse location**: for 'sensor-only' events (i.e. heart rate / work), respect the minSensorInterval and attach the distance or location from the last written trackpoint (i.e. reuse 'last good location'). This allows me to jog in the gym, and capture my cooldown while exercising outdoors. But it doesn't throw away the minimum-distance / jitter protection logic for location changes
3. **harder path**: capture heart rate with null GPS/distance. I suspect this violates some non-null expectations, breaks charting, potentially breaks other math in the app, complicates export logic, and risks app crashes

dennisguse, 2022-04-22:
> ^ For case 1, I meant that if we set `recordingDistanceInterval`=0, GPS events would come through even if stationary, and heart rate events would be attached to them via `fill()`. Are you saying that's not correct?

Until now; and that is something we would need to change.
Due to GPS inaccuracy, `recordingDistanceInterval`=0 would be working, but the summed distance would be inflated.

>     1. **config-only workaround**: leave GPS active in the gym, play around with minimum time / distance / accuracy configs, maybe add minSensorInterval

Together with 3.?

> 2. **reuse location**: for 'sensor-only' events (i.e. heart rate / work), respect the minSensorInterval and attach the distance or location from the last written trackpoint (i.e. reuse 'last good [...]

Attaching old data to the next TrackPoint feels wrong; we do this for sensor data, but have a TTL of 5s.
We would need to do something similar here and then we still sometimes end up with 3.
And this would require the GPS do be turned on.

FYI TrackPoints are write only; it makes a lot of the code easier and also prevents weird states if something crashes.
We already have TrackPoints without coordinates (`TrackPoint.Type.SENSORPOINT`) [might have distance]/.

>     3. **harder path**: capture heart rate with null GPS/distance. I suspect this violates some non-null expectations, breaks charting, potentially breaks other math in the app, complicates export logic, and risks app crashes

That's what we do for SENSORPOINT and it feels like the way to go, so minSensorInterval  + no distance.

Don't worry about NPE - these are easy to discover and the chart/export code is easy to adjust.
The problem is usually TrackRecordingManager and TrackStatisticsUpdater.

Export for KML/KMZ should work out of the box.
for GPX there might be the need to be some changes as this format doesn't allow to export TrackPoints without coordinates, but we already have code in place to merge the data into the next TrackPoint with coordinates.

abe-winter, 2022-04-22:
Path 3 (store sensor with null location + distance) makes sense to me -- also neat that it works without any GPS.

if you think this won't break a lot of other logic, I'm happy to give it a shot.

dennisguse, 2022-04-22:
Awesome!
I would recommend to:
1. start storing heartrate data
2. add test coverage (at least in ExportImportTest add at least a distance-free heartrate TrackPoint) for KML/KMZ
3. check if GPX export needs to be adjusted 
4. check if ExportImportTest for GPX still makes sense
5. add the minSensorInterval (also needs some kind of test coverage)

aka first get it running and then optimize it ;)

And I would recommend to start on top of #1167 - I guess, it will be merged in the next days.

PS/ test coverage for this is really important.
We have too little, but for the essential logic is pretty important.
Otherwise it is so easy to mess things up (been there done that :sunglasses: ).
And you are touching some critical parts.
Still enjoy it ;)

abe-winter, 2022-04-22:
sounds good. thanks for the outline, it's helpful . I'll branch off 1167 + pay attention to tests.

commit 773c31f1 (2022-07-11): Store sensor data without distance.

Fixes #1088.
PR #1325 (2022-07-11): Store sensor-only data (without distance)
Stores sensor-only data like heartrate without requiring GPS/distance data.
Allows to train indoor and still collect heartrate or powemeter over time.

Fixes #1088.

Note: not yet tested.
commit 5667d2d8 (2022-07-11): Store sensor data without distance.

Fixes #1088.
closed 2022-07-11 by dennisguse