OpenTracksApp/OpenTracks #1094: Activity metadata when saving a recording

Niekto350, 2022-01-15:
**Is your feature request related to a problem? Please describe.**
When you stop recording, the file get's automatically saved with default categories, name etc. However every time I want to add a metadata like select an activity category, description etc. Right now I have to manually reopen the recording to do the edits which is annoying.

**Describe the solution you'd like**
I would like the app to automatically ask for such metadata, like asking for an activity type (bike ride, hike etc). Or automatically open the edit screen to finish the recording. This could possibly be configurable option for people who never want's to save such data. 

**Describe alternatives you've considered**
Alternative approach could be a big edit button directly on the bottom of the recording (there is an empty space for it) to save a few taps every time a recording is made.

dennisguse, 2022-04-02:
I recently installed the final version of My Tracks and your solution was implemented there.
Might be fixed in  #1155.

dennisguse, 2022-04-02:
@rgmf FYI

rgmf, 2022-04-02:
> I recently installed the final version of My Tracks and your solution was implemented there. Might be fixed in #1155.

Yes, this will be fixed on #1155 because on #1155 the track will be auto-saved when user click on finish button on TrackStoppedActivity where user can set metadata before auto-saved.

Niekto350, 2022-04-05:
> I recently installed the final version of My Tracks and your solution was implemented there. Might be fixed in #1155.
Thank you, I'll update the app and give it a try on my next activity. :)

One more thing I've been thinking about. We can set a default activity type, let's say hiking or mountain biking, this works fine. However I was thinking we could improve that if we use activity speed (or other metadata) into account. So basically we can have a "smart" prediction of the activity type or at least set low/high speed activity category to make it a bit more user friendly.

Currently if I have a default activity as a hiking and go for a 100km bike ride with average speed of 30km/h the app labels the activity as a hiking despite that is practically impossible for a human as it's the default action.

Maybe depending on the speed and elevation we could predict subcategories of the activity, let's say biking: road biking and mountain biking.

If you like the idea, we could break it up into new issue.

dennisguse, 2022-04-05:
@Niekto350 Why not :)
Don't know how to do this, but it is worth thinking about - separate issue is a great way to start.

Niekto350, 2022-04-10:
Ok, I've made it into a new issue here. https://github.com/OpenTracksApp/OpenTracks/issues/1177 I've made some edits to the original post to explain it a bit more.

I've installed the latest app few days ago but didn't notice any change related to this issue (just some other ui improvements, which are great by the way). I guess the changes are still in the beta version release so I'll give it another try later.

closed 2022-05-08 by rgmf