OpenTracksApp/OpenTracks #1095: No file manager found

Niekto350, 2022-01-15:
**Describe the bug**
I've changed the file manager on my phone to simple file manager. I cannot do any file related actions like import/export because the app says no file manager is found (some actions fails silently).

The same issue exist on the offline dashboard companion app (https://github.com/OpenTracksApp/OSMDashboard). This one shows the notification about missing file manager. So basically I cannot specify a location for the map files and have a blank screen with my track recording.

This is the file manager I'm using:
https://play.google.com/store/apps/details?id=com.simplemobiletools.filemanager.pro&hl=en_US&gl=US
https://f-droid.org/en/packages/com.simplemobiletools.filemanager.pro/

**To Reproduce**
1. Uninstall default phone file manager / switch to simple file manager.
2. Click on any file manager related action in the app (for example export).

**Technical information**
Using android 10 and latest OpenTracks and Simple File manager.

pstorch, 2022-01-15:
Unfortunately it looks like this FileManager doesn't provide DOCUMENTS_PROVIDER functionality which is needed. 
OpenTracks and OSMDashboard need a DocumentProvider to access files.

dennisguse, 2022-01-15:
Can we do something about the silent failures?

pstorch, 2022-01-15:
Yes, we could catch the ActivityNotFoundException as in OSMDashboard and show a message:

https://github.com/OpenTracksApp/OSMDashboard/blob/34ba6b91d0aaffe0c12d4653fa38655d5f388227/src/main/java/de/storchp/opentracks/osmplugin/BaseActivity.java#L235

Niekto350, 2022-01-15:
Thank you, I've opened a ticket here https://github.com/SimpleMobileTools/Simple-File-Manager/issues/550

Do you need the file manager to do some more complex stuff like file transfer or it's used just to generate the file path? If all you need is a file path string, could we do something like text input to type the path manually?

pstorch, 2022-01-16:
@Niekto350 it's more than just the file path. Android has this Storage Access Framework, which allows Apps without general file permission to access files outside of it's private space via a DocumenProvider App like a FileManager (can also be a remote cloud storage app).

Niekto350, 2022-01-16:
Thanks for the explanation.

pstorch, 2022-01-16:
> Can we do something about the silent failures?

I'll implement the message.

pstorch, 2022-01-20:
@Niekto350 if you like you can test our nightly build. Would be good to know if the error message appears that no compatible file manager is found

Niekto350, 2022-01-22:
I'm quite busy these days but yes I could give it a try once I'll find a minute. However I don't know how to install that, I've installed the app from the fdroid app store and I don't see any option to install nightly build (or any other than a regular release). Or maybe I could just test it in the production, it's not something that could break the app so I think it's safe to release even without some in depth testing and in the unlikely case of any problems fix it in the next release. I should also note that I'm absolute beginner with Android as a user and have practically no experience in Android development so please apologize that if it's a stupid question.

dennisguse, 2022-01-23:
Fixed in 1099102e29eaa16ced6f0ad9944516f39f98b22e.

closed 2022-01-23 by dennisguse

Niekto350, 2022-01-25:
So I've gave it a try and the message is working nicely.

I would like to ask, would it be possible for you to submit a PR to simple file manager to implement this feature required by OpenTracks? Or would it be too much work?

I've also tested some more file managers just to get the map somehow working but I had no luck. I've installed 7 file managers and none of them worked. Can you recommend some file managers that are known to be working correctly with OpenTracks?

pstorch, 2022-01-25:
@Niekto350 thanks for testing. I searched for a few minutes, but couldn't find an OSS filemanager which provide this kind of functionality. 
Maybe you can ask in the F-Droid Forum: https://forum.f-droid.org

Implementing a FileManager with this functionality or sending a PR to an existing one is out of my scope at the moment.

dennisguse, 2022-01-25:
That is kind of unexpected. I was so far using the default installed file managers and had never any trouble.

Which ROM are you using?

Niekto350, 2022-01-26:
> @Niekto350 thanks for testing. I searched for a few minutes, but couldn't find an OSS filemanager which provide this kind of functionality. Maybe you can ask in the F-Droid Forum: [...]
No problem, glad that I could help. I understand.

> That is kind of unexpected. I was so far using the default installed file managers and had never any trouble.
> 
> Which ROM are you using?
I would like to install lineage os but I have never done such thing and I'm worried that I would break the phone. For example if some command does not work / I ran into some problems and end up with black screen, I have no idea how to recover from that. Would be a bit easier if I know how to create full disk backup and restore with something like dd on Linux systems but don't know how to do that either on Android. Maybe sometimes in the future when I find some material to learn such skills I'll give it a try.

So right now I'm using just the original stock installation. The only thing I've did is boot into adb shell and uninstall some preinstalled bloatware including original file manager. Instead I've installed some opensource and privacy oriented file managers and ended up with simple file manager because it seemed like the best choice.

Niekto350, 2022-08-13:
I think there is still some kind of a issue with this feature.

I've been recording some tracks over the last months and wanted to export the data to my notebook. I'm still unable to find some file manager that is compatible with OpenTracks, so I've reinstalled the original stock file manager of the phone using adb. The file manager works normally but when I want to export the tracks (or do any of the file manager related actions in OpenTracks or the map dashboard companion app, the file manager does not start it just blinks for 1/2 second and disappears/crashes back to the OpenTracks). I guess the original file manager is compatible because I no longer see the notification that no compatible file manager is found.

In case this is complicated to solve, would it be possible to use other method of exporting? For example using FTP protocol?

pstorch, 2022-08-13:
@Niekto350 does the system file manager has all the permissions it needs? I don't know what happens if you uninstall and reinstall it via adb. Any chance of a logcat. Sounds like the file manager is crashing.

You can implement your own FTP Connector with the OpenTracks Dasboard API.

Niekto350, 2022-08-15:
I suppose it have the permissions, it have asked for the storage access which I've granted. I can startup the file manager as a standalone app and it works normally. I don't know what a logcat is or more specifically how to obtain it. Glad to hear that the ftp connector is possible however I doubt that I would be able to code it as I have zero knowledge in Android development.

pstorch, 2022-08-20:
@Niekto350 I'm struggling a bit on how to help here.

You wrote:

> So right now I'm using just the original stock installation. The only thing I've did is boot into adb shell and uninstall some preinstalled bloatware including original file manager. 

Which vendor/device is it? Samsung, Huawai, OnePlus, OPPO, ...?

Which file manager app(s) did you uninstall exactly? For example on my Samsung device there are two
- an app called "My Files" which can be launched as an app. This app has a link to the (Samsung) App store.
- an app called just "Files", this I can only see in the Settings / Apps list. It is not startable by itself, but I guess this is the one started by the Intent we need. This app doesn't have a link to an App store, so I guess this was part of the operating system.

Niekto350, 2022-08-23:
@pstorch It's a secondary cheap phone I'm only using as a sport tracker and podcast player, model is Realme C11 3GB. I don't know the names of the original apps, I haven't used them and replaced it pretty much right away (after the phone was laying in my desk drawer for a few months doing nothing and before I've gave it the use case described above). However I haven't uninstalled only the file manager but other bloatware as well like Facebook stuff and so on.

I should have some backup of the commands / exact packages I've uninstalled somewhere if it would be some help. More or less it's based on the google research what other users suggested when debloating the phone and grep results when looking for a packages to get the exact name.

Regarding the file manager it should be this:
```
pm uninstall -k --user 0 com.coloros.filemanager
pm uninstall -k --user 0 com.google.android.apps.nbu.files
pm uninstall -k --user 0 com.google.android.documentsui
pm uninstall -k --user 0 com.google.android.apps.docs
```
I have a backup because I've been reinstalling it with:
```
cmd package install-existing --user 0 com.coloros.filemanager
cmd package install-existing --user 0 com.google.android.apps.nbu.files
cmd package install-existing --user 0 com.google.android.documentsui
cmd package install-existing --user 0 com.google.android.apps.docs
```
However from my understanding the uninstall/reinstall process should not have any effect on this and when I start the original file manager I don't see anything out of the ordinary.

My current objective is to export/backup the tracks to my notebook (it does not have to be the dedicated import/export functionality, some workaround would be more than sufficient). First thing is that I don't want to loose my recordings in case I crash on my bike and break the phone / accidentally destroy the phone with water during activity recording. I took me a lot of effort to record hundreds of recordings over the time I've switched to OpenTracks and I don't want to loose it. The second goal is to build Vue based UI in my free time to view the tracks on the big screen / use my recordings as a sample data to develop it.

pstorch, 2022-08-26:
@Niekto350 did you run the "pm" command via adb? If your phone is connected to a PC, you can use `adb logcat` to view the logs of your system and then try to open the file manager via OT and see if you get any error output we can analyze.