OpenTracksApp/OpenTracks #1101: Vocal announce in French language are incorrect

solfred, 2022-01-21:
Hi, 

I am a French user of OpenTracks. 
The vocal annoucements of time are incorrect

The term "second" is replaced by "minute"

For example instead of saying 15 minutes and 30 **seconds**, OpenTracks is saying 14 minutes and 30 **minutes**.

dennisguse, 2022-01-21:
You are right.
Should it be "seconde" and "secondes"?

Translations are handled via Weblate: https://weblate.bubu1.eu/engage/OpenTracks/
If you find anything else... my French is almost non-existing...

PS/ was my fault; I introduced this problem last year....

solfred, 2022-01-21:
In French "Seconde" and "Secondes" sound the same. You can use "Seconde".

dennisguse, 2022-01-21:
@solfred using the same value for singular and plural leads to errors while linting.
Thanks for reporting the issue; will be fixed with the next release.
Or you start using the nightly build (fix will be included tomorrow).

closed 2022-01-21 by dennisguse

solfred, 2022-01-22:
Thanks a lot for the change.

commit 05bd2e6a (2022-01-21): Voice announcement: fix French seconds.

Fixes #1101.