OpenTracksApp/OpenTracks #1103: Public API: stop recording

dennisguse, 2022-01-23:
TrackRecordingActivity was implemented under the assumption that it would be the only activity that would interact with TrackRecordingService while opened.
It needs to be adjusted for the activity-based approach for public api via Intent as it just restarts the recording after the service handled the stop.

Issue is in the bindCallback.

* [ ] Fix TrackRecordingActivity: if it is open, it resumes the track recording after STOP_RECORDING
* [ ] enable STOP_RECORDING
* [ ] add stop to README.md

Follow-up to #1072

arjan-s, 2022-01-30:
Hi, just wanted to let you know that I started working on implementing the start and stop recording APIs for my wearable in Gadgetbridge [in this PR](https://codeberg.org/Freeyourgadget/Gadgetbridge/pulls/2577).

dennisguse, 2022-01-30:
@arjan-s cool!
I was actually assuming that you would be using some IFTTT engine...
... and OpenTracks has actually four packgesId:
* de.denniguse.opentracks
* ... .nightly
* ... .debug
* ... .playstore (was an early fuckup on my side)

So, hardcoding those in Gadgetbridge feels kind of uncool, but I am not sure how to improve this as I don't know of Gadgetbridge...

arjan-s, 2022-01-31:
Thanks!
I'm not an IFTTT user, so maybe I'm missing something here... What do you think could be gained by using that? Gadgetbridge intends to be as dependency free as possible and completely cloud-free as well, so going the plain intent route seems right.

Yes, I know hardcoding the OpenTracks package IDs in Gadgetbridge isn't really pretty, but neither is scanning all installed apps to find a supported GPS tracking app. And retrieving a list of installed apps/packages got limited for privacy reasons in Android 11, and might be limited even more in future versions.

An implicit intent would have solved this partially, but as we discussed before that didn't work right. So explicit it is, and as such we need a package ID. My idea is to just let the Gadgetbridge user choose an OpenTracks variant (f-droid, play, etc) and then use the relevant package ID in the intent.

vanous, 2022-02-06:
> Yes, I know hardcoding the OpenTracks package IDs in Gadgetbridge isn't really pretty, but neither is scanning all installed apps to find a supported GPS tracking app. And retrieving a list of [...]

@arjan-s just have the package name being a preference string, fill in the default value and let the user to edit it...

arjan-s, 2022-02-06:
My plan exactly :)

dennisguse, 2022-04-24:
Related: https://github.com/OpenTracksApp/OpenTracks/issues/1156#issuecomment-1107856451

closed 2022-05-08 by dennisguse

vanous, 2022-05-08:
Awesome, thank you @dennisguse !

dennisguse, 2022-05-08:
@vanous :sunglasses:

arjan-s, 2022-05-09:
Agreed, thank you, awesome @dennisguse ! :)

commit c577ae90 (2022-05-08): Public API: implemented STOP properly.

Fixes #1103.
commit 7b0e3f3a (2022-05-08): Public API: implemented STOP properly.

Fixes #1103.
commit c8db7dba (2022-05-08): Public API: implemented STOP properly.

Fixes #1103.
commit 9d64989e (2022-05-08): Public API: implemented STOP properly.

Fixes #1103.