OpenTracksApp/OpenTracks #1106: Fix linting issues

dennisguse, 2022-01-25:
.... reported by upgrading Android Gradle plugin to 7.1.

dennisguse, 2022-01-26:
... and the OpenTracks cannot be build anymore, because the Android Gradle plugin now expects to have an app directory.
https://developer.android.com/studio/releases/gradle-plugin

Just one of those moments...

@pstorch do you any suggestions beside moving the code into an `app`-folder?
Or reverting the upgrade?

dennisguse, 2022-01-26:
I am confused: executing gradle from commandline (also in CI) works.
Executing it via AndroidStudio 2022.1.1 leads to `Project 'OpenTracks' not found in root project 'OpenTracks'.`.

Downgrading the plugin and AndroidStudio works again.

rgmf, 2022-01-26:
@dennisguse I don't know if it's the same thing, but I've also run on troubles with AndroidStudio and Gradle (it couldn't build OpenTracks) and what I did was download AndroidStudio again (2022.1.1) and install it again.

After that, It works.

dennisguse, 2022-01-26:
"Have you tried turning it off and on again" :D
Will give it a try and hope for the best

pstorch, 2022-01-26:
It's working for me (now).
First I had to upgrade AndroidStudio. Struggled a bit because of "no space left on device". Yeah, did a lot with VMs and Containers lately :wink: 
After that was solved and I pulled the latest commits for OpenTracks, AS told me to upgrade gradle wrapper to 7.2 (which already was) and reimport the project. I clicked the link to perform the action and it is building and running the app without any issues.

dennisguse, 2022-01-26:
... actually deleting my `.android` folder solved the problem.
Now I also have some clean emulators :D

dennisguse, 2022-01-28:
Upgrade completed and linting issues either fixed or made warning.s

closed 2022-01-28 by dennisguse