OpenTracksApp/OpenTracks #1108: Elevation gain/loss is hugely inflated

phiggins, 2022-01-26:
**Describe the bug**
Whenever I record a track, the elevation gain/loss is hugely inflated.

[image]

[image]

**To Reproduce**
This happens every time I record a track.

**Technical information**
 - Device: Samsung S10+
 - OS: Android 11
 - OpenTracks version: 3.2.4 from Play Store

rgmf, 2022-01-26:
In our tests there are smartphones whose barometric sensor works well and others don't (read https://github.com/OpenTracksApp/OpenTracks/blob/main/README_TESTED_SENSORS.md).

I've had smartphones with barometric sensor whose values go up and down when you shake it, also when you press the screen. So they aren't measure the atmospheric pressure like expected but the pressure on the smartphone.

You can test your smartphone with apps like TrailSense. Open it, go to pressure tab and shake/press your smartphone. If pressure value changes then OpenTracks cannot measure gain/loss correctly because when you are running, walking, biking... your smartphone is getting a lot of values of pressure up and down.

phiggins, 2022-01-26:
> You can test your smartphone with apps like TrailSense. Open it, go to pressure tab and shake/press your smartphone. If pressure value changes then OpenTracks cannot measure gain/loss correctly [...]

I tried this and nothing I did appeared to affect the app's pressure value. While using the app the value did change rapidly back and forth between 1025.6 hPa and 1025.7 hPa, could that be causing the discrepancy?

If there's nothing to be done to fix it, can I disable the elevation display? There's no point in displaying it if it is wildly inaccurate.

rgmf, 2022-01-26:
> I tried this and nothing I did appeared to affect the app's pressure value. While using the app the value did change rapidly back and forth between 1025.6 hPa and 1025.7 hPa, could that be causing the discrepancy?

Thes hPa difference values seem very little but enough to OpenTracks to detect gain and loss. Anyway, I expected greater differences to get such a gain/loss values from your activities.

> If there's nothing to be done to fix it, can I disable the elevation display? There's no point in displaying it if it is wildly inaccurate.

It sounds like a good idea.

BTW @dennisguse could we store pressure values together gain and loss? In that way we could debug, see and figure out what happen in such a cases :thinking:

dennisguse, 2022-01-26:
@rgmf Yes, we can do this.
What we can easily: let AltitudeSumManager send a callback to TrackPointCreator to create a SENSORPOINT; then we just need to enable that all TrackPoints are stored (aka ignoring the distance requirement i.e, implementing #1088)
And we need to return not the altitude but the actual sensor value.

dennisguse, 2022-01-26:
@phiggins Yes, you can configure the UI to now show the elevation gain/ loss.

What we are actually doing: we apply exponential smoothing to these air pressure data as it is kind of noisy and if the difference then is larger than 3m, we count it as gain or loss.

Could you check what happens if you don't move the phone?
And what happens if hold the phone in your hand while walking?

We might also have a bug in OpenTracks somewhere...

@rgmf PS/ my new phone doesn't have barometer :/
Have to pull out my good old S4 if I need one...

dennisguse, 2022-01-26:
And according to our current code: the phone needs to report at least 1026.0 over some duration to be above the 3m threshold.
```
        float a = SensorManager.getAltitude(SensorManager.PRESSURE_STANDARD_ATMOSPHERE, 1025.6f);
        float b = SensorManager.getAltitude(SensorManager.PRESSURE_STANDARD_ATMOSPHERE, 1026.0f);
        assertEquals(a,b, 0.01);
``
leading to these values in meters
```
java.lang.AssertionError: expected:<-102.31413269042969> but was:<-105.61168670654297>
```

Tardigrad3, 2022-03-01:
> **Describe the bug** Whenever I record a track, the elevation gain/loss is hugely inflated.
> 
> [image]
> 
> [image]
> 
> **To Reproduce** This happens every time I record a track.
> 
> **Technical information**
> 
>     * Device: Samsung S10+
> 
>     * OS: Android 11
> 
>     * OpenTracks version: 3.2.4 from Play Store

Reverting back to V3.20.2 fixed the issue for me, hope this helps.

dennisguse, 2022-04-16:
@phiggins Please re-open if this is still a problem for you.

closed 2022-04-16 by dennisguse