OpenTracksApp/OpenTracks #1110: Allow backups for Android 12

dennisguse, 2022-01-26:
More details here: https://developer.android.com/about/versions/12/behavior-changes-12#backup-restore

dennisguse, 2022-01-26:
After reading the actual changes: from Android 12 onward, Google Drive backup and device2device transfer are separated and can be deactivated individually.
So, it should just work...

@pstorch @rgmf Do you have devices to test this?
I am still running Android 11 and it might be take quite a while before I get an upgrade to 12.

pstorch, 2022-01-27:
My device is still on Android 11.

rgmf, 2022-01-27:
Yes, I can test it on Android 12.

rgmf, 2022-01-28:
Reading this: https://support.google.com/pixelphone/answer/7129955

It seems you only can transfer data app from one device to another the first time you start your destination Android device, right? In such a case I'd have to reset my phone to test it.

pstorch, 2022-01-31:
Oh, and I hoped local (adb) Backups will be allowed then. 😔

dennisguse, 2022-01-31:
@pstorch ... me too :/

dennisguse, 2023-05-23:
If this works, it works - if not then not.

closed 2023-05-23 by dennisguse