OpenTracksApp/OpenTracks #1116: Optionally provide live track updates to other apps

arjan-s, 2022-02-02:
**Is your feature request related to a problem? Please describe.**
With the addition of the intent API (https://github.com/OpenTracksApp/OpenTracks/pull/1072 and https://github.com/OpenTracksApp/OpenTracks/issues/1103) it is now possible for Gadgetbridge and other apps to start and end recordings while the phone's display is still locked. There is currently no ability for the same apps to receive live track updates (like speed and distance) from OpenTracks.

**Describe the solution you'd like**
My preferred solution for providing live track updates is to implement the explicit intent approach. Perhaps the settings screen could even have a "target package ID" setting or something like that, so give the user choice. I'm thinking about the way [Tiny Weather Forecast Germany](https://f-droid.org/en/packages/de.kaffeemitkoffein.tinyweatherforecastgermany/) does it [here](https://codeberg.org/Starfish/TinyWeatherForecastGermany/src/branch/master/app/src/main/java/de/kaffeemitkoffein/tinyweatherforecastgermany/GadgetbridgeAPI.java#L112).

**Describe alternatives you've considered**
Alternatively, since Gadgetbridge already has access to all notifications, it could try to parse the OpenTracks ongoing notification and extract the relevant data from it. This is, however, a bad practice and prone to break in the future. Also, implementation of this would be a problem due to translations.

The existing dashboard API is able to provide live updates, but it has to be started manually for each new track. The proposed solution is for automatic live updates and hopefully doesn't require a lot of extra code.

dennisguse, 2022-02-02:
What is your use case?
Assumption: the currently recording Track
And which data of a Track do you need?
* last TrackPoints
* all TrackPoints
* overall statistics

We could re-use the DashboardAPI and allow starting it automatically returning the URIs (incl. temp. read permissions) on Intent.

arjan-s, 2022-02-02:
Sorry to keep bothering you with these questions!

My use case is that wearables supported by Gadgetbridge could display the overall statistics of a currently recording track on their watchface or in a workouts app.
Perhaps the URI-on-intent method could work. I haven't looked into the dashboard API yet though. Could it allow an app without internet permissions to periodically (or streaming) receive the overall statistics?

dennisguse, 2022-02-02:
Understood.

Internet/ no Internet permission doesn't matter.
The URIs are pointing to the OpenTracks contentprovider.
What you are interested in is the Track URI.

Relevant code:
* https://github.com/OpenTracksApp/OpenTracks/blob/19a958ccbb9ba569db31eb8a34526dc98e1528bb/src/main/java/de/dennisguse/opentracks/util/IntentDashboardUtils.java#L85
* https://github.com/OpenTracksApp/OSMDashboard/blob/a42ce3117051cd7b2f634e45e0ce84f9d6e1a8a5/src/main/java/de/storchp/opentracks/osmplugin/MapsActivity.java#L181

(For now) My suggestion for this case would be to enhance the StartRecording to  trigger the DashboardAPI (after starting the recording) and let Gadgetbridge handle this intent.

arjan-s, 2022-02-03:
Great, thanks for the pointer! This does indeed sound like exactly what I need. I'll pick this up soon(ish).

commit 1e4e7b8c (2022-02-16): Allow starting the Dashboard API automatically from the StartRecording API

Fixes #1116.
closed 2022-02-16 by dennisguse