OpenTracksApp/OpenTracks #1117: Recording: speed shows 0.0 most of the time.

rgmf, 2022-02-03:
**Describe the bug**
When a track recording is on you can see that speed is 0.0 most of the time but average is computed and showed correctly.

Every once in a while you can see the current speed that vanishes quickly to show 0.0 again.

I've seen this issue using GPS speed (without speed sensor). I couldn't test it with speed sensor yet.

**To Reproduce**
1. Start a recording.
2. Look at speed.

**Technical information**
 - Device: Pixel 4a
 - OS: Android 12
 - OpenTracks version: debug version (current main branch).

dennisguse, 2022-02-03:
I can confirm this. 
Testing without any sensor. 

Problem seems to be that UI gets a new TrackPoint to update the time, but the provided location is considered to old. 
We need to check what the validity duration is. Is either in TrackPointCreator or TrackRecordingManager.

dennisguse, 2022-02-03:
Here is the problem: https://github.com/OpenTracksApp/OpenTracks/blob/19a958ccbb9ba569db31eb8a34526dc98e1528bb/src/main/java/de/dennisguse/opentracks/services/handlers/TrackPointCreator.java#L124

If we store a TrackPoint, we call `GPSHandler.reset()` and thus setting `lastTrackPoint = null`.
This reset is a good idea for storing the data, but for the UI this information is still valid and should (not yet) be discarded.

Still thinking on how to handle this properly.

This effect was introduced in 5998f9127dee84e71821e091760f06d86f288815
... to prevent storing an unrelated TrackPoint for another track.

rgmf, 2022-02-04:
> Here is the problem:
> 
> https://github.com/OpenTracksApp/OpenTracks/blob/19a958ccbb9ba569db31eb8a34526dc98e1528bb/src/main/java/de/dennisguse/opentracks/services/handlers/TrackPointCreator.java#L124
> 
> If we store a TrackPoint, we call `GPSHandler.reset()` and thus setting `lastTrackPoint = null`. This reset is a good idea for storing the data, but for the UI this information is still valid and should (not yet) be discarded.
> 
> Still thinking on how to handle this properly.
> 
> This effect was introduced in [5998f91](https://github.com/OpenTracksApp/OpenTracks/commit/5998f9127dee84e71821e091760f06d86f288815) ... to prevent storing an unrelated TrackPoint for another track.

So lastTrackPoint is lastTrackPointNotStored, right? It would importat to change the name of this property, right?

Also, if we reset sensors then speed will be set to 0, so the issue is also there altough you use speed sensor.

It seems we need another approach to handle all these stuffs, but I don't know how for now.

dennisguse, 2022-02-04:
@rgmf You are absolutely right!
But still no clue how to fix it nicely :)

PR #1124 (2022-02-17): Bugfix: speed showed sometimes 0.0 in UI.
Fixes #1117.

Should only fix the problem for using the GPS.
BLE sensors might still be affected if they were - I had no time to check it.
commit 0848897e (2022-02-17): Bugfix: speed showed sometimes 0.0 in UI.

Fixes #1117.
closed 2022-02-17 by dennisguse