OpenTracksApp/OpenTracks #1118: Heart rate zones and graph in the real of time

MrS1R1US, 2022-02-04:
**Describe the solution you'd like**
There is a very lack of heart rate zones, as in Polar Beat. When the heart rate zones are set and there is a real-time color graph, it is easier to follow the correct heart rate. This is a very important schedule during a serious workout. 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
[This is what the final chart looks like. Also, the training graph with a line is built heart rate, which allows you to see the line that you cannot cross. ](https://i.imgur.com/hsvnR75.jpeg)

P.S. You took into account my previous wish regarding automatic backups to the phone and via Syncthing to the PC, for which I am very grateful to you!

MrS1R1US, 2022-02-05:
[This is how the graph looks like during a workout, the line moves depending on the heart rate zone. Over time, when the workout is long, the first heart rate point goes beyond the graph in order to make the graph smoother. Please note that the heart rate numbers themselves change color depending on the heart rate zone, this is very helpful for real-time analysis to slow down or increase it. ](https://i.imgur.com/tOMxAsX.png)

This is how I would like to see it in OpenTracks:
[https://i.imgur.com/MPQ7DYg.png](https://i.imgur.com/MPQ7DYg.png)

This is a very important function, to see in which pulse zone you are. This feature is available in all sports devices to which a heart rate monitor is connected. In the settings you set the maximum heart rate, and then it is calculated down from 100%. It is also worth noting that it is important to save the graph at the end of the workout, this will allow you to compare with past changes.
Zone 5 - 90-100%
Zone 4 - 80-90%
Zone 3 - 70-80%
Zone 2 - 60-70%
Zone 1 - 50-60%

MrS1R1US, 2022-02-05:
Please pay due attention to this. )

dennisguse, 2022-02-05:
@MrS1R1US somebody might if somebody has time ;)
PS/ it would be helpful if you upload the images to Github instead of imgur.

Is this issue different than #185?

MrS1R1US, 2022-02-05:
> PS/ it would be helpful if you upload the images to Github instead of imgur.
Ok

> Is this issue different than #185?

This topic complements and correctly reveals the question) 

P.S. After 2 years, no one wants to do this? This is an important functionality for a sports application.

dennisguse, 2022-02-05:
From my side, it is lack of time and I rather did other things in OpenTracks (mainly focusing on sensors and fixing internals).
So yeah - it was not a high priority and seemingly everybody else felt the same or there are not enough people out there that could contribute.
Kind of a standard problem for open-source projects :shrug: 
Anyhow: a lot stuff happened already, so somebody might pick us this one as well.

MrS1R1US, 2022-02-06:
Yes, I was surprised when I saw the speed and cadence sensor. I'm going to buy them for the summer. It's also very good.

It would be nice if someone would take care of this functionality .. (Heart rate zones and graph in the real of time)

MrS1R1US, 2022-02-06:
Also, for my part, I can offer help at least in translating the Russian/Ukrainian interface if necessary.

MrS1R1US, 2022-02-07:
@dennisguse, tell me please, can you at least here (temporarily, suddenly there is someone who would do all the functionality) on the graphs to sort the lines, divided by 6? In order to at least indicate the maximum heart rate here, and then sort it into 6 parts based on it. Now, for example, here it is 225-200-175-150-125, but would it be 200-180-160-140-120-100?
[image]

P.S. I have not found how to upload screenshots to the github server ..

dennisguse, 2022-02-07:
Interesting solution to the problem - let me check if this can be easily done in the next days.
Would at least be a nice hack, right? :)

MrS1R1US, 2022-02-08:
> Would at least be a nice hack, right? :)

@dennisguse, yes) And if possible, I would like to ask you to make a color change during training depending on the zone:
100%-90% - red #ff0000
90%-80% - orange #ffbd04
80%-70% - Green #6acc2b
70%-60% - Blue #64d1fd
60%-50% - default color.

[image]

MrS1R1US, 2022-02-16:
@dennisguse, what do you say ?

dennisguse, 2022-02-16:
I really like the color idea!
It is easy to build.

Let me see if I have some time for this in the next weeks.

MrS1R1US, 2022-11-22:
> @MrS1R1US It took quite a while to enable this. Anyhow, it is now feasible. Would you have the time to do it yourself? Code for the UI is `GenericStatisticsViewHolder.SensorHeartRate`. Just create a new layout and have fun.

Hello. Unfortunately, I don't have programming skills. I looked at the settings - there is nothing anywhere, also there are no updates in f-droid.

dennisguse, 2022-11-22:
@MrS1R1US you are right, there is nothing released yet. Let me see if I have some days before X-mas.

dennisguse, 2022-11-25:
Changes will be in tomorrows nightly.
Feedback appreciated.

commit 19fe5275 (2022-11-25): Heartrate zones: adding preference for max heartrate.

Fixes #1118.
PR #1427 (2022-11-25): Heartrate zones
- [x] adding preference for max heartrate.
- [x] render UI

Fixes #1118.
closed 2022-11-25 by dennisguse