OpenTracksApp/OpenTracks #1119: Type: Swimming Open Water Workout

MrS1R1US, 2022-02-04:
Please add activity type: Swimming (Without GPS) and Swimming in open water (with GPS).

dennisguse, 2022-02-04:
Do you need two different ones? OpenTracks does not have a profile support. So the activity does not affect what and how data get recorded (eg which BLE sensor, with or/without GPS).

dennisguse, 2022-02-04:
@pstorch Do you have an icon available?

MrS1R1US, 2022-02-04:
`Do you need two different ones?`

Need activity - icon)

pstorch, 2022-02-04:
Two different icons for "Swimming" and "Swimming in open water"?

MrS1R1US, 2022-02-05:
> Two different icons for "Swimming" and "Swimming in open water"?

Yes please) And "Fitness" if you can. (Need for workout with use heart monitor)

pstorch, 2022-02-05:
I'm not a designer, but I like playing with Inkscape. I don't want to copy something from the "internet" to not run into copyright issues, so I tried my own interpretation of these activities.

Swimming in the open:
[image]

Swimming:
[image]
This should depict a swimming pool. But I'm not happy with it. Maybe the general swimming should be the first icon and "swimming in the open" should have some extra characteristic. But I'm not sure what. Maybe bigger waves or a small island with a palm tree :wink:

Workout activity with heart-rate monitoring:
[image]

PS: ignore the different background color. This will be transparent later in the app.

rgmf, 2022-02-05:
@pstorch they are fantastic!!! I love them.

Anyway, I agree with you: I'd use "swimming in the open" for swimming and for swimming in the open I'd add a sun or something like this, a simple detail like this would be enough IMHO.

I love workout activity.

Great use of Inkscape ;)

MrS1R1US, 2022-02-05:
> This should depict a swimming pool. But I'm not happy with it. Maybe the general swimming should be the first icon and "swimming in the open" should have some extra characteristic. But I'm not sure what. Maybe bigger waves or a small island with a palm tree wink

Very cool) For open water, I think you can leave this logo, just add a small palm tree to the background. And for fitness, maybe some a man with biceps or a icon - dumbbell?) What do you think?

MrS1R1US, 2022-02-05:
> Workout activity with heart-rate monitoring:

I suggest this for crossfit ))

pstorch, 2022-02-05:
Standard swimming:
[image]

Swimming open water:
[image]

Workout: 
[image]

Now with one muscle arm and a dumbbell. It looks a bit funny :laughing:

MrS1R1US, 2022-02-05:
> Now with one muscle arm and a dumbbell. It looks a bit funny laughing

:D
Maybe try two muscle?)))

Swimming good :)

pstorch, 2022-02-05:
Ok, let's settle the swimming icons.

The muscle man with more muscles:
[image]

Maybe it's a bit too much.

Or back to sticky figure with straight lines:
[image]

MrS1R1US, 2022-02-06:
I think the second option matches the general style of the icons:
[image]

pstorch, 2022-02-06:
I think so too.
Other question: despite I like the heart, is it really necessary on this specific activity? I mean most of the sports activities we have affect your heart beat.

MrS1R1US, 2022-02-06:
> I think so too. Other question: despite I like the heart, is it really necessary on this specific activity? I mean most of the sports activities we have affect your heart beat.

The heart symbolizes that this workout can actually record only heartbeat and time, because speed and distance do not make sense. Here, as you wish, you can remove, you can leave.

pstorch, 2022-02-06:
Ah sure, that is the difference to the others where movement is the main thing we track.

@dennisguse and @rgmf what do you think?

rgmf, 2022-02-06:
@pstorch about the workout I like the last version, without muscle and with heartbeat. Also, I think the heartbeat focuses that it's a hr/time activity.

pstorch, 2022-02-06:
I've opened a PR: https://github.com/OpenTracksApp/OpenTracks/pull/1121

commit 7c063270 (2022-02-06): fix #1119 add new activities "swimming", "swimming open" and "workout"
commit 142b58da (2022-02-07): fix #1119 add new activities "swimming", "swimming open" and "workout"
closed 2022-02-06 by pstorch