OpenTracksApp/OpenTracks #1120: Running training does not show pace

MrS1R1US, 2022-02-04:
**Describe the bug**
Running training does not show pace. Need choice pace/speed on ready workout for analyze past workouts.

**To Reproduce**
1. Go to 'Ready workout'
2. See error "Speed"

If applicable:
https://i.imgur.com/Qs24gD6.png

**Technical information**
 - Device: All devices
 - OS: All OS
 - OpenTracks version: Current version

dennisguse, 2022-02-04:
i don't understand. You can show the pace - both while recording (customize UI feature) and for past recordings (speed/pace can be changed in the settings).

MrS1R1US, 2022-02-04:
Sorry! I didn't see this setting! Done, the pace is on the run, and the speed on the bike) Please close this appeal)

dennisguse, 2022-02-04:
:D

PS/ you can also close issues yourself :)

closed 2022-02-04 by dennisguse