OpenTracksApp/OpenTracks #1127: Option to disable overview PNG on track sharing

mardom1, 2022-02-14:
**Is your feature request related to a problem? Please describe.**
When sharing a track it also shares a picture of some statistics (since an update a while ago). This is frustrating for me, as I used this to upload the GPX tracks to Nextcloud. I now have to delete that picture manually afterwards.

Btw: The export feature is even more frustrating, as it modifies the file name which I would like to be the same as the Track name. I would have to change it manually after uploading, which is even more bothersome than deleting files, especially as the UUID makes them randomly sorted.

**Describe the solution you'd like**
Add an option in the settings (or when sharing) to not add the overview picture.

**Describe alternatives you've considered**
Add an option to the export feature to use the track name as file name. Also export of a single track would be helpful here.

**Additional context**
[image]

dennisguse, 2022-02-14:
@rgmf @pstorch I would remove this functionality.
I guess it is not too useful anyhow (without showing a map or similar).
Any objections?

@mardom1 Why are you not using the auto export functionality?

closed 2022-02-14 by dennisguse

mardom1, 2022-02-15:
That went quickly, thank you!

> @mardom1 Why are you not using the auto export functionality?

Well, in theory it would be perfect for my use case, but I don't want the track ID in the file name, as is the case with the (auto) export feature (#512). So in the end, I find it easier to share new tracks with Nextcloud than to rename them after the auto export.

And for the record: the app is simply amazing, so thank you for the continued efforts!

PR #1128 (2022-02-14): Share: removed PNG share pic.
Fixes #1127.
commit 094c97f7 (2022-02-14): Share: removed PNG share pic.

Fixes #1127.