OpenTracksApp/OpenTracks #1141: Auto-Export + Export All Fail while Export via Share remains only working export option

Der-Flip, 2022-02-26:
**Describe the bug**
Autoexport and "export All" fail.

Autoexport: Export fails with message: check Export folder. Folder is set and correct and hasn't changed from folder set in 3.24.1
Export all: Export fails, no error message, app returns to settings menu 

**To Reproduce**
1. set export folder
2. enable auto export
3. start tracking & wait for GPS fix
4. Stop tracking

If applicable:
* ~~add screenshots~~
* very new to android, logcat is massive, not sure if I did it correctly, happy to send via e-mail

**To Reproduce**
1. set export folder (sd card OR internal, doesn't matter)
2. have multiple tracks
3. export all

If applicable:
* ~~add screenshots~~
* very new to android, logcat is massive, not sure if I did it correctly, happy to send via e-mail

**Technical information**
 - Device: Xiaomi Redmi 9C
 - OS: Android 10 QP1A-190711.020
 - MIUI 12.0.11 Stable
 - OpenTracks version: [e.g. v3.25.0]

**Additional Info:**
I did a downgrade to 3.24.1:
  Autoexport set to the same folder again works flawlessly
  Export All works flawlessly
After updating the App tp 3.25.0 again (post downgrade):
  Autoexport fails
  Export All fails.

Multiexport by selecting multiple Tracks and Exporting via the Share button (in my case then cx explorer) remains the only option to export the tracks to device

I'm happy to answer any more question, but I'll be running 3.24.1 for the time being.
If you'd like to have a look at the logcats (which I genuinely do not know whether I did it correct thus, whether they contain the necessary info) let me know and I'll e-mail them to you.

pstorch, 2022-02-26:
This will be fixed by https://github.com/OpenTracksApp/OpenTracks/pull/1136

Der-Flip, 2022-02-26:
Oh okay.
I read that ticket, and was about to reply to it rather than open a new one, but as it was very reliably failing to export (compared to #1080 failing just occasionally), the "export all" was failing too, both of which worked perfectly fine in 3.24.1 and 3.24.0 and with the log reading quite a bit different I decided to open a new one.
My apologies and thanks for the effort!

pstorch, 2022-02-26:
@Der-Flip no worries, there are two different issues. But my PR contains a fix for the issue you discovered, because I stumbled upon it while investigating the older issue.

pstorch, 2022-02-27:
@Der-Flip a fix is in our nightly build. Would be great if you can test it, if it fully fixes this issue.

dennisguse, 2022-04-02:
@Der-Flip Please re-open if the issues is not fixed for you.
Fix was released in [v3.25.1](https://github.com/OpenTracksApp/OpenTracks/releases/tag/v3.25.1)

closed 2022-04-02 by dennisguse

Der-Flip, 2022-04-10:
hi, I apologize for the delay... didn't notice there had been an update and github didn'*t send notifications either, weirdly enough.
Export does indeed work as intended.

dennisguse, 2022-04-10:
@Der-Flip :metal: