OpenTracksApp/OpenTracks #1142: Post-workout export failed, please check the export directory 🤪

vsevjednom-cz, 2022-02-28:
**Describe the bug**
I see the error about export after stop tracking. GPX is exported ok. It's weird.

**To Reproduce**
1. Enable automatic GPX export in settings.
2. Start
3. Stop
4. See error

If applicable:
There isn't error in Run log but maybe this helps:
```
W/....debug: type=1400 audit(0.0:114247): avc: granted { write } for comm=4173796E635461736B202331 path=2F73746F726167652F656D756C617465642F302F4A6564752F31383530303139385F54726173612036362E677078 dev="sdcardfs" ino=376252 scontext=u:r:untrusted_app:s0:c247,c256,c512,c768 tcontext=u:object_r:sdcardfs:s0 tclass=file
D/ActivityThread: Remove activity client record, r= ActivityRecord{667d0b1 token=android.os.BinderProxy@21dd261 {....debug/de.dennisguse.opentracks.TrackRecordingActivity}} token= android.os.BinderProxy@21dd261
W/....debug: type=1400 audit(0.0:114248): avc: granted { write } for comm=4173796E635461736B202331 path=2F73746F726167652F656D756C617465642F302F4A6564752F31383530303139385F54726173612036362E677078 dev="sdcardfs" ino=376252 scontext=u:r:untrusted_app:s0:c247,c256,c512,c768 tcontext=u:object_r:sdcardfs:s0 tclass=file
W/....debug: type=1400 audit(0.0:114249): avc: granted { write } for comm=4173796E635461736B202331 path=2F73746F726167652F656D756C617465642F302F4A6564752F31383530303139385F54726173612036362E677078 dev="sdcardfs" ino=376252 scontext=u:r:untrusted_app:s0:c247,c256,c512,c768 tcontext=u:object_r:sdcardfs:s0 tclass=file
W/....debug: type=1400 audit(0.0:114250): avc: granted { write } for comm=4173796E635461736B202331 path=2F73746F726167652F656D756C617465642F302F4A6564752F31383530303139385F54726173612036362E677078 dev="sdcardfs" ino=376252 scontext=u:r:untrusted_app:s0:c247,c256,c512,c768 tcontext=u:object_r:sdcardfs:s0 tclass=file
```

**Technical information**
 - Device: Honor 8S KSA-LX9
 - OS: Android 9
 - OpenTracks version: v3.25.0

pstorch, 2022-03-01:
Interesting. Isn't the StrictMode only enabled in DEBUG builds? @vsevjednom-cz how did you capture the log? From the release version or from a selfbuild debug version?

pstorch, 2022-03-01:
The error description sounds like #1141, @vsevjednom-cz can you check the nightly build?

vsevjednom-cz, 2022-03-02:
@pstorch: hmm, it's weird. I didn't do anything but problem with export is away 🤪. I'll let you know if the problem comes back.

vsevjednom-cz, 2022-03-05:
The problem is really weird - sometimes export crashes, sometimes not.But it looks like something about export is fixed in version 3.25.1. I started using this version. I'll let you know.

pstorch, 2022-03-05:
It's still trial and error. I have problems with Nextcloud directories also with other Apps (e.g. KeePassDX).

vsevjednom-cz, 2022-03-07:
Everything works without any problem in v3.25.1 🥳.

vsevjednom-cz, 2022-03-08:
Everything is ok in `3.25.1`

closed 2022-03-08 by vsevjednom-cz