OpenTracksApp/OpenTracks #1146: osm dashboard not an option to view map and track from open tracks

nutpantz, 2022-03-04:


dennisguse, 2022-03-05:
I am ok, with removing this functionality. It is a leftover from before OSMDashboard times... And opening the data as a file cannot be used for live updates.

@rgmf @pstorch

pstorch, 2022-03-05:
How do we open the track in Osmand or other map app then? The share button doesn't activate them.

nutpantz, 2022-03-05:
I think it would be nice if it was a setting for the map button. I can see the use for the various options.

eliotb, 2022-05-20:
I seem to have same issue. The ONLY option I get when clicking the map button is to open using ... OpenTracks (as KML)
I have osmand installed.  
My current workaround is to set Instant export, and set the Track export directory to OSMAnd's own tracks directory. `Android/data/net.osmand.plus/files/tracks`  . Then at least I can choose the track(s) to display from OSMAnd

pstorch, 2022-05-20:
> The ONLY option I get when clicking the map button is to open using ... OpenTracks (as KML)

What options do you get next after "OpenTracks (as KML)? 
"OpenTracks (as GPX)" is not available?

eliotb, 2022-05-21:
> What options do you get next after "OpenTracks (as KML)?
> "OpenTracks (as GPX)" is not available?

The ONLY option is KML, there is no other option.

pstorch, 2022-05-21:
And what happens when you click on KML? There should be another app chooser coming next. 

Have you tried this: 
> Please go to the app info (long click on launcher icon), then Defaults/ Set as default, then clear defaults.

eliotb, 2022-05-21:
> There should be another app chooser coming next.

:bulb:  Ahhh right - thanks. There is. I never clicked because I was already using OpenTracks, and it made no sense to me that I would want to use OT to open its own track. This IS confusing I think.

So clicking the map button I get 
```
Complete action using OpenTracks
               Just once   Always
Use a different app
() OpenTracks
   Open as KML
```
Top one gives me Open as gpx, and offers OsmAnd+ or OpenTrac...
Bottom one gives me Open as kml, and offers OsmAnd+ or OpenTrac... or Magic Ea...

I'm pretty sure (quite a long time ago) I would click the map icon and go directly to OsmAnd...

pstorch, 2022-05-22:
At the moment OT itself is listening on the `Intent.OpenTracks-Dashboard` Intents.
What do you think of changing this to a custom IntentChooser with these Options:
- KMZ (with and without photos)
- KML (with and without photos)
- GPX
- OpenTracks-Dashboard

In short all supported file formats + Dashboard with the option to set a default (which should be reset-able in OT settings).
This would flatten the options to choose from and could be (re-)configured within OT.

At the moment the share button uses the export file format. I'm thinking of making this configuration independent. Because a user might want to (auto-)export all tracks as KMZ with photos, but wants to share a GPX now and then.

dennisguse, 2022-05-22:
Sounds good to me.

pstorch, 2022-05-22:
Then I'll work on a PR with a proposal for this.

PR #1253 (2022-05-30): Add "show on map" format selection dialog
Adds a format selection dialog for the "show on map" action.
A chosen format can be set as "always", so this dialog is skipped the next time.
The chosen format can be set and unset in the Preferences UI Settings.

fixes #1146
closed 2022-05-30 by pstorch