OpenTracksApp/OpenTracks #1147: Czech translation at Bubu and Czech voice annoucements 🎤

vsevjednom-cz, 2022-03-05:
I created a lot of proposals of Czech translation at [Bubu](https://weblate.bubu1.eu/engage/OpenTracks/) but I don't see any change in OpenTracks 😨 😰.

... And I can create voice annoucements in Czech language if you want 🎤. I need to know all sentenses.

dennisguse, 2022-03-05:
I just checked and there are no pending translations on Bubu for Czech.
I usually pull changes before every release.
So, they seem to be somehow gone - do you know when you did the translations?
I am sorry for your waste of time... :crying_cat_face: 

Regarding voice announcements: we just use the installed TTS and give it the strings, it should say and then it does :)

pstorch, 2022-03-06:
I've asked Bubu via Matrix chat. He can't see any irregularities. @vsevjednom-cz did you make a suggestion for a translation (which someone else needs to pick up) or an "actual" translation?

vsevjednom-cz, 2022-03-06:
Proposed change for example: https://weblate.bubu1.eu/translate/OpenTracks/opentracks/cs/?checksum=6598cdda6df89646&q=stopy&sort_by=-priority%2Cposition
or here: https://weblate.bubu1.eu/translate/OpenTracks/opentracks/cs/?checksum=2d8b907c825686fb&q=v%C3%BDvoz&sort_by=-priority%2Cposition
Do you see it?

pstorch, 2022-03-06:
Ah ok, these are two suggestions, not actual translations. Either you save them as actual translations or some other translator needs to pick them up.

vsevjednom-cz, 2022-03-06:
Yes. I wasn't logged 🥴. You can see the changes right now.

closed 2022-03-06 by vsevjednom-cz