OpenTracksApp/OpenTracks #1149: No GPS location on LineageOS 18.1

creideiki, 2022-03-06:
**Describe the bug**
After upgrading to LineageOS 18.1 (Android 11), OpenTracks never receives a GPS location. There is not even a location icon in the notifications bar, which suggests to me that the OS doesn't think that OpenTracks has asked for a location. OpenTracks worked before the upgrade, on LineageOS 16 (Android 9). I have removed and reinstalled OpenTracks, which did not help.

The only location provider I have is the onboard GPS. I am running a bare LineageOS ROM, without microG, Mozilla Location Service, or any other network location provider. This was also the case when the phone ran LineageOS 16 (Android 9).

Other applications, e.g. OsmAnd and StreetComplete, manage to get a GPS location, and while they are running, the location icon appears in the notification bar.

Permissions on OpenTracks look right: `ACCESS_COARSE_LOCATION` and `ACCESS_FINE_LOCATION` are both granted. The only denied permissions on OpenTracks are `BLUETOOTH_SCAN` and `BLUETOOTH_CONNECT`.

`adb logcat` output is not particularly helpful. Here's everything from trying to start recording a track until the last mention of either "GPS", "GNSS" or "location":
```
03-06 17:41:55.967   671 19955 I ActivityTaskManager: START u0 {flg=0x14000000 cmp=de.dennisguse.opentracks/.TrackRecordingActivity} from uid 10204
03-06 17:41:55.989   671 19955 W ActivityTaskManager: Tried to set launchTime (0) < mLastActivityLaunchTime (82689784)
03-06 17:41:56.135  1761 19503 D KeyguardViewMediator: setOccluded true
03-06 17:41:56.135 19756 19756 I TrackRecordingServiceConnection: Binding the service.
03-06 17:41:56.136  1761  1761 D KeyguardViewMediator: adjustStatusBarLocked: mShowing=false mOccluded=true isSecure=true force=false --> flags=0x0
03-06 17:41:56.141 19756 19756 I TrackRecordingServiceConnection: Starting the service.
03-06 17:41:56.143 19756 19756 I TrackRecordingServiceConnection: Binding the service.
03-06 17:41:56.144 19756 19756 I TrackRecordingServiceConnection: Binding the service.
03-06 17:41:56.153 19756 19756 I TrackRecordingService: new status null -> RecordingStatus{trackId=null, paused=false}
03-06 17:41:56.181 19756 19756 I TrackRecordingServiceConnection: Binding the service.
03-06 17:41:56.198 19756 19756 I TrackRecordingServiceConnection: Connected to the service.
03-06 17:41:56.200 19756 19756 I chatty  : uid=10204(de.dennisguse.opentracks) identical 1 line
03-06 17:41:56.201 19756 19756 I TrackRecordingServiceConnection: Connected to the service.
03-06 17:41:56.204   671   696 I ActivityTaskManager: Displayed de.dennisguse.opentracks/.TrackRecordingActivity: +237ms
03-06 17:41:56.286 19756 19756 I TrackRecordingService: new status RecordingStatus{trackId=null, paused=false} -> RecordingStatus{trackId=5, paused=false}
03-06 17:41:56.287 19756 19756 I SystemUtils: Acquiring wake lock.
03-06 17:41:56.291 19756 19756 E GPSHandler: Not started.
03-06 17:41:56.298   575   606 E LocSvc_APIClientBase: onResponseCb:894] ERROR: 5 ID: 40
03-06 17:41:56.301 19756 19756 W BluetoothRemoteSensorManager: Bluetooth not enabled.
03-06 17:41:56.302 19756 19756 I chatty  : uid=10204(de.dennisguse.opentracks) identical 3 lines
03-06 17:41:56.302 19756 19756 W BluetoothRemoteSensorManager: Bluetooth not enabled.
03-06 17:41:56.309 19756 19756 W AltitudeSumManager: No pressure sensor available.
03-06 17:41:56.309 19756 19756 D AltitudeSumManager: Reset
03-06 17:41:56.323 19756 19756 D VoiceAnnouncement: Start
03-06 17:41:56.325   671   688 I GnssLocationProvider: GNSS HAL Requesting location updates from network provider for 10000 millis.
03-06 17:41:56.326 19756 19756 I VoiceAnnouncement: TextToSpeech initialized with status -1
03-06 17:41:56.326   671   688 W GnssLocationProvider: Unable to request location.
03-06 17:41:56.326   671   688 W GnssLocationProvider: java.lang.IllegalArgumentException: provider doesn't exist: network
03-06 17:41:56.326   671   688 W GnssLocationProvider:  at com.android.server.location.LocationManagerService.requestLocationUpdatesLocked(LocationManagerService.java:1964)
03-06 17:41:56.326   671   688 W GnssLocationProvider:  at com.android.server.location.LocationManagerService.requestLocationUpdates(LocationManagerService.java:1947)
03-06 17:41:56.326   671   688 W GnssLocationProvider:  at android.location.LocationManager.requestLocationUpdates(LocationManager.java:1178)
03-06 17:41:56.326   671   688 W GnssLocationProvider:  at android.location.LocationManager.requestLocationUpdates(LocationManager.java:1139)
03-06 17:41:56.326   671   688 W GnssLocationProvider:  at com.android.server.location.gnss.GnssLocationProvider.handleRequestLocation(GnssLocationProvider.java:828)
03-06 17:41:56.326   671   688 W GnssLocationProvider:  at com.android.server.location.gnss.GnssLocationProvider.access$2700(GnssLocationProvider.java:100)
03-06 17:41:56.326   671   688 W GnssLocationProvider:  at com.android.server.location.gnss.GnssLocationProvider$ProviderHandler.handleMessage(GnssLocationProvider.java:2079)
03-06 17:41:56.326   671   688 W GnssLocationProvider:  at android.os.Handler.dispatchMessage(Handler.java:106)
03-06 17:41:56.326   671   688 W GnssLocationProvider:  at android.os.Looper.loop(Looper.java:223)
03-06 17:41:56.326   671   688 W GnssLocationProvider:  at android.os.HandlerThread.run(HandlerThread.java:67)
03-06 17:41:56.326   671   688 W GnssLocationProvider:  at com.android.server.ServiceThread.run(ServiceThread.java:44)
```
The Java exception saying "provider doesn't exist: network" looks suspect, but it appears when running OsmAnd as well, and that app manages to get a GPS location.

**Technical information**
 - Device: Moto G5 Plus (XT1685)
 - OS: LineageOS 18.1
 - OpenTracks version: 3.25.1, from F-Droid

dennisguse, 2022-03-07:
No clue.
Relevant code is in `GPSHandler.registerLocationListener()`.

As of 3.25.0, we allow to start recordings __without__ GPS (i.e., BLE sensor only use cases like indoor cycling).
So, it might be that the permission check is problematic.
And we are using: `LocationManager.GPS_PROVIDER` - don't know where the network is coming from.

Could you debug it and see what is coming up?

creideiki, 2022-03-07:
> Could you debug it and see what is coming up?

Maybe, if you could be more explicit in what you want me to do. I am not an Android developer, `adb logcat` is pretty much the limit of my knowledge, and I have no Android app development environment available.

dennisguse, 2022-03-11:
1. Install AndroidStudio
2. Enable development mode on your phone
3. Clone repo
4. Build and install OpenTracks on the phone
5. Start OpenTracks in debug mode via AndroidStudio and set breakpoints...

creideiki, 2022-03-19:
So, that took way more work than expected, but I think I now know what happens, even if I don't know why.

Setting breakpoints in OpenTracks' `GPSHandler` showed me that `registerLocationListener()` was called, which in turn called `locationManager.requestLocationUpdates()`. Toggling location globally called `onProviderEnabled()` and `onProviderDisabled()`, so the location listener has definitely been registered, but `onLocationChanged()` which is supposed to actually receive location updates was never called.

To have a chance of figuring out what was supposed to happen, I looked at the simplest GPS application I could find and easily build, which turned out to be [GPS Cockpit](https://www.f-droid.org/en/packages/org.woheller69.gpscockpit/). Sprinkling its source with breakpoints, I found that the location icon in the phone's status bar (the pin) appeared after the app registered a [GNSS status callback](https://developer.android.com/reference/android/location/GnssStatus.Callback) (at [this line in the GPS Cockpit source code](https://github.com/woheller69/gpscockpit/blob/f77664e8d28e8e099615374b58091e5d66077e18/app/src/main/java/org/woheller69/gpscockpit/GpsSvc.java#L175)).

Blindly copying the code that registers the GNSS status callback from GPS Cockpit into OpenTracks gave me a `GPSHandler.registerLocationListener()` that looks like this:
```diff
--- a/src/main/java/de/dennisguse/opentracks/services/handlers/GPSHandler.java
+++ b/src/main/java/de/dennisguse/opentracks/services/handlers/GPSHandler.java
@@ -2,6 +2,7 @@ package de.dennisguse.opentracks.services.handlers;
 
 import android.content.Context;
 import android.content.SharedPreferences;
+import android.location.GnssStatus;
 import android.location.Location;
 import android.location.LocationListener;
 import android.location.LocationManager;
@@ -148,6 +149,8 @@ public class GPSHandler implements LocationListener, GpsStatus.GpsStatusListener
         }
     }
 
+    private GnssStatus.Callback mGnssStatusCallback;
+
     private void registerLocationListener() {
         if (locationManager == null || context == null) {
             Log.e(TAG, "Not started.");
@@ -156,6 +159,13 @@ public class GPSHandler implements LocationListener, GpsStatus.GpsStatusListener
         if (PermissionUtils.hasGPSPermission(context)) {
             try {
                 locationManager.requestLocationUpdates(LocationManager.GPS_PROVIDER, gpsInterval.toMillis(), 0, this);
+                mGnssStatusCallback = new GnssStatus.Callback() {
+                    @Override
+                    public void onSatelliteStatusChanged(@NonNull GnssStatus status) {
+                        super.onSatelliteStatusChanged(status);
+                    }
+                };
+                locationManager.registerGnssStatusCallback(mGnssStatusCallback);
             } catch (SecurityException e) {
                 Log.e(TAG, "Could not register location listener; permissions not granted.", e);
             }

```

And with that code added, I see the location icon in the phone's status bar, and OpenTracks starts writing trackpoints.

However, I cannot find any documentation stating that you have to use both `requestLocationUpdates()` and `registerGnssStatusCallback()` to get a location. The few articles that do talk about that combination, like [this StackOverflow question](https://stackoverflow.com/questions/42890585/gnssstatus-callback-onsatellitestatuschanged-not-working), talk about using the GNSS status callback to get detailed satellite information, which an app that just wants a location shouldn't care about.

Looking at some other GPS-using apps, [OsmAnd](https://github.com/osmandapp/OsmAnd) (which works for me) [calls `registerGnssStatusCallback()`](https://github.com/osmandapp/OsmAnd/blob/e3aa8c8ba1e2f7a4822a0c591a4c2c320235b4b8/OsmAnd/src/net/osmand/plus/OsmAndLocationProvider.java#L339). [StreetComplete](https://github.com/streetcomplete/StreetComplete) also works for me even though I cannot find any call to `registerGnssStatusCallback()`, but it's written in Kotlin of which I know even less than Java, so I am probably missing something.

Also, I have no idea how horrible my cargo-culted Java code is. It definitely leaks a callback object every time it's called. Hopefully, someone who actually knows Java can turn it into something productive.

pstorch, 2022-03-19:
That's very interesting.

dennisguse, 2022-03-19:
@creideiki nice investigation! 
The Gnss callback can for example be used to access NMEA data and therefore get additional data directly from the GPS without passing it through the driver.

My gut feeling tells me that is a problem with the ROM you are using. Both callbacks are separate and can be accessed individually - although I would prefer one callback only as both provide slightly different data. 

PS/ the Android lifecycle makes the callback hell even more complicated...

dennisguse, 2022-04-02:
@creideiki Any more insights / suggestions on this?
I don't know what we can do about this...

But I received another similar report (also LineageOS 18.1, Poco F3) report via email.

creideiki, 2022-04-02:
I have not looked any further. I was hoping that you would be willing to register a GNSS status callback that doesn't do anything, since that seems to be enough to make it work for me. There would be no need to use the (potentially different) data from the callback, and if it is clearly marked in the code as a bug workaround for LineageOS 18.1 the maintenance burden should be minimal.

dennisguse, 2022-04-02:
@creideiki did you report this to LineageOS?
Then we can add a link to the workaround + we have a place to check if this is still needed.

It would be a plus, if we can determine that we are running on a LineageOS device using Android APIs. I would consider this optional.

creideiki, 2022-04-03:
Unfortunately, my Motorola G5 Plus is not officially supported by LineageOS, so I am running an unofficial build which they [explicitly refuse bug reports from](https://wiki.lineageos.org/how-to/bugreport). However, [the Xiaomi POCO F3 is supported](https://wiki.lineageos.org/devices/alioth/), so if you could get confirmation from that user that their problem is identical I could refer to that device instead.

dennisguse, 2022-04-03:
@creideiki Rather unlikely - it was just an email some days ago and I replied, but have not heard anything back. :shrug: 

Some options (in order of my personal preference):
1. What about putting your hack into a separate branch and you rebase/ build yourself every now and then?
2. Adding a workaround only for LineageOS or even your variant for 6 month (might be helpful: https://developer.android.com/reference/android/os/Build )
3. Adding the workaround for 6 month
  (but we would need to know if there are any ugly side effects before rolling it out)

creideiki, 2022-04-04:
I'll try to take a look at implementing option 2.

dennisguse, 2022-04-27:
Then let's hope that LineageOS is fixing it...

closed 2022-04-27 by dennisguse