OpenTracksApp/OpenTracks #1156: Recording stopped when pressing Map button

mulm, 2022-03-14:
**Describe the bug**
Recording stopped when pressing Map button.  This has occurred twice for me.  Each time, no apparent crash; able to start a new recording and use Map button on the subsequent recording.

*Not* regularly reproducible (tried just now, problem did not occur).

**To Reproduce**
1. Start recording
2. Click on Map button
3. App jumps back to the track list (home?)

No screenshot or logcat output at this time, but happy to try to capture.  Also, happy to try an instrumented version of app.

**Technical information**
 - Device: Samsung S9
 - OS: Android 10
 - OpenTracks version: v3.25.1, Version code: 4755

dennisguse, 2022-03-14:
Sounds like an issue in the state handling of StatisticsRecordingFragment likely a nullpointer. And it is probably affected by the UI update that we trigger every second (and that's why it does not occur everytime).

 If you can get a logcat that would be pretty helpful.

dennisguse, 2022-03-21:
Got an NPE reported via PlayStore that might fit:
```
java.lang.NullPointerException: 
  at de.dennisguse.opentracks.data.ContentProviderUtils.getTrack (ContentProviderUtils.java:234)
  at de.dennisguse.opentracks.ui.intervals.IntervalsFragment.onResume (IntervalsFragment.java:135)
  at androidx.fragment.app.Fragment.performResume (Fragment.java:3039)
  at androidx.fragment.app.FragmentStateManager.resume (FragmentStateManager.java:607)
  at androidx.fragment.app.FragmentStateManager.moveToExpectedState (FragmentStateManager.java:306)
  at androidx.fragment.app.FragmentManager.executeOpsTogether (FragmentManager.java:2189)
  at androidx.fragment.app.FragmentManager.removeRedundantOperationsAndExecute (FragmentManager.java:2100)
  at androidx.fragment.app.FragmentManager.execSingleAction (FragmentManager.java:1971)
  at androidx.fragment.app.BackStackRecord.commitNow (BackStackRecord.java:305)
  at androidx.viewpager2.adapter.FragmentStateAdapter$FragmentMaxLifecycleEnforcer.updateFragmentMaxLifecycle (FragmentStateAdapter.java:726)
  at androidx.viewpager2.adapter.FragmentStateAdapter$FragmentMaxLifecycleEnforcer$1.onPageScrollStateChanged (FragmentStateAdapter.java:632)
  at androidx.viewpager2.widget.CompositeOnPageChangeCallback.onPageScrollStateChanged (CompositeOnPageChangeCallback.java:87)
  at androidx.viewpager2.widget.CompositeOnPageChangeCallback.onPageScrollStateChanged (CompositeOnPageChangeCallback.java:87)
  at androidx.viewpager2.widget.ScrollEventAdapter.dispatchStateChanged (ScrollEventAdapter.java:426)
  at androidx.viewpager2.widget.ScrollEventAdapter.onScrolled (ScrollEventAdapter.java:214)
  at androidx.recyclerview.widget.RecyclerView.dispatchOnScrolled (RecyclerView.java:5173)
  at androidx.recyclerview.widget.RecyclerView$ViewFlinger.run (RecyclerView.java:5338)
  at android.view.Choreographer$CallbackRecord.run (Choreographer.java:1030)
  at android.view.Choreographer.doCallbacks (Choreographer.java:853)
  at android.view.Choreographer.doFrame (Choreographer.java:784)
  at android.view.Choreographer$FrameDisplayEventReceiver.run (Choreographer.java:1015)
  at android.os.Handler.handleCallback (Handler.java:883)
  at android.os.Handler.dispatchMessage (Handler.java:100)
  at android.os.Looper.loop (Looper.java:214)
  at android.app.ActivityThread.main (ActivityThread.java:7397)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:492)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:935)
```

dennisguse, 2022-04-02:
Somehow we lose the trackId in TrackRecordingActivity.
The problem is that this activity also starts the recording, which makes way to complicated.
Will be fixed when implementing #1103 as it must be refactored anyhow.

dennisguse, 2022-04-24:
How to recreate the issue reliably:
1. Start a recording
2. Close activity (not only move it to the background, but really swipe it away)
3. Open via notification: opens TrackRecordingActivity
4. Go to interval tab: crash

The problem is that the 2nd time TrackRecordingActivity is opened, it lost the trackId.
And since it doesn't have it anymore all operations were the trackId needs to be passed will crash (e.g.., interval tab, map, share)

We try to persist the trackId in a savedInstance, but that is not restored as it is a different (aka new) TrackRecordingActivity.

Mindblown.
@pstorch @rgmf FYI

dennisguse, 2022-04-24:
For now, fixed by opening TrackListActivity instead of TrackRecordingActivity from the notification.
Then passing the trackId works as expected.

closed 2022-04-24 by dennisguse

rgmf, 2022-04-24:
> For now, fixed by opening TrackListActivity instead of TrackRecordingActivity from the notification. Then passing the trackId works as expected.

It sounds like a good idea, because TrackListActivity is the "home" Activity, so when user opens OpenTracks from Notification is seems good opening TrackListActivity.

Anyway, I prefer OpenTracks opens TrackRecordingActivity: what about save trackId into preferences? It was removed but I don't know if there are other options to retain trackId in this situation.

dennisguse, 2022-04-24:
We just need to pass via the intent from TrackRecordingServiceNotificationManager, but it requires some refactoring.

commit 6741ef4d (2022-04-24): Bugfix: notification opens TrackListActivity instead of TrackRecordingActivity.

If a recording is running, we need to pass the trackId to TrackRecordingActivity as otherwise it crashes when needing it.

Fixes #1156.