OpenTracksApp/OpenTracks #1159: Current HR voice announcements seem stuck to the same value regardless of real HR

serpico, 2022-03-18:
**Describe the bug**
Installed v3.26 from F-Droid this morning to test the HR voice announcement feature.

Time interval set to 1 min - Distance interval off - Report heart rate ticked/on

With the hiking activity during the first kilometer only Distance, Pace and **Average HR** are announced.

After the first kilometer **Current HR** is also announced but seems to be stuck, announcing the same value every minute and doesn't reflect the HR displayed on the User Interface/increasing-decreasing of the real HR.

I made 3 recordings back to back ( start, walk 1 km, listen to the announcements for the 3-5 minutes following the completion of the first kilometer while increasing/reducing the pace to make HR vary, stop the recording, then start another recording ) it was the same behavior every time.

What's the sample for the value **Current Heart Rate** ? 

I imagine **Average HR** is calculated since the start of the activity/recording.

I was hoping to use Open Tracks to replace an old version of Wahoo Fitness which gives me my average HR ( averaged during the last minute ) every minute while cycling, using earplugs, using this to increase or reduce my pace.

Thanks @spacecowboy for implementing that feature.

**Technical information**
 - Device: Samsung SM-G361F
 - OS: Android 5.1.1 
 - OpenTracks version: v3.2.6 from F-Droid

spacecowboy, 2022-03-18:
current heart rate is a bit misleading. it is the "lap heart rate". so average during the latest 1 km intervall

serpico, 2022-03-18:
> current heart rate is a bit misleading. it is the "lap heart rate". so average during the latest 1 km interval

Thanks for the explanation, any plan to change that ? i.e in line with the options currently offered in Setting>Voice Announcements

* Time interval ( 1, 2, 5 min etc and so on --> averaged over the last 1, 2, 5 min etc )
* Distance interval ( 1, 5, 10 km etc   --> averaged over the last 1, 5, 10 km etc )

Maybe the present behavior is best suited with your and the majority of user's expectation...

spacecowboy, 2022-03-18:
the implementation matches the behavior for lap pace. the voice announcements are currently hard coded to per kilometer / per mile statistics.

it would be up to @dennisguse if this should be changed. I wouldn't mind contributing to make the change

dennisguse, 2022-03-18:
Just go ahead - the belongs as much to you as to me.
Have fun!

I will just complain if too much complexity is added and then we can discuss if we can make it nicer ;)

dennisguse, 2022-06-20:
@serpico is this solved for you?