OpenTracksApp/OpenTracks #1161: distance is not being reported as cumulative distance in GPX files

brianjmurrell, 2022-03-22:
According to 
https://github.com/OpenTracksApp/OpenTracks/blob/0eda1eed1dd4516caf724a701f6941bcac4e3c90/src/main/java/de/dennisguse/opentracks/io/file/exporter/GPXTrackExporter.java#L357 it seems that the `distance` tag is supposed to be _cumulative_ distance.  In looking in some of the OpenTracks GPX export files though, it doesn't appear to be cumulative:

```
$ grep distance example.gpx
            <opentracks:distance>9.6</opentracks:distance>
            <opentracks:distance>2.2</opentracks:distance>
            <opentracks:distance>9.7</opentracks:distance>
            <opentracks:distance>2.9</opentracks:distance>
            <opentracks:distance>11.8</opentracks:distance>
            <opentracks:distance>9</opentracks:distance>
            <opentracks:distance>1.4</opentracks:distance>
            <opentracks:distance>10.4</opentracks:distance>
            <opentracks:distance>11.1</opentracks:distance>
            <opentracks:distance>9.6</opentracks:distance>
            <opentracks:distance>1.5</opentracks:distance>
            <opentracks:distance>9.6</opentracks:distance>
            <opentracks:distance>0.8</opentracks:distance>
            <opentracks:distance>11.1</opentracks:distance>
            <opentracks:distance>9.6</opentracks:distance>
            <opentracks:distance>1.5</opentracks:distance>
            <opentracks:distance>10.2</opentracks:distance>
```

While you are at fixing this, any chance of implementing #1160?

dennisguse, 2022-03-22:
That's just bad naming for the variable.
It is the cumulative distance of previous TrackPoints that cannot be exported to GPX, because they don't have coordinates.
So, it is some kind of local cumulative.
Will fix the naming.

brianjmurrell, 2022-03-22:
So the `<opentracks:distance>` is not the distance from one track point to another, as a more accurate alternative to GPS coordinates?  I.e. such that the total of all of the distance values is the total of the distance covered in the activity.

dennisguse, 2022-03-22:
It is exactly this.
But we also have TrackPoints that cannot be exported and those values are added to the next TrackPoint before exporting.
As a side effect, GPX does not allow to restore a track exactly in OpenTracks as these points are not exported.

For adding #1060: PR welcome ;)

brianjmurrell, 2022-03-22:
> PR welcome

Touché.  I have not even opened the code-base yet, beyond just searching for the `opentracks:distance` tag and finding the above.

I suppose I was hoping in fixing this issue, that for somebody who knows the code well, adding the cluetrust `distance` tag (as a cumulative distance of the whole track, not just the current track point) would be child's play.

I will see when I can find some time to study the code enough to be able to make that addition.

dennisguse, 2022-03-23:
You found exactly the right place where to add it. I addition to the actual data, the xsd must be referenced in the header.

brianjmurrell, 2022-03-24:
I figured.  But before I can do anything there, I have to get an Android development environment set up here and working.  Making the code change is relatively trivial.  Being able to build and test it is less so, not being an Android developer.

That said, the question arises as to where this cumulative distance should be accumulated into from `cumulativeDistance`.  ` writeTrackPoint()` only has access to that single track point data.  So we could jump up to it's caller ` writeTrackPoints()` and create a variable there an accumulate the distance, but that function doesn't have `cumulateSensorData()` (which I've yet to figure out the actual function of -- I'm not a Java coder so the syntax of that function is quite foreign to be yet) to make a distance out of the track point.

One hack would be to have `writeTrackPoints()` return `cumulativeDistance` to it's caller to be accumulated and then passed to to the next `writeTrackPoint()` to written to the GPX file.  But that does seem an ugly hack indeed.

All of this might also be revealing my ignorance with this code base though.

brianjmurrell, 2022-03-27:
How is <opentracks:distance> calculated when both GPS is available as well as distance from a sensor such as the Tickr X?

dennisguse, 2022-03-27:
BLE sensor data supersedes GPS-based data (i.e. speed and distance) as those sensor provide a higher precision.

dennisguse, 2022-04-02:
@brianjmurrell via the export and import functionality in the settings.

dennisguse, 2022-04-08:
Fixed

closed 2022-04-08 by dennisguse