OpenTracksApp/OpenTracks #1163: Issues with French announcements

solfred, 2022-03-25:
Vocal announcements in French have 2 issues:

    Text is in English: "Lap speed" instead of "vitesse au tour"
    Only hours are given now, not the minutes. Example: Lap speed: 11km/h instead of 11,5km/h

OpenTrack version; 3.26.0
Device: Sony M2
Android 5.1.1

dennisguse, 2022-03-25:
English is the default fallback if there is no translation of a certain text.
You can just head over to https://weblate.bubu1.eu/engage/OpenTracks/ and add the missing french ones.
They will be included in the next release.

About the missing minutes: I can take a look at the beginning of next month.

solfred, 2022-03-28:
Hi, 

French Translation was previously OK (I use OpenTracks for 2 years now and no issue before on this translation). The translation level seems ok too on translation website. 

The 2 issues (lost of French translation in voice annoucement and missing minutes) appeared in the same release. 

Regards

dennisguse, 2022-05-03:
@ristello Thanks for taking a look at this!
And you are right, we are "just" missing translations - it is kind of an annoying task to adjust them manually especially if you cannot read them...
In this commit, @spacecowboy split up the stings, so `TtsSpan.MeasureBuilder` could be used to enable the TTS be nicer.

spacecowboy, 2022-05-04:
Yes. It was unfortunately necessary to split the strings up in order to use the TTS Spans.

About it being reported as `11km/h` instead of `11.5km/h`, that may be a rounding that I decided on. Or it is the TTS Engine making that decision. Not sure.

dennisguse, 2022-06-11:
Will be fixed, if somebody translates the strings on Weblate.
Might already be fixed; didn't check.
It is not an issue in the code.

@spacecowboy Still nice work ;)

closed 2022-06-11 by dennisguse