OpenTracksApp/OpenTracks #1171: Display heart rate from BLE before starting track

brianjmurrell, 2022-04-06:
**Is your feature request related to a problem? Please describe.**
It would be nice if one could see the HR from one's BLE sensor prior to starting a track.  I just did a walk and the sensor failed to be found and I got no sensor readings for the whole walk.  This is because it seems to take a short while after starting a track (i.e. starting reading the sensor) to get data, so it would be nice to get that wait over with before starting the track and knowing that it's reading.

**Describe the solution you'd like**
Some kind of display of the HR before pressing start [track].

dennisguse, 2022-04-07:
One workaround: before starting a recording (in the track list), use the GPS button in the top menu.
It starts the GPS and sensor connection.
However, the status it represents is only GPS.

dennisguse, 2022-04-07:
And we could try to put this information into the Foreground notification - that would require no further UI changes.

brianjmurrell, 2022-05-10:
> And we could try to put this information into the Foreground notification - that would require no further UI changes.

But you don't get that notification until the track is started, right?  I'm referring to the notification that that has speed and distance in it.

Because the whole point here is being able to see that OpenTracks can see a signal from the HRM before starting the track.

There is a notification that comes on when starting GPS but I don't want GPS on when I am doing an indoor activity.

dennisguse, 2022-05-10:
Implementing #240 might allow us to solve this, right?
Something like: wait until this sensor is present.

brianjmurrell, 2022-05-11:
Yeah, something like #240 could work where the delay/count-down is effectively just the wait for the BLE sensors to be present.

I would further suggest that the track be stoppable during this count-down/sensor-wait and if it is stopped in that period it is automatically deleted (if it's even created by that time).

Honestly, it sometimes (usually after a phone reboot -- which happens weekly to apply LineageOS updates) takes a bit of effort on my phone to get a working BLE signal from my HRM, such as toggling BT, trying with the HRM vendor's own app, etc.  I have not really figured out the magic incantation yet.  Even the HRM vendor's own app can have difficulty finding the HRM without all of the fiddling.  The one benefit their app has is that it shows more detail about the HRM so seeing when it starts working is more clear.

dennisguse, 2022-05-13:
@brianjmurrell I know your pain - my cycling sensors also sometimes just don't want to play nicely with my phone....