OpenTracksApp/OpenTracks #1174: Autopause annoucement 🎧

vsevjednom-cz, 2022-04-08:
**Is your feature request related to a problem? Please describe.**
Is possible to set announcement of autopause starts and autopause ends? I've used it few years ago in Sports Tracker app. I wrote about it [here](https://github.com/OpenTracksApp/OpenTracks/issues/862#issuecomment-915034592) earlier. This will be turned off by default of course.

**Describe the solution you'd like**
Autopause on: sound announcement.
Autopause off: another sound announcement.

**Additional context**
It's weird thing, I know. But when I'll track: I can be sure app works when I stop at traffic lights for example (announcement every kilometer is really amazing but this is another clever thing).

dennisguse, 2022-04-08:
Good idea: will also help us to understand the actual impact of the moving time detection (aka autopause).
Some discussion here: https://github.com/OpenTracksApp/OpenTracks/issues/968

Approach: this information is determined in the TrackStatistics/ TrackStatisticsUpdater.
When `TrackRcordingService.newTrackPoint()` the statistics will be updated and afterwards, we could directly trigger the announcement if a change took place.

Something like
```
    public boolean newTrackPoint(TrackPoint trackPoint, Distance thresholdHorizontalAccuracy) {
        if (!isRecording() || isPaused()) {
            Log.w(TAG, "Ignore newTrackPoint. Not recording or paused.");
            return false;
        }

        boolean stored = trackRecordingManager.onNewTrackPoint(trackPoint); // Return if idle/non-idle changed
        notificationManager.updateTrackPoint(this, trackRecordingManager.getTrackStatistics(), trackPoint, thresholdHorizontalAccuracy);

       voiceAnnouncementManager.announceIdleChanged(); //TODO :)

        return stored;
    }
```

PS/ PR welcome ;)

dennisguse, 2022-04-10:
There is an easier solution: VoiceAnnouncement is already triggered by the 1s UI update.
https://github.com/OpenTracksApp/OpenTracks/blob/dd1c5667a0e5c54f55a3231de035213e733ece8e/src/main/java/de/dennisguse/opentracks/services/TrackRecordingService.java#L378

So all is in place: we just need to trigger the message in VoiceAnnouncement manager.

dennisguse, 2023-08-22:
Idle detection is now almost finished #1652.
We need to change a tiny detail: the idle announcement has to be triggered in `TrackRecordingService.newTrackPoint()` as `idle` is only present then (not during the time updates).

closed 2023-09-23 by dennisguse