OpenTracksApp/OpenTracks #1176: java.lang.NullPointerException: Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference

brianjmurrell, 2022-04-09:
It seems like this code:

https://github.com/OpenTracksApp/OpenTracks/blob/0f489f1cc9f7bf0a3f21a83ec9b1434aa94c7182/src/main/java/de/dennisguse/opentracks/io/file/exporter/GPXTrackExporter.java#L361-L365

used to expect that `cumulateSensorData()` could return `null` which was handed by line 362.  But now it's being converted directly to a `Distance` which I suspect is causing:

```
04-09 14:28:48.193 25218 27877 E AndroidRuntime: Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'double java.lang.Double.doubleValue()' on a null object reference
04-09 14:28:48.193 25218 27877 E AndroidRuntime:        at de.dennisguse.opentracks.io.file.exporter.GPXTrackExporter.writeTrackPoint(GPXTrackExporter.java:361)
04-09 14:28:48.193 25218 27877 E AndroidRuntime:        at de.dennisguse.opentracks.io.file.exporter.GPXTrackExporter.writeTrackPoints(GPXTrackExporter.java:177)
04-09 14:28:48.193 25218 27877 E AndroidRuntime:        at de.dennisguse.opentracks.io.file.exporter.GPXTrackExporter.writeTrack(GPXTrackExporter.java:116)
04-09 14:28:48.193 25218 27877 E AndroidRuntime:        at de.dennisguse.opentracks.data.ShareContentProvider.lambda$openFile$1(ShareContentProvider.java:224)
04-09 14:28:48.193 25218 27877 E AndroidRuntime:        at de.dennisguse.opentracks.data.ShareContentProvider$$ExternalSyntheticLambda0.writeDataToPipe(Unknown Source:11)
04-09 14:28:48.193 25218 27877 E AndroidRuntime:        at android.content.ContentProvider$1.doInBackground(ContentProvider.java:2304)
04-09 14:28:48.193 25218 27877 E AndroidRuntime:        at android.os.AsyncTask$3.call(AsyncTask.java:394)
04-09 14:28:48.193 25218 27877 E AndroidRuntime:        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
04-09 14:28:48.193 25218 27877 E AndroidRuntime:        ... 3 more
```

I'm just not sure what the graceful mechanism of handling this is other than reverting back to testing the result of `cumulateSensorData()` before casting it into a `Distance` along the lines of:

```patch
--- a/src/main/java/de/dennisguse/opentracks/io/file/exporter/GPXTrackExporter.java
+++ b/src/main/java/de/dennisguse/opentracks/io/file/exporter/GPXTrackExporter.java
@@ -358,8 +358,9 @@ public class GPXTrackExporter implements TrackExporter {
                     trackPointExtensionContent += ("<opentracks:loss>" + ALTITUDE_FORMAT.format(cumulativeLoss) + "</opentracks:loss>\n");
                 }
 
-                cumulativeDistance = Distance.of(cumulateSensorData(trackPoint, sensorPoints, (tp) -> tp.hasSensorDistance() ? tp.getSensorDistance().toM() : null));
+                Double cumulativeDistanceDouble = cumulateSensorData(trackPoint, sensorPoints, (tp) -> tp.hasSensorDistance() ? tp.getSensorDistance().toM() : null);
                 if (cumulativeDistance != null) {
+                    cumulativeDistance = Distance.of(cumulativeDistanceDouble);
                     trackPointExtensionContent += ("<opentracks:distance>" + DISTANCE_FORMAT.format(cumulativeDistance.toM()) + "</opentracks:distance>\n");
                     trackPointExtensionContent += ("<cluetrust:distance>" + DISTANCE_FORMAT.format(trackDistance.plus(cumulativeDistance).toM()) + "</cluetrust:distance>\n");
                 }
```

Thoughts?

rgmf, 2022-04-09:
I think that happens when there's not sensor data.

rgmf, 2022-04-09:
PS/ I opened #1175 with the same issue. I'll close mine.

commit 0a810e71 (2022-04-11): Bugfix: NPE in GPXTrackExporter.
Introduced in 0f489f1cc9f7bf0a3f21a83ec9b1434aa94c7182

Fixes #1176.
closed 2022-04-11 by dennisguse