OpenTracksApp/OpenTracks #1183: Low Max Speed values are displayed

SanskritFritz, 2022-04-14:
The MAX SPEED value is wrong when a track is displayed in OpenTracks (it was recorded by OpenTracks, but imported gpx tracks are affected too). When I open the track in say OsmAnd, there is a correct value displayed. Example: OpenTracks displays MAX SPEED: 22.3 km/h while OsmAnd displays 35 km/h. All other values (like Distance, Average Moving Speed, or Moving Time) are correct). I can provide a gpx export if needed.
Here is a short table of the Max Speed values (in km/h) for some randomly chosen tracks:
|OpenTracks|OsmAnd|
|---------------|------------|
|24.6|28|
|22.8|34|
|21|38|
|9.4|18.8|
|25.2|33|

**Technical information**
 - Device: Samsung Galaxy S7
 - OS: LineageOS 16
 - OpenTracks version: 3.26.1 from Fdroid

dennisguse, 2022-04-14:
That is not unexpected as there are different options to compute the max speed.
At least:
* GPS estimated speed (OpenTracks still does this; there is some simple filtering)
* the traveled distance using the provided GPS coordinates

And GPS is not that precise, so is a multitude of options to filter the data before the actual computation.
Don't know how OsmAnd does the computation and the question is what the "correct" value actually is.
If you use a BLE sensor to measure speed/distance, it is way more precise.

Anyhow, there is the general plan to move to distance-based avg/max speed estimation for OpenTracks.

All computations are in `TrackStatisticsUpdater` if you want to take a look, but don't expect rocket science there :rocket:

rgmf, 2022-04-14:
More information: https://github.com/OpenTracksApp/OpenTracks/issues/1003

dennisguse, 2022-04-14:
@rgmf Somehow I had a feeling that there should be a ticket for this.
Thanks!

closed 2022-04-14 by dennisguse

SanskritFritz, 2022-04-14:
OK, fair enough. I don't know much about the calculation, but I can assure you that those values displayed for max speed by OpenTracks are **clearly** wrong. I know my speed on my bike and just look at the difference in the table. OsmAnd gets it right, but I can't tell you why or how. Anyway, it is what it is, if that is the final verdict, so be it. I still choose OpenTracks for tracking my rides. Thank you for developing this very useful app.

dennisguse, 2022-04-14:
@SanskritFritz the computation is just speed = distance / time. Nothing fancy.
I can try to create a quick (untested) hack in the following days if you want to try it out

And thanks for the info - it is something we should optimize and nudging every now and then helps to keep focus ;)

Anyhow, I can really recommend BLE sensors - then OpenTracks even works without GPS.

dennisguse, 2022-04-15:
@SanskritFritz #1185 can be tested.
Importing existing tracks is sufficient.
PS/ I hope nothing crashes as I have not tested it :sunglasses:

SanskritFritz, 2022-04-16:
Nice, thanks. I'm waiting for the next nightly build in Fdroid as I don't know how (yet) to build this myself.

SanskritFritz, 2022-04-16:
> All computations are in `TrackStatisticsUpdater` if you want to take a look, but don't expect rocket science there

> the computation is just speed = distance / time. Nothing fancy.

I looked at the code briefly and noticed that the calculation very much depends on the isMoving method. I didn't go into that, but why is that necessary at all? My current speed can be calculated as you wrote, speed=distance/time using the last two points. The time I got from the last point to my current is all that matters IMO, checking that the last point isMoving or not seems unnecessary to me. I suspect that isMoving depends on many factors that might not fulfilled in my case (like my GPS is pure in the sense it relies only on GPS signals, no network or whatever. Is there any speed/moving information given by GPS at all? Isn't it just a series of coordinate points?).

dennisguse, 2022-04-16:
GPS provides also the speed as well as a some accuracy measurements. No trouble there.
I guess we will change this in the near future, but one change at a time. We had quite some trouble with changes in this place as the code is quite old and nobody knows exactly why it was implemented this way. We already understood some hidden assumptions and documented it with test cases.

Regarding the Fdroid nightly: it is only build from the main branch. So you will not see this change until it is merged. GitHub also compiles a nightly per branch (same package name different signing key).

SanskritFritz, 2022-04-16:
Ah, so how can I test it? I should learn to build...

dennisguse, 2022-04-16:
@SanskritFritz when you are signed into GitHub you can download the build artifacts and there should be an APK.

Sadly this GitHub feature is well hidden.
Beside each commit you should either see a green arrow or a red cross. Just click on this and then there should be a build job called nightly. And  there you should find the APK.

For this build it should be here:

https://github.com/OpenTracksApp/OpenTracks/runs/6047956574?check_suite_focus=true

SanskritFritz, 2022-04-16:
Woa, I must agree that feature was very well hidden to me :smile: 
Anyway, I tested it, and it gives... different results. Sometimes bigger, other times less than the correct value. I can't put my finger on it there seems to be no correlation. Do you want me to share my gpx files with you?

dennisguse, 2022-04-16:
Kind of expected.

The actual question is: what was the actual max speed?
The measurements that you get are noisy and depending which approach is taken, the estimated max speed can vary a lot.

Here is a discussion at OsmAnd - also including some tools for analysis: https://github.com/osmandapp/OsmAnd/issues/9904

Can you re-create the table from above with the distance-based OpenTracks version?

Sharing the data won't help as the ground truth is missing :shrug:

SanskritFritz, 2022-04-16:
I had no idea GPS is that inaccurate regarding speed.
I took all my gpx files and here I present a table for comparison (yes, I double checked the sometimes huge vaues returned by the nightly). The older files were recorded with GPS Logger and imported into OpenTracks and the rest was recorded with OpenTracks itself.

|File|OpenTracks|OpenTracksNightly|OsmAnd|GPXSee|GPXViewer|GPSLogger|
|----|----------|-----------------|------|------|---------|---------|
|f94389b5|24.1|34.7|35|33.3|34.53|
|4b604b69|24.7|33.7|33|28.3|31.66|
|38d9d85d|22.3|53.3|35|34.6|37.61|
|52be44f3|24.6|32.9|28|27.9|30.54|
|16c27c5c|29.1|37.1|36|35.0|36.99|
|b555d7a6|22.8|34.4|34|32.4|34.46|
|bd8a1834|24.1|38.2|37|31.4|38.29|
|0cf2fd02|22.8|39.8|38|37.1|38.05|
|c91a4c8c|23.6|36.3|35|33.8|36.36|
|ba02d461|21.0|38.4|38|36.9|38.38|
|f27f5f74|22.7|31.6|28|27.0|30.78|
|58fa6d52|24.0|38.3|37|33.9|38.25|37|
|4b236bfc|19.4|38.7|39|37.0|37.69|39|
|b6cbf4f9|23.0|36.8|35|32.7|36.69|35|
|32917f3a|9.4|28.7|19|19.8|22.49|19|
|afa7c545|24.2|37.9|36|33.5|37.85|36|
|d87b12c4|24.2|37.4|37|33.4|37.33|37|
|b926cd16|22.1|74.8|35|33.3|44.01|35|
|570da83e|24.9|32.3|28|26.5|30.91|28|
|d88352be|25.2|33.3|33|31.2|33.21|33|
|43dd62f0|22.9|28.6|28|26.5|27.29|28|
|04653b5a|27.2|37.1|37|36.3|35.95|37|
|37e7796a|20.6|26.8|23|23.9|26.77|23|
|5ea0ad45|19.4|27.8|28|26.1|27.76||