OpenTracksApp/OpenTracks #1184: Crash when taking a photo

vsevjednom-cz, 2022-04-14:
**Describe the bug**
App disappears while creating photo or video. This thing happens again and again. But only sometimes randomly.

**To Reproduce**
1. Start tracking.
2. Create photo or video.
3. App is gone.

If applicable:
```
2022-04-14 21:26:04.110 21218-21243/? E/RestartUtil: Interrupted.
    java.lang.InterruptedException
        at java.lang.Thread.sleep(Native Method)
        at java.lang.Thread.sleep(Thread.java:386)
        at java.lang.Thread.sleep(Thread.java:327)
        at xba.run(:com.google.android.gms@220920026@22.09.20 (100306-434869283):2)
        at java.lang.Thread.run(Thread.java:784)
```
I hope this is right log because I can't repeat this crash again.

**Technical information**
 - Device: Honor 8S KSA-LX9
 - OS: Android 9
 - OpenTracks version: v3.26.1

dennisguse, 2022-04-14:
Mhhh, the stacktrace looks unrelated.
But we have: #1126

vsevjednom-cz, 2022-04-18:
I'm trying to call this error again 😱 but: application crashed, runlog only stops and logcat is empty. You can try to start tracking, paused it, minimize it, open camera and wait, wait, wait.

Log is cleared after crash. I see there only this. It doesn't help you I think.

```
2022-04-18 08:45:18.819 19606-19606/? I/...: Late-enabling -Xcheck:jni
2022-04-18 08:45:18.920 19606-19606/? I/...: Reinit property: dalvik.vm.checkjni= false
2022-04-18 08:45:18.978 19606-19606/? D/ZrHung.AppEyeUiProbe: AppEyeUIP created.
2022-04-18 08:45:18.988 19606-19606/? D/ActivityThread: Attach thread to application
2022-04-18 08:45:19.221 19606-19657/... I/HwApiCacheMangerEx: apicache path=/storage/emulated/0 state=mounted key=...#10247#256
2022-04-18 08:45:19.226 19606-19657/... I/HwApiCacheMangerEx: apicache path=/storage/emulated/0 state=mounted key=...#10247#0
2022-04-18 08:45:19.352 19606-19606/... I/...: QarthPatchMonintor::Init
2022-04-18 08:45:19.353 19606-19606/... I/...: 
2022-04-18 08:45:19.353 19606-19606/... I/...: QarthPatchMonintor::StartWatch
2022-04-18 08:45:19.353 19606-19606/... I/...: 
2022-04-18 08:45:19.353 19606-19606/... I/...: QarthPatchMonintor::WatchPackage: /data/hotpatch/fwkhotpatch/
2022-04-18 08:45:19.353 19606-19606/... I/...: 
2022-04-18 08:45:19.353 19606-19606/... I/...: QarthPatchMonintor::CheckAndWatchPatch: /data/hotpatch/fwkhotpatch/...
2022-04-18 08:45:19.353 19606-19606/... I/...: 
2022-04-18 08:45:19.354 19606-19606/... I/...: QarthPatchMonintor::CheckAndWatchPatch: /data/hotpatch/fwkhotpatch/all
2022-04-18 08:45:19.354 19606-19606/... I/...: 
2022-04-18 08:45:19.354 19606-19606/... I/...: QarthPatchMonintor::Run
2022-04-18 08:45:19.354 19606-19606/... I/...: 
2022-04-18 08:45:19.358 19606-19668/... I/...: QarthPatchMonintor::Reading
2022-04-18 08:45:19.359 19606-19668/... I/...: 
2022-04-18 08:45:19.359 19606-19668/... I/...: QarthPatchMonintor::CheckNotifyEvent
2022-04-18 08:45:19.359 19606-19668/... I/...: 
2022-04-18 08:45:19.359 19606-19668/... I/...: QarthPatchMonintor::CheckNotifyEvent before read
2022-04-18 08:45:19.359 19606-19668/... I/...: 
2022-04-18 08:45:19.380 19606-19657/... I/AwareBitmapCacher: init processName:... pid=19606 uid=10247
2022-04-18 08:45:19.568 19606-19606/... W/...: Accessing hidden field Landroid/os/Trace;->TRACE_TAG_APP:J (light greylist, reflection)
2022-04-18 08:45:19.569 19606-19606/... W/...: Accessing hidden method Landroid/os/Trace;->isTagEnabled(J)Z (light greylist, reflection)
2022-04-18 08:45:19.607 19606-19606/... I/Startup: ...; BuildType: debug; VersionName: v3.26.1-debug/v3.26.1 VersionCode: 4775
2022-04-18 08:45:19.678 19606-19606/... D/Startup: Enabling strict mode
2022-04-18 08:45:19.683 19606-19657/... E/MemoryLeakMonitorManager: MemoryLeakMonitor.jar is not exist!
2022-04-18 08:45:19.701 19606-19606/... D/StrictMode: StrictMode policy violation; ~duration=15 ms: android.os.strictmode.DiskReadViolation
        at android.os.StrictMode$AndroidBlockGuardPolicy.onReadFromDisk(StrictMode.java:1514)
        at java.io.UnixFileSystem.checkAccess(UnixFileSystem.java:251)
        at java.io.File.exists(File.java:815)
        at android.graphics.HwTypefaceUtil.updateFont(HwTypefaceUtil.java:154)
        at android.graphics.Typeface.loadSystemFonts(Typeface.java:1212)
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6728)
        at android.app.ActivityThread.access$2000(ActivityThread.java:273)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2020)
        at android.os.Handler.dispatchMessage(Handler.java:112)
        at android.os.Looper.loop(Looper.java:216)
        at android.app.ActivityThread.main(ActivityThread.java:7625)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:987)
2022-04-18 08:45:19.733 19606-19606/... D/StrictMode: StrictMode policy violation; ~duration=6 ms: android.os.strictmode.DiskReadViolation
        at android.os.StrictMode$AndroidBlockGuardPolicy.onReadFromDisk(StrictMode.java:1514)
        at java.io.UnixFileSystem.checkAccess(UnixFileSystem.java:251)
        at java.io.File.exists(File.java:815)
        at android.graphics.HwTypefaceUtil.getMultyWeightHwFamily(HwTypefaceUtil.java:94)
        at android.graphics.HwTypefaceUtil.updateFont(HwTypefaceUtil.java:160)
        at android.graphics.Typeface.loadSystemFonts(Typeface.java:1212)
        at android.app.ActivityThread.handleBindApplication(ActivityThread.java:6728)
        at android.app.ActivityThread.access$2000(ActivityThread.java:273)
        at android.app.ActivityThread$H.handleMessage(ActivityThread.java:2020)
        at android.os.Handler.dispatchMessage(Handler.java:112)
        at android.os.Looper.loop(Looper.java:216)
        at android.app.ActivityThread.main(ActivityThread.java:7625)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:524)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:987)
2022-04-18 08:45:19.741 19606-19606/... I/TrackRecordingService: new status null -> RecordingStatus{trackId=null, paused=false}
2022-04-18 08:45:19.777 19606-19606/... I/TrackRecordingService: new status RecordingStatus{trackId=null, paused=false} -> RecordingStatus{trackId=null, paused=false}
2022-04-18 08:45:19.789 19606-19606/... D/AwareAppScheduleManager: webViewOpt, not safe 
```

dennisguse, 2022-04-18:
Can you create a video recording?
And is it always reproducible?

vsevjednom-cz, 2022-04-19:
Yes, it's reproducible when tracking is paused 💔. I'll create some video tomorrow.

dennisguse, 2022-04-19:
Are you trying to create a marker from within OpenTracks (that should only be possible while recording)?
Or are you switching to camera application?

vsevjednom-cz, 2022-04-19:
I'm switching to the camera.

dennisguse, 2022-04-19:
Then you are likely experiencing this:  #1156.
There is a bug in the lifecycle handling of `TrackRecordingActivity` or `StatisticsRecordingFragement` - some kind of race condition, when switching to another application.

vsevjednom-cz, 2022-04-20:
It's the same problem maybe. But I don't clicking to the Map button.

It crashes when I'm using camera or anything else. Tinder is good for crash, hahahaha 😋. Look at this video. There was a crash and logcat was cleared. There are too many cuts in video because video must be very short here 🤪.

https://user-images.githubusercontent.com/56153843/164195252-92ab0f46-07df-4efd-a7c0-392c3f8bdf15.mp4

dennisguse, 2022-04-20:
That's a tough one...
One aspect: a crash in the UI also kills the TrackRecordingService - so the behavior is kind of expected.
But the missing logcat...
It still feels like a state management problem.
And good to know that the recording was paused!

vsevjednom-cz, 2022-04-20:
It crash sometimes when recording is not paused - but less often than when recording is paused. I must check recording everytime after I did something with my phone.

dennisguse, 2022-04-21:
Damn that doesn't sound good.
I will try to take a look in the upcoming week.

dennisguse, 2022-04-24:
Hopefully fixed by https://github.com/OpenTracksApp/OpenTracks/commit/6741ef4d9fef99d8a13c7a2f952aa47268e2efd5

@vsevjednom-cz can you check and re-open if not?

vsevjednom-cz, 2022-04-25:
Crash while pause looks solved but crash while tracking is still there. I'll try to get some logcat in next days.

dennisguse, 2022-04-25:
@vsevjednom-cz there is always one more bug to hunt, right? 
:sunglasses:

vsevjednom-cz, 2022-04-26:
Oh, crash while pause still exists. I open calendar today = crash. I'll write here more info this week.

vsevjednom-cz, 2022-05-01:
Or try to take a panoramic photo. Panoramic photo = more crashes 😋.

dennisguse, 2022-05-01:
Another hypothesis: Could it be that your phone is killing OpenTracks due to lack of memory?
I mean the crash is not taking place immediately and you do not re-open OpenTracks again, right?
That suggests that the problem is outside of OpenTracks somewhere.

One thing you could try: 
1. restart the device,
2. start recording with opentracks
3. start one other app and do something

Repeat 3 until OpenTracks crashes.

What I am wondering is that the logcat is deleted as well.
That suggests some problem on operating system/ runtime level rather than application level.

vsevjednom-cz, 2022-05-03:
Interesting idea, sounds good (for you, not for me). Maybe you're right. I tried crash before restart - no problem. But there was not any crash after restart.
So you can take back your last commit in https://github.com/OpenTracksApp/OpenTracks/commit/6741ef4d9fef99d8a13c7a2f952aa47268e2efd5
So I'll try to restart phone always before tracking. `Memory info` app still shows I've got free about 500 MB RAM. That should be enough but it was a cheap phone 🤪.

dennisguse, 2022-05-03:
:)

PS the commit was actually fixing a bug.
Not the one you described, but I found it while looking for yours ;)

closed 2022-05-03 by vsevjednom-cz

vsevjednom-cz, 2022-05-07:
You're absolutely right!
Simple solution:
1. Show recent apps.
2. Swipe up every app.
3. OpenTracks crash is away.

dennisguse, 2022-05-07:
:sunglasses: