OpenTracksApp/OpenTracks #1186: [Request] Option to not put prefix on export track name.

sgofferj, 2022-04-15:
**Is your feature request related to a problem? Please describe.**
Opentracks puts an 8 digit hex number at the beginning of the name of each exported file. This number is not chronological and messes with the sorting of the tracks in the file system or in apps that can't sort by file date/time.

**Describe the solution you'd like**
I would like to suggest an option to disable that prefix (or set a cutom one) so that exported files contain only date and time.

**Describe alternatives you've considered**
Sorting by file date/time is possible but only in filesystem browsers and also only make sense if the files are transferred from the phone to a computer in the correct sequence (not all transfer methods preserve creatinon date/time from the phone).

dennisguse, 2022-04-15:
Related: #512.
And we need to do something about this.

PR welcome.

closed 2022-04-30 by dennisguse

dennisguse, 2022-04-30:
https://github.com/OpenTracksApp/OpenTracks/pull/1207